theyosh/TerrariumPI #563: [BUG]: V4 JS Gauge issue "Uncaught TypeError: e.toFixed is not a function"

chrRtg, 2021-07-09:
**More info**
After logging out on the next reload a JS error shows up. On the dashboard at least one gauge is not shown. If I switch to any sensor the gauge is in a disabled or not initalized state and the line chart shows loading spinner. 
To resolve this situation a new login is sufficient.

To illustrate the problem I had attached three screenshots.

@theyosh if you have something I should try to locate the problem please send me a message, happy to support such great product.

**Setup:**
 - TerrariumPI version: 4.x.y.z
 - Python version:3
 - Raspberry PI: 3B

**Screenshots**
The problem in debug console
[image]

The missing Gauge
[image]

A typical Sensor View
[image]

theyosh, 2021-07-13:
Hi,

I see this sometimes, and a reload fixes it. So for now it is a low priority issue.

theyosh, 2021-07-28:
Hmm, I think I found this  by accident. Could it be that you had also the option to show the min and max values set to on in the settings page? If so, then this should be fixed now.

Do a `git pull` and refresh your internet browser.

chrRtg, 2021-08-21:
Sorry for the late reply @theyosh, had to build the terrarium first.

With the newest V4 the problem is no longer appearing. I'll close the tickets while I consider the issue solved.

Thanks mate!

commit 239a6ada (2021-07-28): Fix Gauge labels. Fix #563
closed 2021-08-21 by chrRtg