theyosh/TerrariumPI #567: [BUG]: Update sispmctl to latest version (4.9)

victorhooi, 2021-07-14:
This is related to this discussion - https://github.com/theyosh/TerrariumPI/discussions/566

Current version of `sispmctl` in TerrariumPi is quite outdated (version 3.1 from 2011).

The latest version at time of writing is 4.9:

https://sourceforge.net/projects/sispmctl/files/sispmctl/sispmctl-4.9/

This seems to fix some bugs (e.g. being able to check the version without an error message), and also adds support for newer boards like the EG-PM2S).

Is it possible to update the sispmctl installed in TerrariumPi? (Would we have to compile our own one here?)

theyosh, 2021-07-19:
I think this will be a FAQ item or other documentation. Building this package during installation will introduce more time. And not everybody needs it. It needs to be compiled manual on the Pi.

So for now I stick with the the outdated OS version. As my code has support for the new version. This is done by checking two different locations on disk. And this keep the installer at a decent speed.

theyosh, 2021-08-13:
Added it to the new documentation website: https://theyosh.github.io/TerrariumPI/hardware/relay/eg-pm-usb/#outdated-software

So that should be enough I think.

closed 2021-08-13 by theyosh