theyosh/TerrariumPI #568: [BUG]: V4 ModuleNotFoundError: No module named 'gevent' after install

Dave0408, 2021-07-16:
Hello @theyosh 

**Setup:**
Raspberry Pi Model B Plus Rev 1.2

Python 3.7.3
Python 2.7.16

Raspberry Pi OS Lite
Release date: May 7th 2021
Kernel version: 5.10

**Describe the bug**
I followed the steps for a new install of V4 as described [here](https://github.com/theyosh/TerrariumPI/tree/4.x.y.z#installation)
I used a freshly installed 32GB sdcard. 
After first restart (part of installation routine) terrariumpi isn't starting. I noticed that my installation process took only a few minutes (5-10) and definitely not the mentioned 45 minutes. 

The install process finished with:
`TerrariumPI installation is rebooting the Raspberry PI in 5 seconds...`
`TerrariumPI installation is rebooting the Raspberry PI in 4 seconds...`
`TerrariumPI installation is rebooting the Raspberry PI in 3 seconds...`
`TerrariumPI installation is rebooting the Raspberry PI in 2 seconds...`
`TerrariumPI installation is rebooting the Raspberry PI in 1 seconds...`

In /var/log/syslog i found:
`May  7 17:40:43 raspberrypi python[427]: Traceback (most recent call last):`
`May  7 17:40:43 raspberrypi python[427]:   File "/home/pi/TerrariumPI/terrariumPI.py", line 4, in <module>`
`May  7 17:40:43 raspberrypi python[427]:     from gevent import monkey`
`May  7 17:40:43 raspberrypi python[427]: ModuleNotFoundError: No module named 'gevent'`
`May  7 17:40:43 raspberrypi systemd[1]: terrariumpi.service: Main process exited, code=exited, status=1/FAILURE`
`May  7 17:40:43 raspberrypi systemd[1]: terrariumpi.service: Failed with result 'exit-code'.`
`May  7 17:40:43 raspberrypi systemd[1]: Failed to start TerrariumPI.`

I found this [this](https://github.com/theyosh/TerrariumPI/issues/536) issue but had no success.
> I was able to fix this by ensuring version 3.7 of Python was always the default with the command:
> 
> sudo update-alternatives --install /usr/bin/python python /usr/bin/python2.7 1
> sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.7 2

Debugging with screen wasn't possible  well. 
`pi@raspberrypi:~ $ screen -r
-bash: screen: command not found`

Any suggestion on this?

Thanks :)

theyosh, 2021-07-16:
Hi,

hmm, strange, but the easiest thing to do is rerun the installer: `sudo ./install.sh`

With TP4 I dropped the screen system. So looking in the syslog is the right thing to do now.

And I use now a virtual Python environment, so using the `update-alternatives` will not change anything for my software.

So, try again the installer and I hope it will work..

theyosh, 2021-07-16:
> After first restart (part of installation routine) terrariumpi isn't starting. I noticed that my installation process took only a few minutes (5-10) and definitely not the mentioned 45 minutes.

Cool, but that could be to fast....

Dave0408, 2021-07-18:
After updating/upgrading the raspbian software and manually setting the correct time, the installation now worked. As expected, it took longer than before. 

The only "error" I have seen was this incorrect number. It was alerted multiple times for different statements:

` ./7nstall.sh: Zeile 207: printf: 27.60: Ungültige Zahl.`

closed 2021-07-18 by Dave0408