theyosh/TerrariumPI #569: [BUG]: Connection to Meross power switch instable

Dave0408, 2021-07-24:
Hello @theyosh ,

**Setup:**
 - Python version: 3
 - Raspberry PI: Raspberry Pi Model B Plus Rev 1.2 

**Describe the bug**
`2021-07-24 07:56:01,545 - ERROR   - __init__             - Error changing state for relay Meross power switches relay named 'Beregnungsanlage' at address '1912207582162990807148e1e91455fa,2' to 100.0 :Error changing relay Meross power switches relay named 'Beregnungsanlage' at address '1912207582162990807148e1e91455fa,2' to state 100.0. Error:`
Unfortunately, the above message is not very meaningful.

I also had this problem under the old version. I had hoped that it could be solved with the TP4. But there still seems to be occasional interruptions in communication with the Meross power strip. I have the model mss425e with firmware 3.1.8 and hardware 3.0.0. Wifi seems to be stable at 100%.

The problem is temporarily solved when I restart the terrariumpi.service. Can anyone report similar behavior? Is there a solution to this?

During restart processing I found this Error messages. I do not know if it is related to the problem described above.

```
Jul 24 07:56:48 raspberrypi python[13385]: 2021-07-24 07:56:48,645 - INFO    - terrariumCloud       - Stopping Meross cloud ...
Jul 24 07:56:49 raspberrypi python[13385]: 2021-07-24 07:56:49,432 - INFO    - terrariumCloud       - Closed Meross cloud connection
Jul 24 07:56:49 raspberrypi python[13385]: 2021-07-24 07:56:49,457 - INFO    - terrariumPI          - Shutdown TerrariumPI 4.0.0 done. Bye bye ...
Jul 24 07:56:49 raspberrypi python[13385]: Exception in thread zeroconf-ServiceBrowser__http._tcp.local._2855295712:
Jul 24 07:56:49 raspberrypi python[13385]: Traceback (most recent call last):
Jul 24 07:56:49 raspberrypi python[13385]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Jul 24 07:56:49 raspberrypi python[13385]:     self.run()
Jul 24 07:56:49 raspberrypi python[13385]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/zeroconf/__init__.py", line 1739, in run
Jul 24 07:56:49 raspberrypi python[13385]:     state_change=service_type_state_change[1],
Jul 24 07:56:49 raspberrypi python[13385]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/zeroconf/__init__.py", line 1495, in fire
Jul 24 07:56:49 raspberrypi python[13385]:     h(**kwargs)
Jul 24 07:56:49 raspberrypi python[13385]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/zeroconf/__init__.py", line 1593, in on_change
Jul 24 07:56:49 raspberrypi python[13385]:     listener.add_service(*args)
Jul 24 07:56:49 raspberrypi python[13385]:   File "/home/pi/TerrariumPI/hardware/relay/sonoff_relay.py", line 127, in add_service
Jul 24 07:56:49 raspberrypi python[13385]:     for switch_nr in range(len(data['Status']['FriendlyName'])):
Jul 24 07:56:49 raspberrypi python[13385]: TypeError: string indices must be integers
Jul 24 07:56:49 raspberrypi python[13385]: /home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py:3345: DeprecationWarning: Absolute template path names are deprecated.
Jul 24 07:56:49 raspberrypi python[13385]:   fname = self.search(name, self.lookup)
Jul 24 07:56:49 raspberrypi python[13385]: /home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/pywsgi.py:930: ResourceWarning: unclosed file <_io.BufferedReader name='/home/pi/TerrariumPI/log/terrariumpi.log'>
Jul 24 07:56:49 raspberrypi python[13385]:   for data in self.result:
Jul 24 07:56:49 raspberrypi python[13385]: ResourceWarning: Enable tracemalloc to get the object allocation traceback
Jul 24 07:56:49 raspberrypi python[13385]: Exception in thread Thread-3:
Jul 24 07:56:49 raspberrypi python[13385]: Traceback (most recent call last):
Jul 24 07:56:49 raspberrypi python[13385]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Jul 24 07:56:49 raspberrypi python[13385]:     self.run()
Jul 24 07:56:49 raspberrypi python[13385]:   File "/usr/lib/python3.7/threading.py", line 865, in run
Jul 24 07:56:49 raspberrypi python[13385]:     self._target(*self._args, **self._kwargs)
Jul 24 07:56:49 raspberrypi python[13385]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/paho/mqtt/client.py", line 3452, in _thread_main
Jul 24 07:56:49 raspberrypi python[13385]:     self.loop_forever(retry_first_connection=True)
Jul 24 07:56:49 raspberrypi python[13385]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/paho/mqtt/client.py", line 1779, in loop_forever
Jul 24 07:56:49 raspberrypi python[13385]:     rc = self.loop(timeout, max_packets)
Jul 24 07:56:49 raspberrypi python[13385]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/paho/mqtt/client.py", line 1181, in loop
Jul 24 07:56:49 raspberrypi python[13385]:     rc = self.loop_read(max_packets)
Jul 24 07:56:49 raspberrypi python[13385]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/paho/mqtt/client.py", line 1572, in loop_read
Jul 24 07:56:49 raspberrypi python[13385]:     rc = self._packet_read()
Jul 24 07:56:49 raspberrypi python[13385]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/paho/mqtt/client.py", line 2310, in _packet_read
Jul 24 07:56:49 raspberrypi python[13385]:     rc = self._packet_handle()
Jul 24 07:56:49 raspberrypi python[13385]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/paho/mqtt/client.py", line 2944, in _packet_handle
Jul 24 07:56:49 raspberrypi python[13385]:     return self._handle_suback()
Jul 24 07:56:49 raspberrypi python[13385]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/paho/mqtt/client.py", line 3151, in _handle_suback
Jul 24 07:56:49 raspberrypi python[13385]:     self, self._userdata, mid, granted_qos)
Jul 24 07:56:49 raspberrypi python[13385]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/meross_iot/manager.py", line 518, in _on_subscribe
Jul 24 07:56:49 raspberrypi python[13385]:     _LOGGER.info(f"Updated {len(list(results))} devices.")
Jul 24 07:56:49 raspberrypi python[13385]: RuntimeError: await wasn't used with future
Jul 24 07:56:49 raspberrypi python[13385]: KeyboardInterrupt
Jul 24 07:56:49 raspberrypi python[13385]: 2021-07-24T05:56:48Z
Jul 24 07:56:52 raspberrypi systemd[1]: terrariumpi.service: Succeeded.
Jul 24 07:56:52 raspberrypi systemd[1]: Stopped TerrariumPI.
```

Since the misting system, the pump and also the lighting are connected to the Meross power strip, this is an essential component and must work reliably. 

Thank you very much.

theyosh, 2021-07-26:
Hi,

I have a MS100 and a mss425e and they work well. I did have almost the same error you have, but I think I fixed that. So in order to make sure you have the latest version do a `git pull`.
Them re-run the installer `sudo ./install.sh`. As this will change some libraries regarding this bug. Then restart the Pi just once to make sure everything is started correctly.

The only other thing is that you use a old RPi? I am testing on a RPi4 and RPi3. And I have the feeling that there is a difference how the RPi works based on RPi type, even when you have the latest OS installed.

As there is now more code running at the same time, it could be that a Pi older then RPi3 is not working correctly.... But I can't test that.

And, a bit strange maybe, but do you use Sonof devices with Tasmota software?

Dave0408, 2021-07-26:
> make sure you have the latest version do a `git pull`.
I will try that!

> The only other thing is that you use a old RPi?
I am using a Raspberry Pi Model B Plus Rev 1.2
Maybe a little bit old?! 

> And, a bit strange maybe, but do you use Sonof devices with Tasmota software?
No, i don't use Sonoff devices.

theyosh, 2021-07-26:
I think Pi Model B Plus Rev 1.2 is a bit old. It is still single core and armv6. So I would suggest to go with a newer PI. Will also make it a bit faster.

https://learn.adafruit.com/introducing-the-raspberry-pi-2-model-b?view=all#performance-improvements

As I did not test it on older Pis, I do not know if it works...

Dave0408, 2021-07-26:
Okay I will update/change my hardware. 

> I am testing on a RPi4

Which version would you recommend? 2GB, 4GB or even 8GB? Are 2 enough for TP4?

theyosh, 2021-07-26:
2GB is enough.

Dave0408, 2021-08-02:
The new raspberry arrived and is in use now. I will observe it for a couple of days...

Dave0408, 2021-08-17:
Now runs on thePI4 without problems and much more reliably. And by the way, it's noticeably faster.

Dave0408, 2021-09-07:
Hello @theyosh ,

unfortunately, I have to reopen this issue. 
Now, in the TP4 version with the new raspberry pi 4, it has happened repeatedly that the light does not turn off. 

Attached is the filtered error log:

```
2021-09-07 20:47:11,774 - ERROR   - terrariumEngine      - Engine can't keep up. For 202 times it could not finish in 30.0 seconds.
2021-09-07 20:47:11,609 - ERROR   - hardware.relay       - Error changing state for relay Meross power switches relay named 'Arcadia Jungle LED' at address '1912207582162990807148e1e91455fa,1' to 0.0 :Error changing relay Meross power switches relay named 'Arcadia Jungle LED' at address '1912207582162990807148e1e91455fa,1' to state 0.0. Error: 
2021-09-07 20:46:38,165 - ERROR   - terrariumEngine      - Engine can't keep up. For 201 times it could not finish in 30.0 seconds.
2021-09-07 20:46:37,992 - ERROR   - hardware.relay       - Error changing state for relay Meross power switches relay named 'Arcadia Jungle LED' at address '1912207582162990807148e1e91455fa,1' to 0.0 :Error changing relay Meross power switches relay named 'Arcadia Jungle LED' at address '1912207582162990807148e1e91455fa,1' to state 0.0. Error: 
2021-09-07 20:46:04,501 - ERROR   - terrariumEngine      - Engine can't keep up. For 200 times it could not finish in 30.0 seconds.
2021-09-07 20:46:04,331 - ERROR   - hardware.relay       - Error changing state for relay Meross power switches relay named 'Arcadia Jungle LED' at address '1912207582162990807148e1e91455fa,1' to 0.0 :Error changing relay Meross power switches relay named 'Arcadia Jungle LED' at address '1912207582162990807148e1e91455fa,1' to state 0.0. Error:
```

It says: **Engine can't keep up.**

[image]

I would currently reject the possibility of an overload of the Pi.

Any idea?

theyosh, 2021-09-08:
Hmm, there is 1 scenario which I could not fix. When your Pi lose internet connection, it loses also the meross connection. And this is hard to detect for me. I have not found a way to figure this out. And it could also be that when your DSL router is rebooting, you loose the Meross connection, even when the Pi does not notice it.

Only solution, restart TerrariumPI so it logs in again.

About the update duration. This is a indicator there is something slowing it down. You can see that in the log file `log/terrariumpi.log`. Every update part has a duration. So, try to see which action(s) takes the most seconds. But there is no 1 single thing causing this. It could be Meross related, but I can't judge without a full log file.

So, I guess, do a full reboot of the Pi and see if it is than still happening....

theyosh, 2022-02-03:
So, there is some progress with this. In the 4.x.y.z branch there is a new update of this Meross library which looks like does better connection restoring. But I have not tested it fully, but I had a internet connection interruption this morning, and my TP did recover from it correctly.

I hope to release a new version within a week from now

theyosh, 2022-03-04:
So, I have changed the code a bit more, and for now, I can now recover from different kind of network errors. And it is hard to test, but I can see in my logs, that reconnect do occur. I have a not so stable internal network, and I see 1 or 2 reconnects a week. But it keeps working and it does recover.

The code is in the 4.x.y.z branch. You could test this if you want, but I think it is a bit hard to do so. So I leave this up to you. This code will be in the next release end of this month I guess.

Dave0408, 2022-03-11:
Hey @theyosh ,

thanks a lot. I'll wait for the update end of march. Then i will try to test it in my environment. But your experiences sound promising so far.

theyosh, 2022-04-08:
Any updates by any change?

Dave0408, 2022-04-10:
> Any updates by any change?

Hey, 
as mentioned i want to wait for the next release in the main branch and not swap to the 4.x.y.z branch. 
The planned "end of march" release isn't published yet or did i miss something?

theyosh, 2022-04-10:
Oh yeah. You are right. And no, there is not a new version yet. I am preparing one, and was going over the issues to see if everything is fixed correctly so I can make a new release.

But then this is clear. I will start a new release within a few days.

Dave0408, 2022-07-26:
Good evening, 

I wanted to give a short feedback on this issue. 
Since April, some time has passed and even several interruptions of the WIFI connection as well as changes of the complete WIFI infrastructure were made. There were no problems or abnormalities anymore.

In my opinion, we can now close this point.

theyosh, 2022-08-01:
Thanks for the feedback. Closing now.

closed 2022-08-01 by theyosh