theyosh/TerrariumPI #570: [BUG]:

PhoenixNP, 2021-07-25:
**Setup:**
 - Software version: 4
 - Raspberry PI: 3B

**Describe the bug**
Denkovi V2 is unable to add addresses.
It will fail with a generic error
If you add it without placing the commandline tool and then place the tool it will still error out
Jul 26 01:13:23 terrarium01 python[545]: hardware.relay.terrariumRelayUpdateException: Error getting new data from relay Denkovi Relay v2 (4 sockets) relay named 'Hydor_Socket4_Denkovi' at address '4'. Error: unknown
Jul 26 01:13:23 terrarium01 python[545]: 2021-07-26 01:13:23,050 - WARNING - terrariumEngine      - Could not take a new measurement from relay denkovi_v2_4 relay named 'Hydor_Socket4_Denkovi' at address '4'. Tried for 1.0

PhoenixNP, 2021-07-26:
It's missing an
"import re"
But there is more missing/wrong

theyosh, 2021-07-26:
Hi,

thanks for testing this relay, as I do not own one. So I could not test it. I found some issues in the code, and made some small fixes. Could you do a `git pull` and see if you can now add a relay.

Also I updated the README file how to install Java. But I see that you already have fixed that.

PhoenixNP, 2021-07-26:
It does get added now but still throws a load of errors;

2021-07-26 10:13:21,463 - WARNING - terrariumEngine      - Could not take a new measurement from relay denkovi_v2_4 relay named 'XXNothing_Socket1_Denkovi' at address '1,0001691995'. Tried for 1.04 seconds. Skipping this update.
hardware.relay.terrariumRelayUpdateException: Error getting new data from relay Denkovi Relay v2 (4 sockets) relay named 'XXNothing_Socket1_Denkovi' at address '1,0001691995'. Error: unknown
raise terrariumRelayUpdateException(f'Error getting new data from relay {self}. Error: unknown')
File "/opt/TerrariumPI/hardware/relay/__init__.py", line 159, in __get_hardware_value
return f()
File "/opt/TerrariumPI/venv/lib/python3.7/site-packages/retry/api.py", line 33, in __retry_internal
logger)
File "/opt/TerrariumPI/venv/lib/python3.7/site-packages/retry/api.py", line 74, in retry_decorator
return caller(func, *(extras + args), **kw)
File "/opt/TerrariumPI/venv/lib/python3.7/site-packages/decorator.py", line 232, in fun
new_data = self.__get_hardware_value()
File "/opt/TerrariumPI/hardware/relay/__init__.py", line 237, in update
Traceback (most recent call last):
2021-07-26 10:13:21,457 - ERROR   - __init__             - Error getting new data from relay Denkovi Relay v2 (4 sockets) relay named 'XXNothing_Socket1_Denkovi' at address '1,0001691995'. Error: unknown
2021-07-26 10:13:21,453 - ERROR   - __init__             - Error getting new data from relay Denkovi Relay v2 (4 sockets) relay named 'XXNothing_Socket1_Denkovi' at address '1,0001691995'. Error: 'terrariumRelayDenkoviV2_4' object has no attribute '_relay_cache_key'

PhoenixNP, 2021-07-26:
pi@terrarium01:/opt/TerrariumPI $ sudo java -jar 3rdparty/DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar ID=0 4v2 all status
0000
pi@terrarium01:/opt/TerrariumPI $ sudo java -jar 3rdparty/DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar 0001691995 4v2 all status
0000

It doesn't seem to switch it either

theyosh, 2021-07-26:
Hi,

made a small change in the code. Could you do a `git pull` and restart: `sudo service terrariumpi restart`

That should fix the above issue. 

The Java lines, are results of a status request. And not a toggle request. So that is not the right command:

sudo java -jar 3rdparty/DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar 0001691995 4v2 1 1

Should toggle a relay to on....

PhoenixNP, 2021-07-26:
Yeah I know, toggling it manually will work to toggle the switches =)

However I still get an error

2021-07-26 13:11:37,412 - WARNING - terrariumEngine      - Could not take a new measurement from relay denkovi_v2_4 relay named 'Nothing' at address '1,0001691995'. Tried for 1.10 seconds. Skipping this update.
hardware.relay.terrariumRelayUpdateException: Error getting new data from relay Denkovi Relay v2 (4 sockets) relay named 'Nothing' at address '1,0001691995'. Error: unknown
raise terrariumRelayUpdateException(f'Error getting new data from relay {self}. Error: unknown')
File "/opt/TerrariumPI/hardware/relay/__init__.py", line 159, in __get_hardware_value
return f()
File "/opt/TerrariumPI/venv/lib/python3.7/site-packages/retry/api.py", line 33, in __retry_internal
logger)
File "/opt/TerrariumPI/venv/lib/python3.7/site-packages/retry/api.py", line 74, in retry_decorator
return caller(func, *(extras + args), **kw)
File "/opt/TerrariumPI/venv/lib/python3.7/site-packages/decorator.py", line 232, in fun
new_data = self.__get_hardware_value()
File "/opt/TerrariumPI/hardware/relay/__init__.py", line 237, in update
Traceback (most recent call last):
2021-07-26 13:11:37,404 - ERROR   - __init__             - Error getting new data from relay Denkovi Relay v2 (4 sockets) relay named 'Nothing' at address '1,0001691995'. Error: unknown
2021-07-26 13:11:37,399 - ERROR   - __init__             - Error getting new data from relay Denkovi Relay v2 (4 sockets) relay named 'Nothing' at address '1,0001691995'. Error: 'serial'
2021-07-26 13:11:36,891 - WARNING - api                  - Error getting new data from relay Denkovi Relay v2 (4 sockets) relay named 'Nothing' at address '1,0001691995'. Error: unknown, retrying in 0.5 seconds...
2021-07-26 13:11:36,887 - ERROR   - __init__             - Error getting new data from relay Denkovi Relay v2 (4 sockets) relay named 'Nothing' at address '1,0001691995'. Error: 'serial'
2021-07-26 13:11:36,380 - WARNING - api                  - Error getting new data from relay Denkovi Relay v2 (4 sockets) relay named 'Nothing' at address '1,0001691995'. Error: unknown, retrying in 0.5 seconds...
2021-07-26 13:11:36,376 - ERROR   - __init__             - Error getting new data from relay Denkovi Relay v2 (4 sockets) relay named 'Nothing' at address '1,0001691995'. Error: 'serial'

PhoenixNP, 2021-07-26:
Latest pull also add's the following error;

2021-07-26 13:15:38,801 - ERROR   - terrariumEngine      - Error loading relay denkovi_v2_4 relay named 'Nothing' at address '1,0001691995' with error: Could not load hardware for relay Denkovi Relay v2 (4 sockets) relay named 'Nothing' at address '1,0001691995': unsupported operand type(s) for -=: 'str' and 'int'.
2021-07-26 13:15:38,294 - WARNING - api                  - Could not load hardware for relay Denkovi Relay v2 (4 sockets) relay named 'Nothing' at address '1,0001691995': unsupported operand type(s) for -=: 'str' and 'int', retrying in 0.5 seconds...
2021-07-26 13:15:37,787 - WARNING - api                  - Could not load hardware for relay Denkovi Relay v2 (4 sockets) relay named 'Nothing' at address '1,0001691995': unsupported operand type(s) for -=: 'str' and 'int', retrying in 0.5 seconds...

theyosh, 2021-07-26:
Crap,

Another approach. Are you willing and able to setup SSH access and port forward to your pi? Then I can try it myself through SSH. 

If this is an option, please mail the SSH login credentials to: github@theyosh.nl and then I will try to debug it on your pi and make a fix. I thinks we are close.

PhoenixNP, 2021-07-26:
Check your e-mail =)

theyosh, 2021-07-26:
Nice, and fixed! It is running now correctly. And I added a relay with address 4 (the latest) and that seems to work.

PhoenixNP, 2021-07-26:
Nice! =)

Shall we close the issue? (I will test it and in case you don't hear anything; it's fixed haha)

theyosh, 2021-07-26:
That's cool. I close it now. Just re-open it when you find something. Do not forget to close the port forward to your Pi.

closed 2021-07-26 by theyosh