theyosh/TerrariumPI #571: [BUG]:

PhoenixNP, 2021-07-26:
**Setup:**
Software version 4
 - Raspberry PI: 3B

**Describe the bug**
Delay on switching relay's does not work.
It logs that it will set a delay, but the delay is not enforced. Thus the relay is switched instantly.

**To Reproduce / Installation steps**
Denkovi V2 board used

**Expected behavior**
Switch off the relay, but delayed. For example; Switch it 30 minutes later or earlier.

theyosh, 2021-07-28:
Hi,

Maybe this needs a bit of explanation. The delay is set in the area setup:
[image]
And that means that the delay will trigger when the lights should go on (in my case). This happens when the day starts.

Manually toggling does not delay. As the area has that setting, and not the relay.

And then there is some extra logic, which will ignore the delay when the lights (in my case) should be already on. So if you set this setting during the day, the first action will be toggle on direct, as the relay should already be on.
The next day, I expect the delay to work, and that your relay is toggling on 30 minutes after the timer time.

The used relay should not be of influence of this. As this is an area setting.

PhoenixNP, 2021-07-28:
Woops, I fixed the title! =)

It appears the delay "off" is not set correctly.
I will check it again this evening, but it appears that both switches turn off at the same time, even tho I have a "Relay delay off" set for 30 minutes =)

PhoenixNP, 2021-07-28:
2021-07-28 21:34:49,206 - INFO    - terrariumEngine      - Updated relay denkovi_v2_4 relay named 'T5_UVB_Socket2_Denkovi' at address '2,0001691995' with new value 0.00 in 0.01 seconds.
**2021-07-28 21:34:49,194 - INFO    - terrariumEngine      - Updated relay denkovi_v2_4 relay named 'JungleDawn_Socket3_Denkovi' at address '3,0001691995' with new value 0.00 in 0.01 seconds.**
2021-07-28 21:34:49,180 - INFO    - terrariumEngine      - Updated relay denkovi_v2_4 relay named 'Nothing_Socket1_Denkovi' at address '1,0001691995' with new value 0.00 in 0.01 seconds.
2021-07-28 21:34:49,162 - INFO    - terrariumEngine      - Updated relay denkovi_v2_4 relay named 'Hydor_Socket4_Denkovi' at address '4' with new value 0.00 in 0.02 seconds.
2021-07-28 21:34:49,081 - INFO    - terrariumEngine      - Starting a new update round with 2 sensors, 4 relays, 0 buttons and 0 webcams.
2021-07-28 21:34:26,095 - INFO    - terrariumEngine      - Engine update done in 7.02 seconds. Waiting for 22.98 seconds for the next round.
2021-07-28 21:34:25,887 - INFO    - terrariumEngine      - Updated Enclosure Rainbow with 1 areas in 5.16 seconds.
2021-07-28 21:34:25,878 - INFO    - terrariumArea        - Set the relay T5_UVB_Socket2_Denkovi to 0.0 with a delay of 0.0 minutes
**2021-07-28 21:34:25,877 - INFO    - terrariumArea        - Set the relay JungleDawn_Socket3_Denkovi to 0.0 with a delay of 0.0 minutes**
2021-07-28 21:34:25,876 - INFO    - terrariumEngine      - Updated Enclosure Rainbow with 1 areas in 0.01 seconds.
2021-07-28 21:34:20,726 - INFO    - terrariumArea        - Toggle the relays for area Lighting named Main_Lights to state off.
2021-07-28 21:34:20,724 - INFO    - terrariumArea        - Toggle off the relays for area Lighting named Main_Lights.
2021-07-28 21:34:20,687 - INFO    - terrariumEngine      - Updated sensor owfs temperature named 'Middle-Temperature' at address '28FF968047160355' with new value 28.25C in 0.72 seconds.
2021-07-28 21:34:19,959 - INFO    - terrariumEngine      - Updated sensor owfs temperature named 'SprayWater-Temperature' at address '28FF3CBE47160386' with new value 31.94C in 0.85 seconds.
2021-07-28 21:34:19,222 - INFO    - terrariumEngine      - Updated relay denkovi_v2_4 relay named 'T5_UVB_Socket2_Denkovi' at address '2,0001691995' with new value 0.00 in 0.01 seconds.
**2021-07-28 21:34:19,210 - INFO    - terrariumEngine      - Updated relay denkovi_v2_4 relay named 'JungleDawn_Socket3_Denkovi' at address '3,0001691995' with new value 100.00 in 0.01 seconds.**
2021-07-28 21:34:19,196 - INFO    - terrariumEngine      - Updated relay denkovi_v2_4 relay named 'Nothing_Socket1_Denkovi' at address '1,0001691995' with new value 0.00 in 0.01 seconds.
2021-07-28 21:34:19,178 - INFO    - terrariumEngine      - Updated relay denkovi_v2_4 relay named 'Hydor_Socket4_Denkovi' at address '4' with new value 0.00 in 0.03 seconds.
2021-07-28 21:34:19,076 - INFO    - terrariumEngine      - Starting a new update round with 2 sensors, 4 relays, 0 buttons and 0 webcams.
2021-07-28 21:33:54,767 - INFO    - terrariumEngine      - Engine update done in 5.69 seconds. Waiting for 24.31 seconds for the next round.
2021-07-28 21:33:54,542 - INFO    - terrariumEngine      - Updated Enclosure Rainbow with 1 areas in 4.59 seconds.
2021-07-28 21:33:54,534 - INFO    - terrariumArea        - Set the relay T5_UVB_Socket2_Denkovi to 0.0 with a delay of 0.0 minutes
2021-07-28 21:33:54,532 - INFO    - terrariumEngine      - Updated Enclosure Rainbow with 1 areas in 0.02 seconds.
2021-07-28 21:33:54,521 - INFO    - terrariumArea        - Set the relay T5_UVB_Socket2_Denkovi to 0.0 with a delay of 0.0 minutes
**2021-07-28 21:33:54,520 - INFO    - terrariumArea        - Set the relay JungleDawn_Socket3_Denkovi to 0.0 with a delay of 30.0 minutes**
2021-07-28 21:33:54,513 - INFO    - terrariumArea        - Toggle the relays for area Lighting named Main_Lights to state off.
2021-07-28 21:33:54,511 - INFO    - terrariumArea        - Toggle off the relays for area Lighting named Main_Lights.

PhoenixNP, 2021-07-28:
It apparently sets a new value of 100.00 (non-existing value?)
And then the next run it set's it to 0, switching the relay off

theyosh, 2021-07-28:
Please post screenshot of the Area setup form. That makes it a bit easier for me to figure this out

PhoenixNP, 2021-07-29:
[image]

theyosh, 2021-07-29:
I have it now also. Apparently I did not test this... sorry. Working on it.

PhoenixNP, 2021-07-29:
No worries, can't test everything by yourself =)

theyosh, 2021-08-02:
Hi,

I fixed it. It is not yet 100% the way I want, but it is working. The only problem right now, is that the logging is saying that it will toggle the lights on/off even when it is not happening. So the log is a bit confusing at the moment. I will work on that later on.

For now, you can do a `git pull` and then restart `sudo service terrariumpi restart` to load the new code.

No changes need in the config, It should just work after you pulled the latest code.

theyosh, 2021-08-02:
I also fixed the logging part now. So make sure you have done a `git pull` to get the latest code.

PhoenixNP, 2021-08-02:
You are now updating most of the values with "100" which, atleast for Denkovi V2, is not a valid value and should be "1"

2021-08-02 15:00:43,282 - INFO    - terrariumArea        - Set the relay JungleDawn_Socket3_Denkovi to 100.0 with a delay of 0.00 minutes

theyosh, 2021-08-02:
Strange,

changed the logic. So do a `git pull` and restart the TerrariumPI service. https://github.com/theyosh/TerrariumPI/blob/4.x.y.z/hardware/relay/denkovi_relay.py#L91

PhoenixNP, 2021-08-02:
Hmm strange, It still does it;
2021-08-02 15:15:21,064 - INFO    - terrariumArea        - Set the relay JungleDawn_Socket3_Denkovi to 100.0 with a delay of 0.00 minutes

theyosh, 2021-08-02:
Do you actually see the lights go on or off? Or are you only looking in the log? Because 100 just means on. And 0 means off. The logging is showing numbers is due to dimmer support.

So, I think something else is happening... as all relays have the same logic, and I have tested at least 5 different types, and they all work...

Else, re-open the port forwarding? Then I will have a second look.

PhoenixNP, 2021-08-02:
Once i switch it off, it will flash on, back to off. Then wait for the 30second timer and turn on.
But once that happens, it will keep switching on/off a few times and then settle at on.

PhoenixNP, 2021-08-02:
You should have access to the PI using the old email credentials =)

theyosh, 2021-08-02:
ssh is timing out...

PhoenixNP, 2021-08-02:
Can you try again now? =)
Firewall was still set to drop

theyosh, 2021-08-02:
Hi, maybe continue per email? Seems like the Pi is offline? I was able to login once, and then it died....

PhoenixNP, 2021-08-02:
No problem, I've send you an e-mail

PhoenixNP, 2021-08-06:
I will close it, it works fine if you don't manually switch anything =)

closed 2021-08-06 by PhoenixNP