theyosh/TerrariumPI #572: [BUG]: Incorrect Average Temperature

PhoenixNP, 2021-07-28:
**Setup:**
 - Software Version 4
 - Raspberry PI: 3B

**Describe the bug**
The average temperature calculated is "incorrect".
While testing the setup and excluding one of the sensors from the average, it still get's used for the caculation.

**To Reproduce / Installation steps**
[image]

**Expected behavior**
The sensor that has the "Exclude Average" set to be excluded from calculations

theyosh, 2021-07-28:
Strange, I really thought this has worked.

But it is fixed now. Do a `git pull` and restart TerrariumPI. That should fix this.

PhoenixNP, 2021-07-28:
The average is working now, but the "All Sensors" page doesn't show the temperatures anymore.

theyosh, 2021-07-28:
> The average is working now, but the "All Sensors" page doesn't show the temperatures anymore.

It is unclear what you mean with "doesn't show the temperatures anymore"

PhoenixNP, 2021-07-29:
It works right now =)
Apparently it was a matter of waiting for the sensors to update a few times =)

closed 2021-07-29 by PhoenixNP