theyosh/TerrariumPI #573: [BUG]: Can't add new switches

MP98n, 2021-07-31:
I can make changes to the 7 switches I already have setup, but I can't add the 8th GPIO relay on my 8 channel board.

theyosh, 2021-08-02:
Hi,

I need some more information. Are you using TerrariumPI 3 or version 4? If TP3, do you use Python 2 or 3?

And as GPIO is not auto-detected, so you should have been able to add the others, so then I think you are trying to add a GPIO pin that is already used by a different relay?

What error message do you get? Now it is not clear to me what is happening.

MP98n, 2021-08-03:
Apologies, I have managed to solve it by reinstalling Terrarium Pi. I'm not sure what was causing it. I wasn't getting an error message; it was just not updating the switch list after I clicked submit. I'm not able to reproduce it since I have reinstalled. I'm using Terrarium Pi 3.10.0

I also wanted to say thank you for your efforts with Terrarium Pi. It's been a great help for rescuing a smart vivarium with a dead circuit board from a certain Kickstarter.

closed 2021-08-04 by theyosh