theyosh/TerrariumPI #575: [Minor BUG]: No suitable area type for a pump

Dave0408, 2021-08-03:
**Setup:**
 - Python version: 3
 - Raspberry PI: 4

**Describe the bug**
When I try to add my pump as an installation in TP4, I can only do this via the "Lighting" selection. 
If I use the more obvious option "Humidity" and set the switch to timer, the dependency on a sensor remains. I would assume that the display would also have to switch to the selection of a start and end time as well as duration and timeout. Just as is the case with the lighting. 

What is the right area type for a water pump?

**Screenshots**
Light:
[image]

Humidity:
[image]

theyosh, 2021-08-04:
Hi,

hmm, your idea using Humidity is correct. I have the same test:
[image]

I did change the code a bit, so that it should be possible to add a area without sensors if you have not selected the sensor mode. And only the alarm low or high timer fields are required when there are relays selected.

In my screenshot I setup a pump that works between 8:00 and 20:00. Then I used the On/Off duration to specify an operating window of 1 minute per hour. End in that 1 minute, the pump is put on for 25 seconds and settles for 60 seconds. Which is higher then the 1 minute interval window. And therefore, it should pump 12 times during the day.

It does not need the sensors for operating.

In order to make sure you have the latest code do a `git pull` and restart TerrariumPI. The refresh your browser holding the shift key so you will get the latest HTML. Or clear browser cache. Then it should work if I understand your issue correctly.

theyosh, 2021-08-25:
Hi, is the above working for you? Do you have any updates on this?