theyosh/TerrariumPI #579: [FEATURE]: Sonoff D1 Dimmer support?

fabiopbx, 2021-08-06:
**Is your feature request related to a problem? Please describe.**
Any chance we could get support for the Sonoff D1 Dimmer? I Got three of these and trying to get the new enclosure ready to go :)

**Describe the solution you'd like**
I see Tasmota is already supported and these just need a cmd sent to it with `cm?cmnd-Dimmer=<%value>`.
I had a look into the code but haven't had a chance to play around with it, maybe this weekend, but since you know your way around the code, thought it be good to put a request in :)

**Additional context**
On a side note, any plans to support MQTT? since this would open quite a few possibilities :D

theyosh, 2021-08-09:
Hi,

I think I can add the D1 dimmer. But I need some more information. 

- What is the correct url to get the actual dim value of the device?
- The value is the amount to dim. So 100% means the lights are out?

This can then be added to TP4. But I am not able to test, so that means you need to do some testing for me.

About MQTT, I am not sure how people would like to use. I can just push all updates to a topic. But do you want to control the relays through MQTT? That means I have to listen to certain commands. That is something I am not happy with. That makes it a lot more complex.
So using MQTT as a kind of notification system, that I can make. But I need some more info about it. How about opening a discussion and see what other people are saying.

fabiopbx, 2021-08-09:
Hi I just was looking at the code to see how you were deceiving the dimmer values :D
I got a Pi with version 4 code I'm more then happy to test :)

Per Docs (Using Tasmota Firmware)
`Dimmer | 0..100 = set dimmer value from 0 to 100%`

`http://<ip>/cm?cmnd=Dimmer%20100`
> full brightness at 100%

`http://<ip>/cm?cmnd=Dimmer%200`
> device is off

These should also work for simple on/off top
http://<ip>/cm?cmnd=Power%20On
http://<ip>/cm?cmnd=Power%20off

So MQTT is a lets say, msg event system based around topics, sensors can push updates and vice versa can also send control cmds, its quite powerful and versatile would would open a few doors :D but yh we can discuss in a different topic :)

theyosh, 2021-08-09:
Ok, I think I can fix this. I will update later today when I am able to make some changes in code...

theyosh, 2021-08-09:
I have seen the documentation, but I need to know for sure: how to get the current dimmer status?

Does: `http://<ip>/cm?cmnd=Dimmer` give the current dimmer value? And that is the JSON output if you enter that url in your browser.

fabiopbx, 2021-08-09:
It did respond with a json reply indeed, not sure if it’s current status or acknowledgment of the request. 

Ill give you an example in a minute or so.

theyosh, 2021-08-09:
Just updated the code to support the D1 dimmer. Do a `git pull` and `sudo service terrariumpi restart` and you should be able to add a Sonoff D1 Dimmer.

As address enter: http://10.10.0.184/

That should be enough. I hope it works the first time :)

fabiopbx, 2021-08-09:
So far it looks good :)

[image]

Adding the 3rd Dimmer gives me a db integrity error only enough... but then I have few weird errors adding/deleting relays/sensors throwing 500 errors but might just need a fresh db....

Anything you'd like me to test in relation to the dimmers however ?

Only log I've seen so far is that it sometimes fails to get current status (maybe it timeouts or so)

theyosh, 2021-08-09:
Nice!,

no, at the moment I would suggest use them as you would like to use. Create an enclosure and add an light area. There you can set delay and duration for on and off.

I also see some timeouts with my Sonoffs. But that is network related. So I have a ping running to my sonoff device from my Pi. This way the connections stays active and then I see less errors.

theyosh, 2021-08-25:
Hi, is it still working correctly? Can we close this issue?