theyosh/TerrariumPI #580: [BUG]: V4 Can't Access Web GUI

ntfd0, 2021-08-09:
First of all, thank you so much for creating TerrariumPI! It's super cool and was working okay, but then stopped for some reason...

**Setup:**
 - Python version: 3
 - Raspberry PI: 4

**Describe the bug**
Can't access the web GUI. It was wroking fine, then I turned off the PI and turned it back on a few hours later - can't access anymore (This site can’t be reached. 127.0.0.1 refused to connect.)
Looked at the logs:
```
2021-08-09 23:39:38,329 - INFO    - terrariumEngine      - Starting TerrariumPI 4.0.0 ...
2021-08-09 23:39:38,393 - INFO    - terrariumEngine      - Loaded 28 settings in 0.06 seconds.
2021-08-09 23:39:38,394 - INFO    - terrariumEngine      - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=shinagawa,jp&appid=*********
2021-08-09 23:39:39,253 - INFO    - weather              - Loaded new weather data in 0.856 seconds.
2021-08-09 23:39:39,717 - INFO    - terrariumEngine      - Loading existing sensors from database.
2021-08-09 23:39:40,610 - INFO    - terrariumEngine      - Loaded sensor 1wire temperature named 'temp1' at address '28-0117151e64ff' with value 26.81C in 0.89 seconds.
2021-08-09 23:39:41,478 - INFO    - terrariumEngine      - Loaded sensor 1wire temperature named 'temp2' at address '28-aacb151e64ff' with value 24.38C in 0.87 seconds.
2021-08-09 23:39:42,356 - INFO    - terrariumEngine      - Loaded sensor 1wire temperature named 'temp3' at address '28-de2f0c1e64ff' with value 23.25C in 0.88 seconds.
2021-08-09 23:39:42,358 - INFO    - terrariumEngine      - Scanning for new sensors ...
2021-08-09 23:39:51,242 - INFO    - terrariumEngine      - Loaded 3 sensors in 11.52 seconds.
2021-08-09 23:39:51,243 - INFO    - terrariumEngine      - Loading existing relays from database.
2021-08-09 23:39:51,555 - INFO    - terrariumEngine      - Loaded relay tplinkkasa relay named 'Channel 10 Gallon Lamp' at address '192.168.100.54' value 0.00 in 0.29 seconds.
2021-08-09 23:39:51,606 - INFO    - terrariumEngine      - Loaded relay tplinkkasa relay named 'Channel Pantheon Lamp' at address '192.168.100.60' value 0.00 in 0.05 seconds.
2021-08-09 23:39:51,607 - INFO    - terrariumEngine      - Scanning for new relays ...
2021-08-09 23:40:07,325 - INFO    - terrariumEngine      - Loaded 2 relays in 16.08 seconds.
2021-08-09 23:40:07,326 - INFO    - terrariumEngine      - Loading existing buttons from database.
2021-08-09 23:40:07,336 - INFO    - terrariumEngine      - Loaded 0 buttons in 0.01 seconds.
2021-08-09 23:40:07,337 - INFO    - terrariumEngine      - Loading existing webcams from database.
2021-08-09 23:40:12,793 - WARNING - retry.api            - cannot identify image file <_io.BytesIO object at 0xa99cbf30>, retrying in 0.5 seconds...
2021-08-09 23:40:13,607 - WARNING - retry.api            - cannot identify image file <_io.BytesIO object at 0xa997d450>, retrying in 0.5 seconds...
```
The only issue I'm seeing are the two lines at the bottom. Any idea what this is and if it's causing the issue?
Cheers!

theyosh, 2021-08-09:
It is somehow hanging on loading the webcam. What kind of webcam are you using?

And it should start after a while as I see 2 times a retry.api call. So that means it should try 1 time more, and then continue. But it could take some time. This is not happening?

PS: You are using TP Link Kasa relays.... those are also untested... Not sure if that will have any effect on this issue

ntfd0, 2021-08-10:
Thanks a lot, you were right - it seems like the webcam might have been causing this issue.
Just for future debugging purposes - I think if you add a camera, add it to an enclosure, then pull out the cable from raspi, then reboot, you might get the issue above.
And thanks for the headsup on TP Link Kasas! They are working okay so far. I have to say that I haven't tried any automation with them yet, but at least the manual controls are working as intended.
On the topic of camera, the reason I disconnected it is that I was not getting an image, or rather I was getting a black screen no matter what I did (tried both the normal camera and the live camera options). It's a standard Raspberry Pi Camera V2. I also tested it using raspistill -o testshot.jpg, and got a normal photo without any issues... (let me know if I should create a separate issue for this).
I also noticed a couple of bugs here and there, will post in a moment.
Sorry for the wall of text, just really excited to get this working!

theyosh, 2021-08-10:
That is strange. I cannot pull the cable of a Raspberry Pi Camera. That is locked.

But I use the same raspistill program to take the pictures. So that is strange... Maybe you had set a white balance setting that is causing this..
I will test what happens when there is no camera attached and trying to take a picture. That should not lock the system, only give an error message.

About the TP Link relays. If they toggle when you click on them I guess they work correctly. The only test you could do, is that you toggle the relay with an app. So that you can check if my software is able to update the state without being toggled in the interface.

So, lets do the following. I test/debug the camera in this issue. I will only fix the startup issue. Not the black image... If that is still the case after my fix, we look further.
If you have problems with the TP Link Kasa relays, open a new issue.

And thanks for the Coffee!! :)

ntfd0, 2021-08-11:
I feel very dumb now... You were right, the black image from the camera was because the white balance setting was off. Apologies! It's now working as intended.
I also tried adding a light schedule for two of my enclosures using the TP Link relays, and they worked great! Happy to report that the Kasa integration seems to be doing okay.
As for the camera bug, I'm not sure if it's a high priority issue... Might be just a one-off thing. On my end, I will try my best to replicate again, and if it happens again, I'll let you know. It's working fine now with the camera connected, so I'm not too worried.

theyosh, 2021-08-25:
Hi, I think I leave it as is. I see this like a hardware issue, and this one is not happening that often..

Ok on closing this issue?

ntfd0, 2021-08-25:
Hi, should be okay to close! Haven't had any issues whatsoever since.

closed 2021-08-25 by ntfd0