theyosh/TerrariumPI #581: [BUG]: V4 Unable to see the full list of sensors if all of them are auto-detected

ntfd0, 2021-08-10:
**Setup:**
 - Python version: 3
 - Raspberry PI: 4b
 - Sensors: DS18B20 x 3 (GPIO 4, 1-Wire)

**Describe the bug**
When you click Sensors, only New sensor and Scan sensors show up, even though the sensors themselves are detected correctly and can be used when setting up Enclosures.
The only way to get to them is to create a fake sensor (say, add a non-existing AM2302) - then the options for Temperature and Humidity will show up, from where you can finally edit these sensors.

**To Reproduce / Installation steps**
Clean installation of TerrariumPI, connect just the DS18B20s.

**Expected behavior**
Would be great to have either the Temperature option to show up if these are automatically detected, or have access to a full list of all sensors at all times (like with Enclosures)

**Screenshots**
[image]

theyosh, 2021-08-10:
Hi,

this is strange, as the list of sensor types is shown when you are connected. You see the round green check mark on right top, and you had a green notification that you are connected.
When that happens, the list of available sensor types should be filled in the menu.

What happens if you reload the page? Is it than still an issue? And you have now only 1 sensor type added, temperature. Maybe that is a bug... I had always multiple type of sensors..

theyosh, 2021-08-11:
Hmm, I think I found the issue. So also, if you do a `git pull` and reload the web interface in your browser, it should work again.

Make sure you have cleared your browser cache.

ntfd0, 2021-08-11:
Working now, thanks a lot for a prompt response!
[image]

closed 2021-08-11 by ntfd0