theyosh/TerrariumPI #582: [BUG]: Area settings have the wrong enclosure selected by default

ntfd0, 2021-08-10:
**Setup:**
 - Python version: 3
 - Raspberry PI: 4b
 - 3 animal enclosures

**Describe the bug**
When adding a new Area via the Enclosure List (Enclosures>List>click wrench icon>Add area), the Enclosure selected by default is the first Enclosure in the system instead of the one you wanted to add an Area for.

**To Reproduce / Installation steps**
Example:
1) Set up three Enclosures: Enclosure 1, Enclosure 2, Enclosure 3.
2) Select Enclosure 3 and click Add area
3) The Enclosure selected by default in the Area settings is Enclosure 1, not Enclosure 3

**Expected behavior**
The Enclosure selected by default in the Area settings is the one that the user clicked on, not Enclosure 1.

**Screenshots**
[image]

Granted, this is a very minor bug, but it's a tad annoying if I'm honest (especially during the initial setup)

theyosh, 2021-08-10:
Hi,

that is indeed annoying.. I will look into it. Hoop to fix it in a couple of days.

ntfd0, 2021-08-11:
Wow, thanks for such a quick fix! All good now.

closed 2021-08-11 by ntfd0