theyosh/TerrariumPI #583: I've 3.9.9 version. Can I Update it?

vikrilow, 2021-08-10:
Hello Bro,

Actually I'm installed 3.9.9 version and I'd like update it.
I don't know that is better, 4.0 or 3.10.

Any case, What version update  is better? How can I update my system?

I'm sorry for this question, but I'm not controle this system a lot. hehe

Thanks in advanced

theyosh, 2021-08-11:
Hi,

for now, I still advise just to upgrade to 3.10 . As there is no update path to version 4. That is a migration way. Because TP4 needs a different install.

So for now, just follow https://github.com/theyosh/TerrariumPI/releases/tag/3.10.0 and I hope to release a first version of TP4 soon. And then I can explain how to migrate. It is not that hard to do, but I need to make sure it is clear for everybody.

theyosh, 2021-08-25:
Hi, can we close this issue?

As TP3 is still working, and if you want to go to TP4, there is some documentation about that: https://theyosh.github.io/TerrariumPI/install/#migration-from-v3-to-v4

And this can be done when you want.