theyosh/TerrariumPI #585: Possible problem with environment configuration

vikrilow, 2021-08-12:
Hello.
I'm try my new terrarium with terrarium PI and I've experiment some performance that I don't like so much.
I've configure this:
[image]
 And in dashboard show this:
[image]

Isn't first time that I see it, about every 5 or 6 hours sho it.

Then, in the register, show these messages:

[image]

Does anyone know what happens to my terrariumPI?

Thanks!!

vikrilow, 2021-08-12:
Today, I continue with more problems. For Example:
If my maxime temperature is 32ºC, why with 39ºC don't active BIG_FANs...?
[image]

[image]

Really, I don't Understand why temperature is shutdown with 43ºC when I've configured 32ºC maximun.
[image]

Today, I've restarted 4 times Raspberry and continue with this errors:
_2021-08-12 19:28:18,963 - ERROR   - terrariumEnvironment - Environment humidity sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2021-08-12 19:28:18,962 - INFO    - terrariumEnvironment - Environment temperature alarm max is triggered to state True, but the settle time has not passed: 300.0 seconds._

PLease @theyosh can you help me again? My tortoise can home in 2 weeks and I'm a bit worried.

Thanks!!

theyosh, 2021-08-12:
Euh,

I guess there is a lot wrong with your hardware.. as the sensors do not give a value every 30 seconds. So it looks like there is problem with the Pi and it hardware.

Or could it be that the temperatures get outside the min and max values? That count also as bad hardware.

What kind of PI do you have?
Did you install it using python 2 or python 3?

Could you upload the complete logfile here

vikrilow, 2021-08-12:
Ey, thanks for your reply.
Really, I don´t understand why there is problem with my hardware (sensors)...

I've Raspberry PI4 Model B, Rev 1.2 and I use 2.7 version Python. Maybe, I'll have update it?

I've downloaded log file. You can download it in next link:

https://mega.nz/file/540GALbD#IXB6jKsRj_RvfnuDvB_caJ_CH6R_7DpQ-LBu9bOktQM

thanks!!

theyosh, 2021-08-13:
Hmm,

a few things need to be checked.

- So first, if this is a new install, please restart and use Python3 to install. That works a bit better.
- The limit min and max values for humidity are not wide enough. You have sometimes values below 30, and that gives errors, as you have specified that limit at a sensor. For humidity use limit min like 5 or 10 and max 100. To make sure all measurements are valid.
- Also change the temperature sensors limit min/max so that all measurements are valid.
- And based on the images, it looks like there were some issues with the temperature sensors. Therefore the environment cannot act, and will power off. As it has no information from the sensors.
- Remove the 1800 and 300 from the cooling environment part. And change them to 0. So that the fan should always go on an stay on when it is to hot.  As you have a settle time of 5 minutes (300).

That should solve at least a part. The environment will not work when you have the exclamation marks in the environment dashboard. Then the senors are somehow 'down' for 5 measurements. Wrong min/max limits can cause this.

Also, you have a temperature drop during the night. That means you should not change the sensor configs during the night. That will result in wrong settings due to the temporary temperature drop of -10. This is known bug in TP3.

So, make you changes during the day, and just to be sure, restart after done changing the settings, so we are sure the new setup is loaded correctly.

And again, if this is a re-install, please start over with a clean SD card and install it using Python 3. That will save you in the future.

vikrilow, 2021-08-13:
Thanks @theyosh for your help!!
I've changed points 2,3,4 and 5. Now I'm seeing that don't show these messages at the moment.

My System was installed last feb 2021, but If you think that I need to reinstall with Python 3, I'll do it.

PLease, can you explain me how I must do it? Do you think that I need save important files before re-install or I'd have install as new installation completily?

Thanks !!!!!!

vikrilow, 2021-08-18:
Hi @theyosh can you send me. More explain, that how can I reinstall with Python 3?
Thanks

vikrilow, 2021-08-19:
Lo encontré. Este fin de semana, probaré

theyosh, 2021-08-20:
Hi,

yeah, sorry, I did not post the upgrade url. But the https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-python-35 is the correct url where I describe how to upgrade to Python 3.

Make sure you change `/etc/rc.local` file! Else you will still running Python2 version.

vikrilow, 2021-08-23:
Hi again, 

I'm trying to update to pyhton3 but I think I've problem with that.

I'm seeing that I've installed both version, 2.7 and 3.7.
[image]

And If I run command python --version default version is 2.7

[image]

However, I've seeing that rc.local is configured correctly ( I think so)

[image]

PLease, can you help me with that? MY small tortoise can to home next week and I'd like to have system correctly.

Thanks!!

theyosh, 2021-08-23:
Hi,

everything looks ok. The '3' in the rc.local is making sure your are running Python3 with my software. So when you reboot, it should work with Python3

So what is now the error?

vikrilow, 2021-08-23:
OK, 
When I run command python --version, and show it:

[image]
I thought default version was 2.7. 
If you told me that important is configuration in rc.local, there isn't problem.

thanks!

theyosh, 2021-08-23:
Yeah, Raspberry default is still Python 2.7. But as you have installed my software with Python3 (adding '3' to the install.sh), you will install everything for Python3. But I do not change the default python version, as there are more software packages depending on Python.

So by adding the `3` in the /etc/rc.local, you force my software to use Python3. And I make sure that my software uses Python3.

So, I think it should be right now. Have you tried a reboot yet?

vikrilow, 2021-08-24:
Hello,
Yes. I've rebooted some times.
At the moment, all right 

One Question more, for your experience.
How I must configure Humidity environment? 
[image]
0 and 0 like with temperature or you think that I must configure other parameters?

Thanks!

vikrilow, 2021-08-24:
Thanks!
I'm using Humidifier similar to this:
https://es.aliexpress.com/item/1005001644347785.html?src=google&albch=shopping&acnt=439-079-4345&slnk=&plac=&mtctp=&albbt=Google_7_shopping&gclsrc=aw.ds&albagn=888888&ds_e_adid=475881929441&ds_e_matchtype=&ds_e_device=c&ds_e_network=u&ds_e_product_group_id=743612850954&ds_e_product_id=es1005001644347785&ds_e_product_merchant_id=107769768&ds_e_product_country=ES&ds_e_product_language=es&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&ds_dest_url=https%3A%2F%2Fs.click.aliexpress.com%2Fdeep_link.htm%3Faff_short_key%3DUneMJZVf&albcp=11492995081&albag=117794251848&isSmbAutoCall=false&needSmbHouyi=false&gclid=Cj0KCQjwsZKJBhC0ARIsAJ96n3UdBL5wbs3N7XN4h3cqWqBUILsDDw1IAt62Tv8iNr39jqL1lXouYhcaAsjnEALw_wcB&aff_fcid=18758e8519454f489151290d733b327e-1629816005946-03444-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=18758e8519454f489151290d733b327e-1629816005946-03444-UneMJZVf&terminal_id=dc919218e5db41ddb50ba00836a61b22

Now I've configured with 0 and 0 and I think terrarium has tense atmosphere, so I'll try 20 and 120 sec

Thanks

theyosh, 2021-08-24:
That is a mister, which will raise the humidity slowly. So you can just use two times a '0'. That way, the mister will go on when the humidity is to low, and turn off, when the humidity is higher then the high alarm.

As long as the value is between the min and max alarm value, nothing will change. So use the sensor alarm values to tune this.

vikrilow, 2021-08-24:
ONE MILLION THANKS!!!

vikrilow, 2021-09-02:
Hi,
System is working without problem, but I'd like to take your message to answer you one possible problem with my humidifier. 

Really, I'm seeing that humidify parameters aren't continnuos. Normally up and down constantly. 
[image]
I don't know if these graphics are normals or not.

Maybe, as I've some fans that runs when temperature up and others runs 5 minutes every two hours, these fans help to down humidity, right? 

I send you some register from my system:
2021-09-02 22:54:49,744 - INFO    - terrariumEnvironment - Update done in 0.21320 seconds. Waiting for 14.78680 seconds for next update
2021-09-02 22:54:49,642 - INFO    - terrariumSwitch      - Changed power switch 'BIG_FANs' of type 'gpio' at address '29' from state 'True' to state 'False' (Forced:False)
2021-09-02 22:54:49,642 - INFO    - terrariumEnvironment - Environment humidity is turning off the alarm max powerswitches based on sensor
2021-09-02 22:54:49,533 - INFO    - terrariumSwitch      - Changed power switch 'BIG_FANs' of type 'gpio' at address '29' from state 'False' to state 'True' (Forced:False)
2021-09-02 22:54:49,532 - INFO    - terrariumEnvironment - Environment temperature is turning on the alarm max powerswitches based on sensor
2021-09-02 22:54:49,528 - INFO    - terrariumEngine      - Update done in 3.40070 seconds. Waiting for 26.59930 seconds for next update
2021-09-02 22:54:49,282 - INFO    - terrariumSwitch      - Updated gpio power switch 'BIG_FANs' status in 0.00003 seconds
2021-09-02 22:54:49,181 - INFO    - terrariumSwitch      - Updated gpio power switch 'FRONTAL_FANs' status in 0.00150 seconds
2021-09-02 22:54:49,079 - INFO    - terrariumSwitch      - Updated gpio power switch 'HUMIFICADOR_N4' status in 0.00003 seconds
2021-09-02 22:54:48,979 - INFO    - terrariumSwitch      - Updated gpio power switch 'RESERVA' status in 0.00002 seconds
2021-09-02 22:54:48,878 - INFO    - terrariumSwitch      - Updated gpio power switch 'LUZ-HID_N1' status in 0.00003 seconds
2021-09-02 22:54:48,777 - INFO    - terrariumSwitch      - Updated gpio power switch 'CERÁMICA_N2' status in 0.00004 seconds
2021-09-02 22:54:48,665 - WARNING - terrariumSensor      - Measured humidity value 65.46859851093275% from bme280 sensor 'HUMEDAD_COOL_ZONE' is erratic compared to previous value 54.10755968141866% in 0.23201 seconds.
2021-09-02 22:54:48,321 - WARNING - terrariumSensor      - Measured humidity value 73.20808437947689% from bme280 sensor 'HUMEDAD_HID_ZONE' is erratic compared to previous value 61.77998958159636% in 0.23153 seconds.
2021-09-02 22:54:47,991 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2021-09-02 22:54:47,990 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2021-09-02 22:54:47,980 - INFO    - terrariumSensor      - Updated w1 sensor 'TEMP_HOT_HID_ZONE' temperature from 26.75C to 26.75C in 0.84976 seconds
2021-09-02 22:54:47,020 - INFO    - terrariumSensor      - Updated w1 sensor 'TEMP_COOL_ZONE' temperature from 26.25C to 26.19C in 0.89237 seconds
2021-09-02 22:54:34,752 - INFO    - terrariumEnvironment - Update done in 0.24149 seconds. Waiting for 14.75851 seconds for next update
2021-09-02 22:54:34,650 - INFO    - terrariumSwitch      - Changed power switch 'BIG_FANs' of type 'gpio' at address '29' from state 'True' to state 'False' (Forced:False)
2021-09-02 22:54:34,650 - INFO    - terrariumEnvironment - Environment humidity is turning off the alarm max powerswitches based on sensor
2021-09-02 22:54:34,514 - INFO    - terrariumSwitch      - Changed power switch 'BIG_FANs' of type 'gpio' at address '29' from state 'False' to state 'True' (Forced:False)
2021-09-02 22:54:34,513 - INFO    - terrariumEnvironment - Environment temperature is turning on the alarm max powerswitches based on sensor

Thanks!!

theyosh, 2021-09-06:
Hi,

not sure, but I see the message:

```
2021-09-02 22:54:48,665 - WARNING - terrariumSensor - Measured humidity value 65.46859851093275% from bme280 sensor 'HUMEDAD_COOL_ZONE' is erratic compared to previous value 54.10755968141866% in 0.23201 seconds.
2021-09-02 22:54:48,321 - WARNING - terrariumSensor - Measured humidity value 73.20808437947689% from bme280 sensor 'HUMEDAD_HID_ZONE' is erratic compared to previous value 61.77998958159636% in 0.23153 seconds.
```

These lines are saying that the humidity changed more then the max difference you have selected. This means that it takes 5 measurements to get the new actual value.

So, I would change the values for the field Maximum difference (not sure what the exact name is, as you have a different language then English) Change this value to '0' so that it will be ignored, and every measurement is then valid.

That could smooth the graph a bit more.

vikrilow, 2021-09-07:
OK, I'll test it to see if they improve.
Now, Maximum difference configured is 1 (really I don't know why I put it hehe)

I'll change 1 for 0.
Thanks again!

theyosh, 2021-09-08:
And does it now look better? It is almost 24 hours later, so the graphs should show a difference I think.

vikrilow, 2021-09-09:
Maybe the problem is for humidify sensors, but I thinks my humidify sensors are good, because they are BME280.
In addition, I've configure when there is high humidify nothings, so that, don't run fans. If there is, for example 95% humidify, Will down slowly.

However, graphics are similar:

[image]

As you know, I've to review temperature cable sensors, so I've added two Mi bluetooth sensor and humidity its diferent:

[image]

Maybe reading BME280 is more exact than Mi Bluetooth  sensor.

vikrilow, 2021-09-10:
@theyosh I send you yesterday logfile.

https://mega.nz/folder/Rp9wxTwA#rMRhhLAk0u08NCSdl9vgCg

Although I've configured Maximum difference to 0, I continue with these message type:

_2021-09-09 20:21:01,317 - WARNING - terrariumSensor      - After 5 erratic humidity measurements is the current value 72.19380543417532% is promoted to a valid value for bme280 sensor 'HUMEDAD_HID_ZONE' in 0.23419 seconds.
2021-09-09 20:21:01,661 - WARNING - terrariumSensor      - Measured humidity value 52.430003874861555% from bme280 sensor 'HUMEDAD_COOL_ZONE' is erratic compared to previous value 54.84104281625369% in 0.23175 seconds.

I think I'll reboot system, maybe there is anything in cache giving problems...

vikrilow, 2021-09-10:
Today I've more warning message that I don't know really if there is a real problem.

_2021-09-10 16:03:21,391 - WARNING - terrariumSensor      - Measured humidity value 75.0761029773693% from bme280 sensor 'HUMEDAD_COOL_ZONE' is erratic compared to previous value 75.0761029773693% in 0.00012 seconds.
2021-09-10 16:03:21,279 - WARNING - terrariumSensor      - Measured humidity value 100.0% from bme280 sensor 'HUMEDAD_HID_ZONE' is erratic compared to previous value 100.0% in 0.00020 seconds.
2021-09-10 16:03:21,165 - WARNING - terrariumSensor      - Measured humidity value 88.9% from mitemp sensor 'HUM_HOT_BLUE' is erratic compared to previous value 88.7% in 3.56273 seconds.
2021-09-10 16:03:17,477 - WARNING - terrariumSensor      - Measured temperature value 30.1C from mitemp sensor 'TEMP_HOT_BLUE' is erratic compared to previous value 30.1C in 0.00013 seconds.
2021-09-10 16:03:17,358 - WARNING - terrariumSensor      - Measured humidity value 97.8% from mitemp sensor 'HUM_COOL_BLUE' is erratic compared to previous value 97.7% in 4.23908 seconds.
2021-09-10 16:03:13,056 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2021-09-10 16:03:13,008 - WARNING - terrariumSensor      - Measured temperature value 29.5C from mitemp sensor 'TEMP_COOL_BLUE' is erratic compared to previous value 29.5C in 0.00012 seconds.
2021-09-10 16:03:12,896 - WARNING - terrariumSensor      - Measured temperature value 29.25C from w1 sensor 'TEMP_HOT_ZONE' is erratic compared to previous value 29.25C in 0.84796 seconds.
2021-09-10 16:03:11,935 - WARNING - terrariumSensor      - Measured temperature value 30.312C from w1 sensor 'TEMP_COOL_ZONE' is erratic compared to previous value 30.312C in 0.84343 seconds._

My current configuration:
[image]
[image]
[image]

Actually I've next sensor:

2 x BME280 (1 in cool zone and 1 in hot zone)
2 x DS18B20 (1 in cool zone and 1 in hot zone)
2 x Mi Temperature and humidity Bluetooth ( 1 in cool zone and 1 in hot zone. Added recently for problem with DS18B20 cable sensors)

However, In environment temperature and humidity I only use TEMP_HOT_BLUE (MI Temperature) and HUMEDAD_HID_ZONE and HUMEDAD_HID_BLUE.
[image]
[image]

I'm very sorry to send you so many questions

theyosh, 2021-09-12:
Hmm, I guess I made a mistake. The max diff should be high for each sensor to de-activate it. In TP4 the value 0 does disable it, but in TP3, you just need to set a high number. So enter the value of 99 so that every measurement is valid.

Sorry for the mistake. I am running TP4 for 6 months now, and working on it for more than a year. So I can mix up some logic between TP3 and TP4.

theyosh, 2021-09-21:
Hi, is it getting better? And sorry for the mistake in the explanation.

vikrilow, 2021-10-16:
> ¿Alguna actualización? ¿Te funciona?

Hello, I'm sorry for delay to reply you.
At the moment, everything is running correctly.

Thanks very much for your time!!!