theyosh/TerrariumPI #587: [BUG]: All links under Help lead to non-existent pages

ntfd0, 2021-08-14:
**Setup:**
- Python version: 3
- Raspberry PI: 4b

**Describe the bug**
All links under the "Help" section in About lead to non-existent and unrelated pages (mailbox.html, mailbox-compose.html, mailbox-read-mail.html)

**Expected behavior**
Change links to lead to the actual About, Usage, and Hardware pages

**Screenshots**
[image]

Not sure if it's just a feature that hasn't been implemented yet or a bug? If it hasn't been implemented, perhaps better to lead to an "under construction" page or smth?

theyosh, 2021-08-16:
Hi,

Those are general info pages, which are not done yet. Also I am thinking of changing those pages, as a lot of info is now online. And I do not like to write manuals and howtos multiple times.

Therefore I have reduced the menu to only an about page, where some simple default information is shown. I think this is enough for now.

theyosh, 2021-08-25:
Updated the menu and updated the About page with information about setup and installation to go to: https://theyosh.github.io/TerrariumPI/

closed 2021-08-25 by theyosh