theyosh/TerrariumPI #588: [BUG]: Can't add Telegram or Pushover notifications

ntfd0, 2021-08-14:
**Setup:**
- Python version: 3
- Raspberry PI: 4b

**Describe the bug**
Trying to add Telegram notifications and/or Pushover notifications, both 
**To Reproduce / Installation steps**
Please provide the steps to reproduce or how you have installed the software.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]
[image]
[image]

Am I doing something wrong here...?

theyosh, 2021-08-16:
Hi,

Nope, this is not working. As I am struggling a bit with the amount of code to refactor, and as I do not use any social media anymore or other services like pusher or telegram, I am afraid that this will not be restored.

So, I am thinking of changing this, so that it all works with MQTT, where you then can at least create some own clients that push the info to Pusher or Telegram. So I make a more general notification system, and then it is for the user to integrate with Telegram of other systems.

Also, I am currently thinking of supporting Matrix somehow, so that you can 'chat' with TerrariumPI. And as Matrix also support all kind of bridges between Matrix and Telegram, Whatsapp, etc... It makes it easier for me to finish the notification part.

So for now, the notification part is not working. If you really need it, stick at version 3 for now.

theyosh, 2021-08-25:
I added more information here: #597

closed 2021-08-25 by theyosh