theyosh/TerrariumPI #589: Question about function in CSS811_RPi.py

wolli1234, 2021-08-17:
Hello Yosh,

I doubt my CO2 readings from the CSS811. 

I looked into CSS811_RPi.py:
If I understood correctly the default values of 25°C and 50% relative humidity are used to determine CO2. Can I change these values using this function?
```
def setCompensation(self, temperature, humidity):
                temperature = round(temperature,2)
                humidity = round(humidity,2)
                print('Setting compensation to {} C and {} %'.format(temperature, humidity))
                hum1 = int(humidity//0.5)
                hum2 = int(humidity*512-hum1*256)

                temperature = temperature+25
                temp1 = int(temperature//0.5)
                temp2 = int(temperature*512-temp1*256)

                s = [CSS811_ENV_DATA,hum1,hum2,temp1,temp2,0x00]
                s2 = bytearray( s )
                CCS811_fw.write( s2 )
                
                return
```

I guess I would enter floats into the function e.g. "setCompensation(20, 80)" to change the values to 20°C and 80% humidity. Is that correct? I tried to call this function in the terminal, but I don't know how and where exactly I would do that. Can you help me with this?

Best regards,
Wolli

theyosh, 2021-08-20:
Hi,

you could do that, but that means, with an update, you will lose that again. As TP3 is at the end with update, you could do a `git pull` so you have the latest code. And then change those values. But I do not know how to add this to my code. As I have no TP3 test environment anymore.

So, if this is a big issue, please try TP4. There you can specify the sensors for compensation. So that the values will be based on your actual setup/environment.

You could try to add code at the line: https://github.com/theyosh/TerrariumPI/blob/master/terrariumI2CSensor.py#L620
```
self.__device.setCompensation([TEMP],[HUMIDITY])
```
So it would become:
```python
      if(error):
        print('ERROR:{}'.format(self.__device.checkError(statusbyte)))

      self.__device.setCompensation([TEMP],[HUMIDITY])

      if not self.__device.checkDataReady(statusbyte):
        print('No new samples are ready')
```

Take care about the space indentation!

wolli1234, 2021-08-21:
Hi Yosh,

thanks for your reply, I didn't realize TP4 is able to do that already, since last time I used it it wasn't stable enough for my testings. I'll switch to TP4 then and give it a try!

Adding the code you gave in your reply you mean in TP4 or TP3?

theyosh, 2021-08-21:
TP4 is not official released yet, but the code is stable enough to use it. Only the notification part is not working.

And, as it looks like, you are the only one with this CO2 sensor. So it could be that it was some bugs, but that should be fixed fast.

The code above is for TP3 only. As TP4 does it based on the selected sensors.

So, if you want to switch to TP4, the easiest way is:
- Stop TP3 https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-stop-terrariumpi
- Move `/home/pi/TerrariumPI` to `/home/pi/TerrariumPI.old`
- Install TP4: https://github.com/theyosh/TerrariumPI/tree/4.x.y.z#installation
- After a reboot, a clean TP4 should be running.

If you want to go back to TP3, it is a bit hard, but can be done as long you have the backup. Also this backup can be used to migrate relay history, to keep the total power and water usage in TP4.

wolli1234, 2021-09-05:
Hi Yosh,

I installed TP4 now. Thank you for putting the effort in this new version!

Unfortunately the CCS811 sensor gives an error message. 

> 2021-09-05 16:03:31,606 - ERROR   - hardware.sensor      - Error updating sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1'. Check your hardware! Invalid reading from sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1'

I did not use the compensation yet. The sensor itself should be working fine. It is recognized with the address 5a if I type "i2cdetect -y 1" in the console.

Do you have any idea how to approach this bug? Is there more information I can provide to you?

theyosh, 2021-09-06:
Hi,

I added some debug code to see what is going on. Please follow the steps below to get the latest code and how to run.

1. `cd /home/pi/TerrariumPI`
2. Stop existing terrariumpi `sudo service terrariumpi stop`
3. Update to the latest code: `git pull`
4. Enable python environment: `source venv/bin/activate`
5. Manually start TerrariumPI: `python terrariumPI.py`

And now we wait until you see some messages about `CSS811 DEBUG` And then you can press `Ctrl+C` to stop and copy the debug output here. Then I can see what is happening.

wolli1234, 2021-09-06:
Here it is:

> 2021-09-06 15:48:48,729 - INFO    - terrariumEngine      - Starting TerrariumPI 4.0.0 ...
2021-09-06 15:48:48,817 - INFO    - terrariumEngine      - Loaded 28 settings in 0.09 seconds.
2021-09-06 15:48:49,789 - INFO    - terrariumEngine      - Loading existing sensors from database.
CSS811 DEBUG: Device loaded
<hardware.sensor.ccs811_sensor.CCS811_RPi object at 0x751f07f0>
CSS811 DEBUG: Loaded calibration
{'temperature': 20, 'humidity': 50}
CSS811 DEBUG: load compensation
CSS811 DEBUG: DONE!
CSS811 DEBUG: Load read status
CSS811 DEBUG: 
152
CSS811 DEBUG: error? False
CSS811 DEBUG: result
{'eCO2': 0, 'TVOC': 0, 'status': 152, 'errorid': 0, 'raw': 0}
CSS811 DEBUG: Exception
name 'sensor_data' is not defined
CSS811 DEBUG: final data
None
2021-09-06 15:48:50,073 - WARNING - hardware.sensor      - Invalid reading from sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1', retrying in 0.5 seconds...
CSS811 DEBUG: load compensation
CSS811 DEBUG: DONE!
CSS811 DEBUG: Load read status
CSS811 DEBUG: 
144
CSS811 DEBUG: error? False
CSS811 DEBUG: Data is not ready?
2021-09-06 15:48:50,673 - WARNING - hardware.sensor      - Invalid reading from sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1', retrying in 0.5 seconds...
CSS811 DEBUG: load compensation
CSS811 DEBUG: DONE!
CSS811 DEBUG: Load read status
CSS811 DEBUG: 
144
CSS811 DEBUG: error? False
CSS811 DEBUG: Data is not ready?
2021-09-06 15:48:51,274 - ERROR   - hardware.sensor      - Error updating sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1'. Check your hardware! Invalid reading from sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1'
2021-09-06 15:48:51,276 - WARNING - terrariumEngine      - css811 co2 named 'CO2 CCS811' at address '5a, 1' had problems reading a new value during startup in 1.48 seconds. Will be updated in the next round.
2021-09-06 15:48:51,278 - INFO    - terrariumEngine      - Scanning for new sensors ...
2021-09-06 15:48:57,669 - INFO    - terrariumEngine      - Loaded 1 sensors in 7.88 seconds.
2021-09-06 15:48:57,671 - INFO    - terrariumEngine      - Loading existing relays from database.
2021-09-06 15:48:58,191 - INFO    - terrariumEngine      - Loaded relay eg-pm-usb relay named 'Energenie USB device nr: 1(01:01:51:44:b7), Socket: 1' at address '1,01:01:51:44:b7' value 0.00 in 0.49 seconds.
2021-09-06 15:48:58,333 - INFO    - terrariumEngine      - Loaded relay eg-pm-usb relay named 'Energenie USB device nr: 1(01:01:51:44:b7), Socket: 2' at address '2,01:01:51:44:b7' value 0.00 in 0.14 seconds.
2021-09-06 15:48:58,515 - INFO    - terrariumEngine      - Loaded relay eg-pm-usb relay named 'Energenie USB device nr: 1(01:01:51:44:b7), Socket: 3' at address '3,01:01:51:44:b7' value 100.00 in 0.18 seconds.
2021-09-06 15:48:58,655 - INFO    - terrariumEngine      - Loaded relay eg-pm-usb relay named 'Energenie USB device nr: 1(01:01:51:44:b7), Socket: 4' at address '4,01:01:51:44:b7' value 0.00 in 0.14 seconds.
2021-09-06 15:48:58,658 - INFO    - terrariumEngine      - Scanning for new relays ...
2021-09-06 15:49:08,781 - INFO    - terrariumEngine      - Loaded 4 relays in 11.11 seconds.
2021-09-06 15:49:08,782 - INFO    - terrariumEngine      - Loading existing buttons from database.
2021-09-06 15:49:08,795 - INFO    - terrariumEngine      - Loaded 0 buttons in 0.01 seconds.
2021-09-06 15:49:08,797 - INFO    - terrariumEngine      - Loading existing webcams from database.
2021-09-06 15:49:08,808 - INFO    - terrariumEngine      - Loaded 0 webcams in 0.01 seconds.
2021-09-06 15:49:08,809 - INFO    - terrariumEngine      - Loading existing enclosures from database.
2021-09-06 15:49:08,818 - INFO    - terrariumEngine      - Loaded 0 enclosures in 0.01 seconds.
2021-09-06 15:49:08,819 - INFO    - terrariumEngine      - TerrariumPI is up and running at address: http://0.0.0.0:8090 in 21.24 seconds.
2021-09-06 15:49:08,000 - INFO    - terrariumWebserver   - Running webserver at 0.0.0.0:8090
2021-09-06 15:49:09,136 - WARNING - terrariumEngine      - Could not take a new measurement from sensor css811 co2 named 'CO2 CCS811' at address '5a, 1'. Tried for 0.00 seconds. Skipping this update.
CSS811 DEBUG: load compensation
CSS811 DEBUG: DONE!
CSS811 DEBUG: Load read status
CSS811 DEBUG: 
204
CSS811 DEBUG: error? False
CSS811 DEBUG: result
{'eCO2': 400, 'TVOC': 0, 'status': 152, 'errorid': 0, 'raw': 6599}
CSS811 DEBUG: Exception
name 'sensor_data' is not defined
CSS811 DEBUG: final data
None
2021-09-06 15:49:39,377 - WARNING - hardware.sensor      - Invalid reading from sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1', retrying in 0.5 seconds...
[... 3 lines cut ...]
CSS811 DEBUG: 
144
CSS811 DEBUG: error? False
CSS811 DEBUG: Data is not ready?
2021-09-06 15:49:39,979 - WARNING - hardware.sensor      - Invalid reading from sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1', retrying in 0.5 seconds...
CSS811 DEBUG: load compensation
CSS811 DEBUG: DONE!
CSS811 DEBUG: Load read status
CSS811 DEBUG: 
144
CSS811 DEBUG: error? False
CSS811 DEBUG: Data is not ready?
2021-09-06 15:49:40,581 - ERROR   - hardware.sensor      - Error updating sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1'. Check your hardware! Invalid reading from sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1'
[... cut ...]
2021-09-06 15:49:44,518 - INFO    - terrariumEngine      - Stopped Energenie USB relay named 'Energenie USB device nr: 1(01:01:51:44:b7), Socket: 2' at address '2,01:01:51:44:b7'
2021-09-06 15:49:44,519 - INFO    - terrariumEngine      - Stopped Energenie USB relay named 'Energenie USB device nr: 1(01:01:51:44:b7), Socket: 3' at address '3,01:01:51:44:b7'
2021-09-06 15:49:44,521 - INFO    - terrariumEngine      - Stopped Energenie USB relay named 'Energenie USB device nr: 1(01:01:51:44:b7), Socket: 4' at address '4,01:01:51:44:b7'
2021-09-06 15:49:44,522 - INFO    - terrariumEngine      - Shutdown TerrariumPI 4.0.0 done. Bye bye ...

theyosh, 2021-09-06:
Yesh, I think I have fixed it.

So follow the 5 steps above once more. It should give you just a small code update, but this should fix it.

The debug is still there. I remove that later on. Maybe it is better to leave it so we can also test the calibration option with extra sensors.

theyosh, 2021-09-06:
Hmm, I think I did not finish the compensation part with extra sensors.... hmmm... I have to look at a bit more. But at least the errors should be gone, and you should get some CO2 values now.

wolli1234, 2021-09-06:
The first error is gone, it now gives this error:

> CSS811 DEBUG: 
152
CSS811 DEBUG: error? False
CSS811 DEBUG: result
{'eCO2': 400, 'TVOC': 0, 'status': 152, 'errorid': 0, 'raw': 7634}
CSS811 DEBUG: final data
{'co2': 400}
CSS811 DEBUG: load compensation
CSS811 DEBUG: DONE!
CSS811 DEBUG: Load read status
CSS811 DEBUG: 
152
CSS811 DEBUG: error? False
CSS811 DEBUG: result
{'eCO2': 410, 'TVOC': 1, 'status': 152, 'errorid': 0, 'raw': 6540}
CSS811 DEBUG: final data
{'co2': 410}
CSS811 DEBUG: load compensation
CSS811 DEBUG: DONE!
CSS811 DEBUG: Load read status
CSS811 DEBUG: 
152
CSS811 DEBUG: error? False
CSS811 DEBUG: result
False
2021-09-06 16:55:23,665 - WARNING - hardware.sensor      - Invalid reading from sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1', retrying in 0.5 seconds...
CSS811 DEBUG: load compensation
CSS811 DEBUG: DONE!
CSS811 DEBUG: Load read status
CSS811 DEBUG: 
144
CSS811 DEBUG: error? False
CSS811 DEBUG: Data is not ready?
2021-09-06 16:55:24,266 - WARNING - hardware.sensor      - Invalid reading from sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1', retrying in 0.5 seconds...
CSS811 DEBUG: load compensation
CSS811 DEBUG: DONE!
CSS811 DEBUG: Load read status
CSS811 DEBUG: 
144
CSS811 DEBUG: error? False
CSS811 DEBUG: Data is not ready?
2021-09-06 16:55:24,867 - ERROR   - hardware.sensor      - Error updating sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1'. Check your hardware! Invalid reading from sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1'
2021-09-06 16:55:24,869 - WARNING - terrariumEngine      - Could not take a new measurement from sensor css811 co2 named 'CO2 CCS811' at address '5a, 1'. Tried for 1.45 seconds. Skipping this update.
CSS811 DEBUG: load compensation
CSS811 DEBUG: DONE!
CSS811 DEBUG: Load read status
CSS811 DEBUG: 
152
CSS811 DEBUG: error? False
CSS811 DEBUG: result
{'eCO2': 400, 'TVOC': 0, 'status': 152, 'errorid': 0, 'raw': 6546}
CSS811 DEBUG: final data
{'co2': 400}

Might be due to "burn in" time of the sensor? I had some issues there with TP3 as well. Regarding to the specs it takes 30minutes to give proper data. I'll check back again. 
Regarding the compensation sensors, I tried using the BME280 temperature and humidity. No complaints in the frontend, but I can't see if the data is really used for compensation.

theyosh, 2021-09-06:
It could be that it needs some time to produce stable output. So I think those errors should reduce over time, and stay away. Can you confirm this.. And does the graph show some value in the GUI?

But then again, this is a valid error, and should be treated that way.

And you have just added the BME sensor. Does those sensors also show up when you edit the CO2 sensor again? I have not been able to test this. So, when you edit the CO2 sensor, the select BME sensors are selected again?

Then I need now to think how to load the values of the BME sensor during the updates of the CO2 sensor...

theyosh, 2021-09-06:
Ok, apparently I added some calibration code. So, again, repeat the 5 steps and see what happens :)

wolli1234, 2021-09-06:
The graph is showing data, for now only 400ppm. But this might be the same as in TP3, so after some time this might change. I'll report back.

After editing the CO2 sensor, the BME compensation sensors still show up in the frontend under the CCS811. Both showing the current temperature and humidity values.

The debug info now is:

> CSS811 DEBUG: final data
{'co2': 400}
CSS811 DEBUG: Got new data: {'co2': 400} -> cache key: 5a43f423cd33fa99f2f25a40cde21467
CSS811 DEBUG: Sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1' -> 400
CSS811 DEBUG: final data
{'co2': 400}
CSS811 DEBUG: Got new data: {'co2': 400} -> cache key: 5a43f423cd33fa99f2f25a40cde21467
CSS811 DEBUG: Sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1' -> 400
CSS811 DEBUG: final data
{'co2': 400}
CSS811 DEBUG: Got new data: {'co2': 400} -> cache key: 5a43f423cd33fa99f2f25a40cde21467
CSS811 DEBUG: Sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1' -> 400
CSS811 DEBUG: final data
{'co2': 400}
CSS811 DEBUG: Got new data: {'co2': 400} -> cache key: 5a43f423cd33fa99f2f25a40cde21467
CSS811 DEBUG: Sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1' -> 400
CSS811 DEBUG: ERROR result
False
2021-09-06 17:42:57,841 - WARNING - hardware.sensor      - Invalid reading from sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1', retrying in 0.5 seconds...
CSS811 DEBUG: final data
{'co2': 400}
CSS811 DEBUG: Got new data: {'co2': 400} -> cache key: 5a43f423cd33fa99f2f25a40cde21467
CSS811 DEBUG: Sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1' -> 400
CSS811 DEBUG: Sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1' -> 400
CSS811 DEBUG: final data
{'co2': 400}
CSS811 DEBUG: Got new data: {'co2': 400} -> cache key: 5a43f423cd33fa99f2f25a40cde21467
CSS811 DEBUG: Sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1' -> 400

wolli1234, 2021-09-06:
[
[image]
](url)

theyosh, 2021-09-06:
Ok, found a typo in the code that prevented the calibration to update with the BME values. So repeat the 5 steps again, and then I hope it will also take in account the BME sensor values.

And I think that that the value of 400 will change after a while. It probably needs some time to work correctly.

wolli1234, 2021-09-06:
Thank you for your help! Looks to take calibration into account now:

> CSS811 DEBUG: Update callibration data
{'temperature': 22.542487259791233, 'humidity': 57.21735512697788}
CSS811 DEBUG: Sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1' -> 414
CSS811 DEBUG: Start updating calibrations
CSS811 DEBUG: Found sensor: BME280 sensor temperature named 'Temperature BME280' at address '76, 1'
CSS811 DEBUG: Found sensor: BME280 sensor humidity named 'Humidity BME280' at address '76, 1'
CSS811 DEBUG: Update callibration data
{'temperature': 22.527178075310076, 'humidity': 57.26899577085642}
CSS811 DEBUG: final data
{'co2': 415}
CSS811 DEBUG: Got new data: {'co2': 415} -> cache key: 5a43f423cd33fa99f2f25a40cde21467
CSS811 DEBUG: Sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1' -> 415
CSS811 DEBUG: Start updating calibrations
CSS811 DEBUG: Found sensor: BME280 sensor temperature named 'Temperature BME280' at address '76, 1'
CSS811 DEBUG: Found sensor: BME280 sensor humidity named 'Humidity BME280' at address '76, 1'
CSS811 DEBUG: Update callibration data
{'temperature': 22.542487259791233, 'humidity': 57.211554044267935}
CSS811 DEBUG: Sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1' -> 415

theyosh, 2021-09-06:
COOL! I removed the debug and it should now work nicely. Also thanks for testing!

So, if you stop TerrariumPI, and update the code with `git pull`. Then you can now start it again with the service command: `sudo service terrariumpi start`

And keep it running and see what the CO2 is doing

theyosh, 2021-09-08:
And, how is it running? Does it look stable and ok? Could we close this issue?

wolli1234, 2021-09-09:
Sometimes there comes a warning:

> WARNING - hardware.sensor      - Invalid reading from sensor CCS811 sensor co2 named 'CO2 CCS811' at address '5a, 1', retrying in 0.5 seconds...

But this was the same in TP3. I'd say it is stable and ok now and you can close this issue. Thank you!

theyosh, 2021-09-09:
I did take in account this kind of 'errors' and therefore, it should not be a big issue. If this is the only sensor, and only happens less then 10 times an hour, I do not see an a problem. Especially, when there are only warnings. That means, it just took some more time to get the data. When it failed, you will get an ERROR message.

Then, if the total update stays below 30 seconds, there is no problem.

So thanks for the update, and I close it now.

closed 2021-09-09 by theyosh