theyosh/TerrariumPI #590: Gpio sensor errors

tgrantjr, 2021-08-19:
I have 2 am2302 sensors connected and no matter what address I put them on I get errors that it can't communicate. I have the positive and negative wires going to the 5v power supply for both enclosures. It's a 150 amp supply so I k ow there is enough power. The data is going to the pi. Do I have this wired correctly?  Thanks

theyosh, 2021-08-20:
Hi, I do not know. I would just try to follow this tutorial: https://thepihut.com/blogs/raspberry-pi-tutorials/am2302-temp-humidity-sensor

If that is working, you know that you hardware is correct. Then you can added to my software. That is what I can do at the moment.

tgrantjr, 2021-08-20:
I got that figured out thanks. Now I'm trying to get the sensors to talk to the relay to turn the heat lamp on and off when it's in the temperature range and I cannot find that setting. Any help with that?

closed 2021-08-20 by tgrantjr