theyosh/TerrariumPI #593: [BUG]: openweathermap source documentation error

chrRtg, 2021-08-21:
Raspi 3b with TerrariumPI, last version from today

**Describe the bug**
The documentation say the weather source URL should look like:  
https://api.openweathermap.org/data/2.5/weather?q=[CITY],[COUNTRY_2LETTER]&units=metric&appid=[YOUR_API_KEY]

The parameter "&units=metric" is not covered by the RegEx in weather/openweathermap_org_weather.py, see "VALID_SOURCE"

If I remove the parameter "&units=metric" TerrariumPI is fetch and shows the weather.

theyosh, 2021-08-23:
Hi,

With TP version are you using? Because in TP3, there is a requirement to add `&units=metric` https://github.com/theyosh/TerrariumPI/blob/3.x.y.z/terrariumWeather.py#L369

In TP4 it needs to be stripped, so I can manually add it to make sure I get the right values: https://github.com/theyosh/TerrariumPI/blob/4.x.y.z/weather/openweathermap_org_weather.py#L14

So then which documentation are you referring? All documentation at this time is for TP3.

chrRtg, 2021-08-23:
sorry, forgot to write the version :/
It's the most current V4. Had to look into the code to understand how the URL has to be build.

theyosh, 2021-08-24:
Hmm, did you click on the question mark icon at the weather source form? Because there is a small explanation with the right url. And a link to: https://openweathermap.org/current which explains the url.

And I am working on the documentation at: https://theyosh.github.io/TerrariumPI/

So I think I will close this, as I think there is enough information. But maybe a bit more clearer. But that can be done in the README file.