theyosh/TerrariumPI #594: I2c display and relay issue

tgrantjr, 2021-08-23:
I have a 20x4 display connected to the proper pins on the pi and have no output. Is there some setting somewhere I would have to change to activate it?  Also I'm trying to use a relay for my heat lamp to be controlled by the am2302 sensor when temp is out of spec. Any assistance with this?  Thanks

theyosh, 2021-08-24:
Hi,

I would like you to ask to open a new issue for the relay. It is not wise to combine issues in one ticket.

So first off, the display is not working in TP4. That is not finished. So if you are using TP4, display does not work. For TP3 it should work.

So how do you know that the display is connected correctly? You have checked this with i2cdetect?

closed 2021-08-24 by tgrantjr

theyosh, 2021-08-24:
So, I guess you are using TP4. I still have the wish to enable the Displays again. But for now, it took to much work.