theyosh/TerrariumPI #595: Relay controlled by am2302

tgrantjr, 2021-08-24:
I have my setup and the last part I need is to figure out how to trigger a relay for the heat lamp when my Temps are at range. I cannot find any setting for this. Is this possible to do?

theyosh, 2021-08-24:
Yes!

First, the AM2302 sensor is working? You get the right temperatures?

Then you should go to 'System' -> 'Environment'. There you can a temperature part. Use the sensor mode and use the low alarm part to select your relays.

tgrantjr, 2021-08-24:
Yeah the sensors are working perfectly. I'll look in that area.

theyosh, 2021-08-25:
Euh, oh, you are using TP4? That is fine. 

Then it is a bit different. Then you need to go to enclosures, and add a enclosure first. Then you can add area's to an enclosure. There you can then select the lights and temperature control.

So by adding a heating area to your enclosure, you can select your sensor and then the needed relays. It is pretty straightforward.

theyosh, 2021-09-01:
Is it working for you? Any updates?