theyosh/TerrariumPI #600: TP v4 - no dashboard available

jornobe, 2021-08-30:
v4 installed nicely today, well done on the multi enclosure and areas component! 👍

In the sidebar, when expanding <home> and then clicking on <dashboard>, nothing happens. I have the <all sensors> page opening just fine but no dashboard.

Anything I can do to activate a dashboard as we had it in the previous version?

theyosh, 2021-08-30:
Hi, I am not sure what you mean? I guess you do not have sensors that are auto-detected. So it could be that the dashboard is empty, or not showing up. I will try to test this.

But could you post a screenshot in the mean time? That makes it easier to understand.

theyosh, 2021-08-30:
If I have zero hardware connected, I get this empty dashboard:
[image]
So it should work. If you post a screenshot we can compare

captainboto, 2021-08-30:
I have this issue yesterday 
you try Clear cached browsing data. it's fixed for me

jornobe, 2021-08-30:
> I have this issue yesterday you try Clear cached browsing data. it's fixed for me

Should have been my first attempt to resolution, awesome! Works great now!

@theyosh , solved by clearing cache

theyosh, 2021-08-30:
Yeah, there is some heavy caching in the code. This to reduce the load on the Pi. But it has it downsides. So I am currently working on a caching fix.

theyosh, 2021-08-30:
I will close it for now. As I also have updated the code. So to make sure you have the latest code that should tackle about 80% of the caching issues after adding/updating/deleting data.

So do a `git pull` and restart TerrariumPI: `sudo service terrariumpi restart`

closed 2021-08-30 by theyosh