theyosh/TerrariumPI #602: [FEATURE]: Time of day temperature

PhoenixNP, 2021-09-01:
**Is your feature request related to a problem? Please describe.**
Would it be possible to adjust the min/max temperature based on the time of day?
Right now it's only possible to set a min/max with a deviation for the night time (where im semi-uncertain it works)

**Describe the solution you'd like**
Options to add times and temperates as to emulate sunrise/sundown.
For example;
08:00 - 28 degree's
11:00 - 29 degree's
13:00 - 30 degree's
15:00 - 29,5 degree's
18:00 - 29 degree's
22:00 - 28 degree's

Just listing arbitrary values

PhoenixNP, 2021-09-02:
Well I actually think there are multiple ways of achieving it.
Relative Settings, user configureable
Absolute settings, user configurable
Follow the Sun/Weather settings and raise/lower temperature based on the weather. This would probably be an annoying one since a lot of our animals come from tropical area's with a way different time zone. So that would also mean mapping their "normal" time to our time.

At the moment I heat up most of my environments with Radiant Heat Panels or Deep Heat Projectors on a dimmer (they don't really emit light). That means that for the sunlight im less reliant on the lightings.
For example in my case I switch the lights at sunrise, in a delayed fashion, to mimick sunrise, and do the same with the shutdown in the evening.

As for suggestions;
Both relative as well as absolute value's could and would work, I think this mostly comes down to user preference and easy of setup. I think that just setting a min and max value for the day and a min/max value for the night and calculate percentages throughout the 24/7 would be less predicatable and perhaps for users end up with wrong times

theyosh, 2021-09-06:
> Well I actually think there are multiple ways of achieving it.
> Relative Settings, user configureable
> Absolute settings, user configurable

I think it is much easier to enter '-4' and '+4' for relative values and '28' as an absolute value. Else you get so much options, that is not handy

> Follow the Sun/Weather settings and raise/lower temperature based on the weather. This would probably be an annoying one since a lot of our animals come from tropical area's with a way different time zone. So that would also mean mapping their "normal" time to our time.

Not sure, but I am afraid that this is not correct. At the moment, I use the weather for sun rise and sun set. And if you specify a location in the tropics, it only uses the on and off times. And yes, that is mapped to our location. Else your reptile can have daylight while it is night at your location. And I still offer the option to specify weather source for the lights, and that needs to be supported. Weather data only tells how long there is light.

The only 'new' thing is, that you want your sensors to go up and down with the alarms based on the weather / location temperatures. That is harder to do.
> At the moment I heat up most of my environments with Radiant Heat Panels or Deep Heat Projectors on a dimmer (they don't really emit light). That means that for the sunlight im less reliant on the [...]

This is possible with my software (In TP4 better then TP3). Dimming is using the PID principle https://en.wikipedia.org/wiki/PID_controller

> As for suggestions;
> Both relative as well as absolute value's could and would work, I think this mostly comes down to user preference and easy of setup. I think that just setting a min and max value [...]

I think this is still an option to use/explore

> 
> Edit: right now my herpstats and microclimates are set up using absolute values throughout the day, maybe I'm spoiled. (but I'd love to replace those with TerrariumPI)

The only problem I found is that when you have multiple heating area's in your enclosure, do you set the temperature changes on all area's. Or do you specify a special (new) area type, which will only control the min and max alarm values of all the temperature sensors that are being used in the enclosure...

I need some more thinking about this.

theyosh, 2021-12-10:
Closing this issue, as with the new [release 4.1.0](https://github.com/theyosh/TerrariumPI/releases/tag/4.1.0) this is now possible.

closed 2021-12-10 by theyosh