theyosh/TerrariumPI #603: [BUG]: Script relay not working

wolli1234, 2021-09-05:
Hello Yosh,

I tried to use a script delay which is not working.

**Setup:**
 - TerrariumPI version: 4
 - Python version: 3
 - Raspberry PI: 3B

I added a script relay with the address "/home/pi/customScript/LED.py". It gives the following error messages:

> 2021-09-05 16:12:02,626 - ERROR   - hardware.relay       - Error getting new data from relay Script relay relay named 'Script LED' at address '/home/pi/customScript/LED.py'. Error: unknown
2021-09-05 16:12:02,624 - ERROR   - hardware.relay       - Error getting new data from relay Script relay relay named 'Script LED' at address '/home/pi/customScript/LED.py'. Error: 'NoneType' object has no attribute 'decode'

**To Reproduce / Installation steps**
This is the same script I used in TP3, where it was working fine:

```
#!/usr/bin/python3
import sys
import pigpio

RED_PIN   = 17
GREEN_PIN = 22
BLUE_PIN  = 24

#Green Pin is not connected, instead a small white 6500K LED-Strip is controlled
#Ampere per color: Red (alone) = 1,1 A, White (alone) = 1,43 A, Blue (alone) = 1 A
# Ampere all together, full brightness: 2,98 A
# Ampere red 255, white 255, blue 166: 2,98 A

pi =pigpio.pi()

bright = 255

def setLights(pin, brightness):
	realBrightness = int(int(brightness) * (float(bright) / 255.0))
	pi.set_PWM_dutycycle(pin, realBrightness)

# This is just an example script to test the functionality for TerrariumPI. It is also a reference for implementing your own scripts

if len(sys.argv) == 1:
  raise SyntaxError('No arguments specified')

try:
  value = int(sys.argv[1])
except Exception as ex:
  raise ValueError('Invalid input value. Should be a number between 0 and 100 including')

if not (0 <= value <= 100):
  raise ValueError('Invalid input value. Should be a number between 0 and 100 including')

if value == 0:
  print('Toggle power switch to state OFF')
  # Call your code here....
  setLights(RED_PIN, 0)
  setLights(GREEN_PIN, 0)
  setLights(BLUE_PIN, 0)
else:
  print('Toggle power switch to state ON or dimming value: {}'.format(value))
  # Call your code here....
  setLights(RED_PIN, 255)
  setLights(GREEN_PIN, 128)
  setLights(BLUE_PIN, 128)`
```
I also tried manual mode for the script delay, trying to switch the relay on and off, but this is also not working.

Best regards,
Wolli

theyosh, 2021-09-06:
Hi, for TP4 the script has to change a bit. In TP4, I do expect also to  be able to readout the script relay. New example: https://github.com/theyosh/TerrariumPI/blob/4.x.y.z/contrib/external_switch.py

And take care about https://github.com/theyosh/TerrariumPI/blob/4.x.y.z/contrib/external_switch.py#L10 . If there are no parameters given, that means I want to know the current state. So either return -1 which tells TP4 to use the existing DB value. Or, if you are able to readout your lights current state, you could also do that. But for now, I suggest to return -1.0 so that it should work again.

I assumed that when you run the script by hand, it works? So that means all the python libraries are installed correctly. Correct?

So I made a small update of your script which incorporates the -1.0 value for the readout.

```
#!/usr/bin/python3

# This is just an example script to test the functionality for TerrariumPI. It is also a reference for implementing your own scripts

import sys
import pigpio

RED_PIN   = 17
GREEN_PIN = 22
BLUE_PIN  = 24

#Green Pin is not connected, instead a small white 6500K LED-Strip is controlled
#Ampere per color: Red (alone) = 1,1 A, White (alone) = 1,43 A, Blue (alone) = 1 A
# Ampere all together, full brightness: 2,98 A
# Ampere red 255, white 255, blue 166: 2,98 A

pi =pigpio.pi()

bright = 255

def setLights(pin, brightness):
  realBrightness = int(int(brightness) * (float(bright) / 255.0))
  pi.set_PWM_dutycycle(pin, realBrightness)

# When there is no extra arguments, TerrariumPI expect a float value to return the current state.
# Return -1.0 when not supported.
if len(sys.argv) == 1:
  # Provide a fake dimmer readout...
  #print(random.randint(0,100))

  # Or no readout test
  print('-1.0')

else:

  try:
    value = int(float(sys.argv[1]))
  except Exception as ex:
    raise ValueError('Invalid input value. Should be a number between 0 and 100 including')

  if not (0 <= value <= 100):
    raise ValueError('Invalid input value. Should be a number between 0 and 100 including')

  if value == 0:
    print('Toggle power switch to state OFF')
    # Call your code here....
    # Call your code here....
    setLights(RED_PIN, 0)
    setLights(GREEN_PIN, 0)
    setLights(BLUE_PIN, 0)
  else:
    print('Toggle power switch to state ON or dimming value: {}'.format(value))
    # Call your code here....
    setLights(RED_PIN, 255)
    setLights(GREEN_PIN, 128)
    setLights(BLUE_PIN, 128)
```

wolli1234, 2021-09-06:
There was indeed the pigpio library missing. After installing I can run your script in the console and it prints "-1".

In TP4 I can't add it:

> 2021-09-06 17:03:18,686 - WARNING - hardware.relay       - Could not load hardware for relay Script relay relay named 'Script LED' at address '/home/pi/customScript/LED_TP4.py': Script /home/pi/customScript/LED_TP4.py for relay Script relay relay named 'Script LED' at address '/home/pi/customScript/LED_TP4.py' is not executable.

I did "chmod u+x LED_TP4.py" as well.

wolli1234, 2021-09-06:
Now it works, seems I just had to use "chmod +x LED_TP4.py" instead of "chmod u+x LED_TP4.py".

Just a note: When I switch this custom relay, the popup message is the opposite. So when I turn the lights on it says "turned off" and vice versa.

theyosh, 2021-09-06:
Made some small updates. Do a `git pull` and then `sudo service terrariumpi restart`. And I hope it works now better.

Also made the executable check a bit better and should support u+x now also

wolli1234, 2021-09-07:
I could not access the frontend anymore. I don't know if this was a bug, I freshly reinstalled TP4 and frontend is working again.

Since then the script is not working anymore. I tried doing a git pull and restart TP. It stays switched on, but I can't switch it off:

> 2021-09-07 13:43:25,655 - ERROR   - hardware.relay       - Error changing state for relay Script relay relay named 'LED_TP4' at address '/home/pi/customScript/LED_TP4.py' to 0.0 :Error changing relay Script relay relay named 'LED_TP4' at address '/home/pi/customScript/LED_TP4.py' to state 0.0. Error: Error changing relay Script relay relay named 'LED_TP4' at address '/home/pi/customScript/LED_TP4.py' to state 0.0. Error: unknown

I really don't know what is the difference, as it is the exact same script as before.

theyosh, 2021-09-07:
Sorry, my mistake.

Do a `git pull` and restart `sudo service terrariumpi restart`

theyosh, 2021-09-08:
So, it is now all working? For me it is working now with scripts

wolli1234, 2021-09-09:
Yes now it is working and popup message for switching on and off is correct.

theyosh, 2021-09-09:
Thanks for the update.

closed 2021-09-09 by theyosh