theyosh/TerrariumPI #604: Error with reading temperature.

vikrilow, 2021-09-05:
Hello
Reviewing TerrarirumPI Register, I've seen this error:

2021-09-05 19:29:09,394 - WARNING - terrariumSensor      - Measured temperature value NoneC from w1 sensor 'TEMP_HOT_HID_ZONE' is outside valid range 5.00C - 100.00C in 0.00211 seconds.
FileNotFoundError: [Errno 2] No such file or directory: '/sys/bus/w1/devices/28-0119355be4fd/w1_slave'
with open(os.path.join(terrarium1WSensor.W1_BASE_PATH,self.get_address(),'w1_slave'), 'r') as w1data:
File "/home/pi/TerrariumPI/terrariumSensor.py", line 334, in load_data
Traceback (most recent call last):
2021-09-05 19:29:09,392 - ERROR   - terrariumSensor      - Error loading 1 Wire data at location: /sys/bus/w1/devices/28-0119355be4fd/w1_slave with error: [Errno 2] No such file or directory: '/sys/bus/w1/devices/28-0119355be4fd/w1_slave'
TypeError: '<=' not supported between instances of 'float' and 'NoneType'
if current is None or not (self.get_limit_min() <= terrariumUtils.conver_to_value(current,self.get_indicator()) <= self.get_limit_max()):
File "/home/pi/TerrariumPI/terrariumSensor.py", line 119, in update
self.sensors[sensorid].update()
File "/home/pi/TerrariumPI/terrariumEngine.py", line 623, in __engine_loop
Traceback (most recent call last):
2021-09-05 19:29:09,290 - ERROR   - terrariumEngine      - Engine loop: Sensor has problems: '<=' not supported between instances of 'float' and 'NoneType'

Really I don't know whats is the problem.

The graphics show this form:

[image]

And in the temperature configuration, really system read parameter:
[image]

theyosh, 2021-09-06:
This sounds like a hardware failure or loose cables. As this is the error:

```
FileNotFoundError: [Errno 2] No such file or directory: '/sys/bus/w1/devices/28-0119355be4fd/w1_slave'
with open(os.path.join(terrarium1WSensor.W1_BASE_PATH,self.get_address(),'w1_slave'), 'r') as w1data:
File "/home/pi/TerrariumPI/terrariumSensor.py", line 334, in load_data
```

For 1-Wire I am using the Raspberry PI default overlay. And that should produce folders in `/sys/bus/w1/devices/` which holds IDs of the sensors and a file called: `w1_slave`. If there no folders visible in the `/sys/bus/w1/devices/` then you have a hardware error with the 1-Wire sensors/cables.

If there are problems with the 1-wire cabling then this could happen. This is outside the scope of TerrariumPI. So you should look at your 1-wire cabling if everything is still connected.

Rewire the 1-Wire devices? Make sure the pins are connected fully (maybe a bit loose)

Fully update the Raspberrry PI and reboot: `apt update && apt full-upgrade`

vikrilow, 2021-09-07:
OK, thanks!
I'll review cables.
Meanwhile, I've configured 2 Mi Temperature and Humidity. 
You can close this issue.
Thanks!

theyosh, 2021-09-08:
Ok, good look with the cabling...

closed 2021-09-08 by theyosh