theyosh/TerrariumPI #605: Timer Intervall for Relay

wolli1234, 2021-09-07:
**Setup:**
 - TerrariumPI version: 4
 - Python version: 3
 - Raspberry PI: 3B

**Describe the bug**
I was looking for a solution to run an exhaust air fan four times an hour for five minutes.

Finally this area setup worked for me:
[image]

This gave me quiet a headache. If you change and save values in the area settings, e.g. duration time, this will only take effect if you restart terrariumpi. Elsewise it will interfere? with the old values. My fan went on and off with no logic according to the saved values. This is important to know when you play around with it. Maybe cache or something should to be deleted while saving values in area settings?

Best regards,
Wolli

theyosh, 2021-09-07:
Hi,

if you want your fan to run 4 times an hour for 5 minutes each time, you should use something like this:

[image]

In this setup the fans will start at 10 in the morning, and will end at 10 in the evening. 

Then I want that the power is on for 5 minutes, and then off for 10 minutes. So this makes a single cycle of 15 minutes. And this will repeated from 10:00 to 22:00 hours.

Because the 'Power on time' and 'Settle time' are zero, they will be ignored.

So, now you have a fan running 5 minutes, and 10 minutes off. And that for 12 hours in total. I hope this makes it clear.

wolli1234, 2021-09-08:
So far my setup on the screenshot works as well. I just wanted to let you know, that changing the values caused weird behavior until I restart terrariumpi. Old values seem not to be overwritten completely until a restart. I don't know if this is a local bug on my pi or general behavior.

theyosh, 2021-09-08:
Hi,

thanks for the feedback. It could be that editing area's is not working correctly. I have not fully tested that. So I will look into that.

But, your fans are now working like the way you want?

wolli1234, 2021-09-09:
yes my fans work the way they should now.

closed 2021-09-09 by theyosh