theyosh/TerrariumPI #606: [BUG]: offline right after setting the areas up

kebs19, 2021-09-20:
**Setup:**
 - TerrariumPI version:  4]
 - Python version: r 3]
 - Raspberry PI: 2

**Describe the bug**
I just installed the new version of Terrariumpi, entered the address for the switches by hand and tried to configure the areas. it just got offline and doesn't get back.

**Additional context**
using the manual debug of Terrariumpi4, I got this log:

2021-09-20 07:09:21,625 - INFO    - terrariumEngine      - Loaded 8 relays in 10.90 seconds.
2021-09-20 07:09:21,628 - INFO    - terrariumEngine      - Loading existing buttons from database.
2021-09-20 07:09:21,647 - INFO    - terrariumEngine      - Loaded 0 buttons in 0.02 seconds.
2021-09-20 07:09:21,650 - INFO    - terrariumEngine      - Loading existing webcams from database.
2021-09-20 07:09:21,667 - INFO    - terrariumEngine      - Loaded 0 webcams in 0.02 seconds.
2021-09-20 07:09:21,669 - INFO    - terrariumEngine      - Loading existing enclosures from database.
Traceback (most recent call last):
  File "terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 154, in __init__
    self.__load_existing_enclosures()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 921, in __load_existing_enclosures
    new_enclosure.update()
  File "/home/pi/TerrariumPI/terrariumEnclosure.py", line 118, in update
    area_states[area_id] = self.areas[area_id].update(read_only)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 660, in update
    super().update(read_only)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 382, in update
    toggle_relay = self._is_timer_time(period)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 289, in _is_timer_time
    return main_lights._is_timer_time(period)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 292, in _is_timer_time
    for time_schedule in self.setup[period]['timetable']:
KeyError: 'low'
^A^CKeyboardInterrupt
2021-09-20T10:11:34Z
Exception ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/monkey.py", line 911, in _shutdown
    orig_shutdown()
  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown
    t.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/thread.py", line 121, in acquire
    acquired = BoundedSemaphore.acquire(self, blocking, timeout)
  File "src/gevent/_semaphore.py", line 180, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_semaphore.py", line 249, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_abstract_linkable.py", line 521, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait
  File "src/gevent/_abstract_linkable.py", line 487, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 490, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 442, in gevent._gevent_c_abstract_linkable.AbstractLinkable._AbstractLinkable__wait_to_be_notified
  File "src/gevent/_abstract_linkable.py", line 451, in gevent._gevent_c_abstract_linkable.AbstractLinkable._switch_to_hub
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch

theyosh, 2021-09-21:
Hi,

unfortunately, I can only see that (you) pressed Ctrl+C

So, at this point, there is no logging which indicate something is wrong. So I am afraid that this is something else.

Is it still the case? Could you else upload the logfile?

kebs19, 2021-10-06:
sure. it's working now :)