theyosh/TerrariumPI #608: [BUG]: Cant delete enclosure

kebs19, 2021-09-20:
When I try to delete an enclosure, I receive thefollowing error:
Error:
Error deleting enclosure 885bb0c8-1e91-40e5-8da9-e7be573ed6cc. [Errno 21] Is a directory: '.'

theyosh, 2021-09-21:
Hmm, ok, it should also be fixed now without an image. I was able to delete an enclosure without an image.

theyosh, 2021-09-21:
Ah, to be clear, you need to update the code and restart terrariumpi

kebs19, 2021-09-22:
Helo, :)

I tried to update the code and create new enclosures, here is what happens:

When I create an enclosure the 'image file' space is filled with an image that I already uploaded and the new enclosure uses that image and can be deleted fine.
If I reload the browser and create an enclosure the 'file image' space is blank and the enclosure has no image (which is the right thing) but when I try to delete this enclosure it says:
Error:
Error deleting enclosure 997e51ee-81ca-4d6b-a6b3-4f3bdc5d4b98. [Errno 21] Is a directory: '.'

The only workaround is to upload an image to this enclosure and then delete it.

theyosh, 2021-09-26:
Hmm, this is strange. This will take some more time to figure this out.

kebs19, 2021-10-12:
I tried it again and it works now, with enclosures with or without images 
I'll close the issue