theyosh/TerrariumPI #609: [BUG]: Enclosures can't turn on more than a switch at the same time

kebs19, 2021-09-21:
I have a following enclosure:

A Lid enclosure that will control the lights and cool down the lid when it gets too hot. It's setup this way:
1 timer switch from 06h to 18h (dusk to dawn)
1 weather switch (that turns on and off according to the sunrise/sunset, usually 07-17h)
1 timer switch from 11h to 15h (the period of time when light is strongest in the tropics)
and 1 time switch from 11h to 18h toturn a fan on and coool down the led system (it will be controlled by a temp sensor, but I'm sticking to timered settings in this test setup yet)

What should happen:
Let's say it's 13h, it need to turn on all the switches.

What happens:
Only one switch is on. 

I created 3 more enclosures with similar setups and the same thing happens: only one switch is turned on at the time.

kebs19, 2021-09-21:
Here, 4 switches are supposed to be on:
[image]

Here, only 3 are on. One in which enclosure
[image]

theyosh, 2021-09-26:
This is hard to debug at this time. Are you sure you have not re-used a relay in the area's?

What is the logfile telling you?

closed 2021-10-12 by kebs19