theyosh/TerrariumPI #610: [BUG]: Dashboard is empty

kaju666, 2021-09-23:
**Setup:**
 - TerrariumPI version: 4
 - Python version: 3
 - Raspberry PI: 4

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce / Installation steps**
As in instruction.
Stopped old.
mv /home/pi/TerrariumPI /home/pi/TerrariumPI.old
git clone --branch 4.x.y.z --depth 1 https://github.com/theyosh/TerrariumPI.git
cd TerrariumPI
sudo ./install.sh
Restarted Pi

**Screenshots**
[image]

theyosh, 2021-09-26:
Hmm, I am not sure what is causing this. But I think the easiest solution now is to stop TP4, and remove the database and restart TP4.

So remove the file `/home/pi/TerrariumPI/terrariumpi.db` and restart.

Else, try to if there are javascript errors in the browser. But I can't reproduce this.

kaju666, 2021-09-26:
it's chrome issue but i can't figure it out. deleting webpage data and cache didn't help.
on edge is working

theyosh, 2021-09-26:
Hmm, with Chromium I do not have any issues. I use Firefox normally. But does Chrome not give an error in the developer console?

kaju666, 2021-10-01:
for no reason it started to work... :)

closed 2021-10-01 by kaju666