theyosh/TerrariumPI #611: [BUG]: API on version 4 not fully working.

kaju666, 2021-09-23:
****Setup:**
 - TerrariumPI version:  4
 - Python version: 3
 - Raspberry PI: 4

**Describe the bug**
Most API's are not working on version 4.
for example: /api/uptime/ giving 404 also on your site terrarium.theyosh.nl
I think only api/sensors/ is working.

Also i found that on verion 4 there must be "/" on the end. On version 3 api link was without last "/"

theyosh, 2021-09-26:
The API is different. I did not make the new documentation for this. The urls are changed. Long story....

kaju666, 2021-09-26:
ok :) will be waiting for some docs :)

closed 2021-09-26 by kaju666

theyosh, 2021-09-30:
Here is an update. Not fully up to date... but working on it: https://terrarium.theyosh.nl/api/doc/