theyosh/TerrariumPI #612: [FEATURE]: V4.x SONOFF multi switch (same as v3.x #421 & #445)

ewiandr73, 2021-09-27:
Thanks for all your work the v4.0 is looking very promising now.

**Is your feature request related to a problem? Please describe.**
I raised cases 421 & 445 against version 3.xto allow addressing of multiple relay's on the one SONOFF device.
I've now got what looks like a stable v4.0 install and wanted to see if I could add the sonoff control to multiple enclosures

SONOFF multi switch #421 
Read Temperature from SONOFF device #445

I have a 4 Relay SONOFF device that also has a couple of  DS18DB20 sensors.

**Describe the solution you'd like**
Ability to address all 4 relays 

http://192.168.86.91/#1 (Works)
http://192.168.86.91/#2 (Can't add)
http://192.168.86.91/#3 (Can't add)
http://192.168.86.91/#4 (Can't add)

Currently Relay error after the first relay was added:

_Save Error
Relay could not be added. Object sonoff relay named 'Heater' at address 'http://192.168.86.91/#2' cannot be stored in the database. IntegrityError: UNIQUE constraint failed: Relay.id_

Also be able to address 
http://192.168.86.91/cm?cmnd=sonoff/status 10#StatusSNS/DS18B20-1/Temperature
http://192.168.86.91/cm?cmnd=sonoff/status 10#StatusSNS/DS18B20-2/Temperature
http://192.168.86.91/cm?cmnd=sonoff/status 10#StatusSNS/DS18B20-3/Temperature
http://192.168.86.91/cm?cmnd=sonoff/status 10#StatusSNS/DS18B20-4/Temperature
http://192.168.86.91/cm?cmnd=sonoff/status 10#StatusSNS/DS18B20-5/Temperature
http://192.168.86.91/cm?cmnd=sonoff/status 10#StatusSNS/DS18B20-6/Temperature
http://192.168.86.91/cm?cmnd=sonoff/status 10#StatusSNS/DS18B20-7/Temperature
http://192.168.86.91/cm?cmnd=sonoff/status 10#StatusSNS/DS18B20-8/Temperature

Current Temperature Error

_Save Error
Sensor could not be added. float() argument must be a string or a number, not 'dict'_

**Additional context**

theyosh, 2021-09-27:
Hi, hmm, for the relays I need to update the code. But could you post the JSON from the temperature sensor?

And could it help to change the spaces to `%20` in ` http://192.168.86.91/cm?cmnd=sonoff/status 10#StatusSNS/DS18B20-4/Temperature` so it would become: ` http://192.168.86.91/cm?cmnd=sonoff/status%2010#StatusSNS/DS18B20-4/Temperature`

theyosh, 2021-09-27:
Ok, I found the relay issue. So do a `git pull` and restart. Now you should be able to add more sonof relays as described.

And for the temperature sensors, you need to change the space to %20 https://github.com/theyosh/TerrariumPI/issues/445#issuecomment-670882848

ewiandr73, 2021-09-28:
Awesome I can now add relay's and temp sensors via the SONOFF.

However the who system seems to crash and I lost the GUI, so I renamed the terrariumpi.db and rebooted the pi all can back up so I created just one relay to see if it remains stable. I get the following error and warnings when operating this relay.

Switch On
2021-09-28 16:09:12,696 - ERROR   - hardware.relay       - Error changing state for relay Sonoff (Tasmota) relay named 'Relay-2-1 (Lights)' at address 'http://192.168.86.92/#1' to 100.0 :Error changing relay Sonoff (Tasmota) relay named 'Relay-2-1 (Lights)' at address 'http://192.168.86.92/#1' to state 100.0. Error: 'POWER'
2021-09-28 16:09:12,115 - WARNING - hardware.relay       - Error changing relay Sonoff (Tasmota) relay named 'Relay-2-1 (Lights)' at address 'http://192.168.86.92/#1' to state 100.0. Error: 'POWER', retrying in 0.5 seconds...
2021-09-28 16:09:11,559 - WARNING - hardware.relay       - Error changing relay Sonoff (Tasmota) relay named 'Relay-2-1 (Lights)' at address 'http://192.168.86.92/#1' to state 100.0. Error: 'POWER', retrying in 0.5 seconds...

Switch Off
2021-09-28 16:11:05,127 - ERROR   - hardware.relay       - Error changing state for relay Sonoff (Tasmota) relay named 'Relay-2-1 (Lights)' at address 'http://192.168.86.92/#1' to 0.0 :Error changing relay Sonoff (Tasmota) relay named 'Relay-2-1 (Lights)' at address 'http://192.168.86.92/#1' to state 0.0. Error: 'POWER'
2021-09-28 16:11:04,571 - WARNING - hardware.relay       - Error changing relay Sonoff (Tasmota) relay named 'Relay-2-1 (Lights)' at address 'http://192.168.86.92/#1' to state 0.0. Error: 'POWER', retrying in 0.5 seconds...
2021-09-28 16:11:03,991 - WARNING - hardware.relay       - Error changing relay Sonoff (Tasmota) relay named 'Relay-2-1 (Lights)' at address 'http://192.168.86.92/#1' to state 0.0. Error: 'POWER', retrying in 0.5 seconds...

If things remain stable I will try added the temp sensors and then the timers, to see if I can find which bit crashes the system (if it happens again).

Regards
Andy

theyosh, 2021-09-28:
Hmm, found an issue. So I updated the code. Lets see if this will fix it. Do a `git pull` and restart.

Could you test to toggle a single Sonoff relay outside TerrariumPI. Toggle it with the Tasmota interface. Then my software should see the change, and change it in the interface. Then it should work again.

ewiandr73, 2021-09-28:
I tried with the single as you described and that seems to work correctly.

I then added a second but didn't toggle it with the Tasmota interface and the 2nd relay didn't seem to function correctly, so I deleted the 2nd relay again, the 1st still worked correctly, so I then added the remaining 3 relays and toggled them all in Tasmota one at a time, now all 4 seem to be working correctly. 👍 

How many of these 4CH relays should the system be able to support?
I have 10 snakes that I was hoping to eventually connect  to the same RPi4 (2G) each with it's own 4CH sonoff relay, do see any issues with this?

Thanks, regards
Andy

ewiandr73, 2021-09-29:
I connected up the lights and heaters for 2 of the enclosures (Via the same 4CH SONOFF) and noticed the lights flicker even 30 seconds or so in both enclosures.

'Relay-2-1 (Lights-Snake-1)' at address 'http://192.168.86.92/#1' 
'Relay-2-2 (Heater-Snake-1)' at address 'http://192.168.86.92/#2'
'Relay-2-3 (Lights-Snake-2)' at address 'http://192.168.86.92/#3'
'Relay-2-4 (Heater-Snake-2)' at address 'http://192.168.86.92/#4'

The below log seems to repeat ever 30 seconds 

Log:
2021-09-29 12:43:28,494 - INFO    - terrariumEngine      - Engine update done in 0.95 seconds. Waiting for 29.05 seconds for the next round.
2021-09-29 12:43:28,337 - INFO    - terrariumEngine      - Updated Enclosure Snake-2 with 2 areas in 0.15 seconds.
2021-09-29 12:43:28,332 - INFO    - terrariumArea        - Set the relay Relay-2-3 (Lights-Snake-2) to 100.0 with a delay of 0.00 minutes
2021-09-29 12:43:28,332 - INFO    - terrariumEngine      - Updated Enclosure Snake-2 with 2 areas in 0.01 seconds.
2021-09-29 12:43:28,326 - INFO    - terrariumEngine      - Updated Enclosure Snake-1 with 2 areas in 0.01 seconds.
2021-09-29 12:43:28,188 - INFO    - terrariumArea        - Toggle the relays for area Lighting named Lights-2 to state on.
2021-09-29 12:43:28,187 - INFO    - terrariumEngine      - Updated Enclosure Snake-1 with 2 areas in 0.10 seconds.
2021-09-29 12:43:28,185 - INFO    - terrariumEngine      - Updated Enclosure Snake-2 with 2 areas in 0.01 seconds.
2021-09-29 12:43:28,178 - INFO    - terrariumEngine      - Updated Enclosure Snake-1 with 2 areas in 0.01 seconds.
2021-09-29 12:43:28,092 - INFO    - terrariumArea        - Set the relay Sonoff (Tasmota) relay named 'Relay-2-3 (Lights-Snake-2)' at address 'http://192.168.86.92/#3' to 0.0 with 0 seconds delay
2021-09-29 12:43:28,091 - INFO    - terrariumArea        - Toggle the relays for area Heating named Heat-1 to state off.
2021-09-29 12:43:28,046 - INFO    - terrariumEngine      - Updated relay sonoff relay named 'Relay-2-2 (Heater-Snake-1)' at address 'http://192.168.86.92/#2' with new value 0.00 in 0.05 seconds.
2021-09-29 12:43:27,994 - INFO    - terrariumEngine      - Updated relay sonoff relay named 'Relay-2-3 (Lights-Snake-2)' at address 'http://192.168.86.92/#3' with new value 100.00 in 0.10 seconds.
2021-09-29 12:43:27,947 - INFO    - terrariumEngine      - Updated sensor remote temperature named 'Temp-Snake-1' at address 'http://192.168.86.92/cm?cmnd=sonoff/status%2010#StatusSNS/DS18B20-2/Temperature' with new value 41.00C in 0.10 seconds.
2021-09-29 12:43:27,895 - INFO    - terrariumEngine      - Updated relay sonoff relay named 'Relay-2-4 (Heater-Snake-2)' at address 'http://192.168.86.92/#4' with new value 0.00 in 0.10 seconds.
2021-09-29 12:43:27,847 - INFO    - terrariumEngine      - Updated sensor remote temperature named 'Temp-Snake-2' at address 'http://192.168.86.92/cm?cmnd=sonoff/status%2010#StatusSNS/DS18B20-1/Temperature' with new value 33.40C in 0.29 seconds.
2021-09-29 12:43:27,797 - INFO    - terrariumEngine      - Updated relay sonoff relay named 'Relay-2-1 (Lights-Snake-1)' at address 'http://192.168.86.92/#1' with new value 100.00 in 0.23 seconds.
2021-09-29 12:43:27,542 - INFO    - terrariumEngine      - Starting a new update round with 2 sensors, 4 relays, 0 buttons and 0 webcams.

Thanks

Andy

ewiandr73, 2021-09-29:
Seems to have stopped flickering after about an hour, very odd...

I'll keep and eye on it and let you know how things go.
let me know if you want me to test anything.

Thanks
Andy

theyosh, 2021-09-29:
Could there be another system/timer running? I have seen that some relays have their own schedule in it. Or an app is controlling the relays.

And I am not sure how much relays it can handle. The issue is how much time does it take to update a single relay. As my software is trying to update every 30 seconds. That means that all actions needs to happen within 30 seconds.

theyosh, 2021-09-29:
Hmm, is it possible to do some research? I am wondering if there is 1 url, to get the state of all 4 relays. 

So what is the output of the following: `http://[sonoff_ip]/cm?Status%200`

Does this give the information about all the relays on the device? Could you post the outcome here?

ewiandr73, 2021-09-29:
It's does.

**http://[sonoff_ip]/cm?cmnd=Status%200**

{"Status":{"Module":7,"DeviceName":"Tasmota","FriendlyName":["Heater","Lights","Water","Tasmota4"],"Topic":"SnakeRelays5_C9A4EC","ButtonTopic":"0","Power":8,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T17:30:48","StartupUTC":"2021-09-29T05:25:49","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2020-07-30T02:33:03","SaveCount":3189,"SaveAddress":"FB000"},"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"391/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WiFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","01000080","00006000","00000000"]},"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"SnakeRelays5_C9A4EC-1260","IPAddress":"192.168.86.91","Gateway":"192.168.86.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.86.1","Mac":"C4:4F:33:C9:A4:EC","Webserver":2,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C9A4EC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2021-09-29T22:56:37","Local":"2021-09-29T23:56:37","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:47","Sunset":"18:33"},"StatusSNS":{"Time":"2021-09-29T23:56:37","DS18B20":{"Id":"01143F9ABCAA","Temperature":20.8},"TempUnit":"C"},"StatusSTS":{"Time":"2021-09-29T23:56:37","Uptime":"0T17:30:48","UptimeSec":63048,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"_**POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON",**_"Wifi":{"AP":1,"SSId":"WiFi","BSSId":"E4:F0:42:F0:3C:DF","Channel":6,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":2,"Downtime":"0T00:00:10"}}}

This gives you the same info but without all the extra stuff.

**http://[sonoff_ip]/cm?cmnd=State**
{"Time":"2021-09-29T23:58:13","Uptime":"0T17:32:24","UptimeSec":63144,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":0,**_"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON"_**,"Wifi":{"AP":1,"SSId":"WiFi","BSSId":"E4:F0:42:F0:3C:DF","Channel":6,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":2,"Downtime":"0T00:00:10"}}

Also works on a dual

{"Time":"2021-09-30T09:01:26","Uptime":"0T01:19:46","UptimeSec":4786,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"WiFi","BSSId":"E4:F0:42:F0:37:96","Channel":1,"RSSI":24,"Signal":-88,"LinkCount":1,"Downtime":"0T00:00:07"}}

theyosh, 2021-09-30:
Cool, with this I can make some caching state code, so we only have to check 1 physical device at a a time. That will speed up things.

Give me 1 or 2 days to make this.

theyosh, 2021-09-30:
Ah, it was a bit easier.... So I made the change already. Update with `git pull` and restart.

But, before you do, please look at the logfile and search for the lines with `Updated relay sonoff relay named` and look at the time it took. I have about 0.10 seconds for the update.

But when you run the new code, I would expect that 3/4 of all the sonoff logfiles should contain updates with 0.00 seconds. As that is the cache effect. Every physical device should be polled once per 30 seconds, and will update all the relays states. So only 1/4 of the actual updates are needed, and those should take a bit more time.

theyosh, 2021-09-30:
So, another caching update. As I saw your temperature sensor urls, I start thinking.... 10 * 8 sensors is 80 calls. That is a bit much. So I have reduced that now to 10 calls.

Here I again get all the sensor data in 1 call. And store that in memory for 30 seconds. So other remote calls with the same host and url, will get the cached data, and will not get online data...

So, with those changes, I think using 10 Sonoff devices with 4 relays and 8 sensors each should work. Not sure if the interface will work well with that much...

theyosh, 2021-10-06:
Did you have any time to test? As far I can judge, it should work

ewiandr73, 2021-10-29:
I now have had 4 enclosure running flawlessly for about 3 weeks. 🥇 

Regards
Andy

theyosh, 2021-10-30:
Nice! Then I close this issue. Cool that it is all working.

closed 2021-10-30 by theyosh