theyosh/TerrariumPI #622: [BUG]: TerrariumPI 4 installation fails on RaspberryPI 4B

WarlordKorolov, 2021-11-10:
Hi,

I have been trying to setup a new Pi 4B with TerrariumPI 4 and the installation keeps failing. I have tried multiple installations on multiple SD cards on the same Pi4 hardware but keep getting the same errors. 

**Setup:**
 - TerrariumPI version: [4]
 - Python version: [ 3]
 - Raspberry PI: [ 4]

**Describe the bug**
When launching the installer on a clean version of Raspberry Pi OS Lite following the instructions as stated in the installation guide, the install process dies with the errors attached in the screen shots. I searched through the bug reports and generally on the web but found no instances of this issue being reported. 

**Screenshots**
[image]
[image]

**Additional context**
Any help you can give would be greatly appreciated.

theyosh, 2021-11-10:
Hmm, this is due to the fact Raspberry PI is now running on Debian bullseye. Apparently, that OS is different then buster on which I am currently running.

If you are a bit handy, you can create your own Pi image with an old OS: https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/ using `dd`

Or maybe the image writer can also load older of downloaded images that can be written to the SD card. But, for now, the fastest way is to install the old Raspberry PI image.

I hope to do some testing on Bullseye this week.

WarlordKorolov, 2021-11-10:
Awesome, that worked like a charm. Thanks!

closed 2021-11-10 by WarlordKorolov

theyosh, 2021-11-10:
Thanks for the feedback. Now I have to figure out what is the difference...