theyosh/TerrariumPI #627: Migrating from v3 to v4, instructions reference terrariumpi.db but doesn't exist?

Benjin, 2021-11-20:
I'm following the instructions [here](https://theyosh.github.io/TerrariumPI/install/#migrate) to migrate from v3 to v4 in order to test the fix for #231, but I'm not spotting the referenced terrariumpi.db.  I've renamed the old directory, cloned the new installation, and run `install.sh`.  Where should this be located?

alec-pinson, 2021-11-21:
try

`find / -name "*terrariumpi.db*" 2>/dev/null`

theyosh, 2021-11-21:
It could be that the documentation is not that clear. The file `terrariumpi.db` should be in the same folder as the code. But it will only be there, when TP4 is started at least once. Then the db is created. So after running the new `install.sh` you should reboot once. That should trigger the start of TP4, which will create the new database.

Also, with 4.1, with docker support, will change the database location, so I think I have to address that as well in the documentation.

theyosh, 2021-11-21:
And you have to configure the relays and sensors in TP4 first, before a migration can take place. So make sure TP4 is configured and running. Then shutdown TP4 and do the migration step.

Benjin, 2021-11-21:
I see; yeah, I think the documentation could be improved a bit?  Things that might be worth calling out explicitly:

* After TP3 is backed up and renamed, do a normal, complete installation of TP4 (to the point that the web UI is live)
* Relays/sensors must be configured _before_ the migration
  * Why can't this part of the migration step?  `copy_relay_settings.py` clearly has the core info: name, type of relay, address.
  * What is it migrating if not the name/type/address; just history?  Since the script name includes "settings", I figured those would be included since they're the main settings.

In any case, I've got it running now.  Feel free to close this issue, or perhaps leave it open to track documentation updates?

theyosh, 2021-12-06:
I updated the documentation a bit. But I think it is now clear that I only copy the relay history.
https://github.com/theyosh/TerrariumPI/commit/31a97e3c1a7338b24e78a1b2ff4834e53fa1e70f

closed 2021-12-06 by theyosh