theyosh/TerrariumPI #628: [FEATURE]: "Better" IRF520 dimmer calibration (brightness)

Jexkin, 2021-11-22:
**Is your feature request related to a problem? Please describe.**
The brightness at a low % is (in my opinion/case) a bit too much. Not sure if it really counts as a problem but I figured I'll just type it here. If this feature request is not worth it, then feel free to close it, I won't be mad about it! :) 

**Describe the solution you'd like**
I'd like to have a better calibration for the (IRF520) dimmer module. I personally swapped the IRF520 mosfet for a IRF3708 mosfet because my LED strip did not light up with the IRF520 mosfet. But I feel like the light output/brightness is too much in the low % range. Would it be possible to add e.g. a decimal point to the % or something like that to achieve a "wider" amount of brightness? 
Or anything else that would make the LEDs not as bright in the lower % range?
If that's possible, of course.

**Additional context**
I swapped the IRF520 mosfet for a IRF3708 mosfet (just want to note it here as well).
I'm using a 12V 5054 LED strip, and I currently have the "Max Power" set to 30%.
In general the dimming itself works perfectly fine, no problems in that regards. (Thank you for the project btw!)

theyosh, 2021-11-22:
Hi,

I was hoping you could change the 'Dimmer frequency in Hz' and that would fix it. But I do not know what works best. Also, I am not sure if you need to restart after changing the Dimmer frequency in Hz value. As I do change it in code, I got the feeling it needs a restart.

So, start with low value of 1000 hz, save that and restart TP4. And see what happens. Do not go higher then 10000Hz. Not sure if you could fry your device with to high frequency.

Jexkin, 2021-11-22:
Hey,

you are right! I was able to achieve a lower brightness by adjusting the dimmer frequency.
I had it set to 1000Hz the whole time because it did not work with the preset value of 10000Hz.
So far I was able to see a drop in brightness when I changed it from 1000Hz to 3000Hz, at 6000Hz and at 9000Hz (the higher the value, the lower the brightness at the same %). 
There was no change in brightness in between all of those values (2000Hz, 4000Hz etc.).
The change of brightness is fairly minor though, comparing 1000Hz to 9000Hz right now.

As a side note: I did not have to restart TP4, it worked flawlessly by just saving the value. 
Also, 10000Hz is the max value possible and I did not see any brightness change at that value either.

I'm honestly not sure if the decrease in brightness is enough though, as it's just fairly minor. 
Could you think of any other possible calibration?
If not, then that's no problem!

theyosh, 2021-11-24:
Hmm, I think what you can do is the max that I can do with software. The only thing I can think of is to check with some external software if you can dim your leds enough. https://www.waveformlighting.com/home-residential/how-to-dim-led-strip-lights
But that is pretty expensive: https://store.waveformlighting.com/collections/power-supplies-and-dimmers/products/filmgrade-flicker-free-dimmer-for-led-strip

And they also mention that the dimmer should be close to the light strip. And use a longer power and data cable.

So maybe it is not possible with cheap hardware to get the dimming right.

And I have also that the dimming of a normal light is not 100% accurate. But for me it is enough compared to the expensive commercial products.

Jexkin, 2021-11-24:
Before I tried the dimming with TerrariumPi, I tried it with ESPEasy. If I can remember correctly, they let you set a value between 0 and 1023. (0 would be off/0%, 1023 would be 100%)
The brightness range therefore was better in the low %/value range.

That's why I asked if it might be possible to add a decimal to the % value, so the value in theory goes from 0 to 1000 instead of 0 to 100?

Please excuse me if that's not possible, I have basically zero coding knowledge.

theyosh, 2021-11-24:
No, there are not a lot of options here. But what you describe looks more like: https://github.com/codmpm/pca9685-net-udp

Which is a PCA9685-dimmer in my software. Originally it is used for servo motors. But you can also use it for dimming leds. Maybe going that route will work? Else I am out of options.

theyosh, 2021-11-24:
And using the GUI there are always 100 steps. From 0 - 100. And I translate that back to the valid values. But that does us not allow more steps or values like 0.5% and 0.7% dimming. With the API it could be done, but for that, you need some python scripting knowledge.

Jexkin, 2021-11-24:
I'd like to try the PCA9685-dimmer option but I currently have issues with TerrariumPi.
I did a `git pull` yesterday noon and ever since then I can't access the dashboard. I probably shouldn't have done that.
`journalctl -xe` gives me the following output:

`Nov 24 18:00:04 TerrariumPi python[843]: Traceback (most recent call last):`
`Nov 24 18:00:04 TerrariumPi python[843]:   File "/home/pi/TerrariumPI/terrariumPI.py", line 4, in <module>`
`Nov 24 18:00:04 TerrariumPi python[843]:     from gevent import monkey`
`Nov 24 18:00:04 TerrariumPi python[843]: ModuleNotFoundError: No module named 'gevent'`
`Nov 24 18:00:04 TerrariumPi systemd[1]: terrariumpi.service: Main process exited, code=exited, status=1/FAILURE`

Already tried to reinstall TerrariumPi but that didn't solve it, probably because it pulled the most recent update.

theyosh, 2021-11-24:
Hmm, I changed the python libraries. Some updates are not working nice. So do a `git pull` and run the installer once more. I hope that would solve it

Jexkin, 2021-11-24:
I just did a `git pull` and ran the installer, rebooted the Pi but I still get the exact same output. The only difference is that the number after `python` changed from `[843]` to `[782]`

theyosh, 2021-11-24:
Hmm, there is a folder called `venv` inside TerrariumPI. If you delete that folder and rerun the installer, maybe that would solve it. This way the installer will install all python libraries again from fresh. If that does not help, I do not know what is happening.

As I did a reinstall 1 hour ago, and that worked with the latest code.

Jexkin, 2021-11-24:
Sounds like it's a problem on my side. I already did multiple fresh installs (flashing the micro SD card, then following your guide) but it seems like even though it's a fresh install, it doesn't really properly install? The very first install took me about 20-30 minutes or even more. Now all the other installs take only 1-2 minutes, it seems like it doesn't really do fresh installs and rather tries to do updates.
I already deleted the pi from my network in case something is up with it but it seems like that didn't help.

I'll try to figure out what's wrong and once I can properly run TerrariumPi again I'll test the dimming with PCA9685 and share my results!

Jexkin, 2021-11-29:
That's exactly what I did, downloading the latest Raspbian Image/using the Raspberry Pi Imager. I now tried to set everything up with this lite image:

https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/

But I have some problems getting it to run. Which image did you use?

JoshuaR1337, 2021-11-29:
@fujexo and @Jexkin https://github.com/theyosh/TerrariumPI/issues/623#issuecomment-965911172

fujexo, 2021-11-29:
I think I downloaded the same image as you did. After `sudo ./install.sh`, TerrariumPi started just fine.
Have you tried `sudo apt update && sudo apt full-upgrade` before installing TerrariumPi? Maybe this makes a difference?

Are you certain that you installed the right branch? On installation, I accidentally copied the commands from the master branch README which tried to install TP3 instead of TP4, took some minutes until I figured that out :innocent:

Jexkin, 2021-11-29:
Just flashed the version from #622 onto the pi and everything works like a charm now!
Thanks y'all!

I just ordered a PCA9685 and I'll try if it's better than the IRF520 module.
As I'm unsure how long it'll take, feel free to close this issue.

Jexkin, 2021-12-04:
So, I got the PCA9685 module yesterday and I just tried to hook it up but it doesn't seem to work correctly.

Everything should be hooked up properly. I'll post the log entries, maybe it's just me doing something wrong.

`2021-12-04 13:33:15,950 - WARNING - terrariumEngine       - Could not take a new measurement from relay pca9685-dimmer dimmer named 'Test' at address '0x40'. Tried for 1.61 seconds. Skipping this update.`
`hardware.relay.terrariumRelayUpdateException: Error getting new data from relay PCA9685-dimmer dimmer named 'Test' at address '0x40'. Error: unknown`
`raise terrariumRelayUpdateException(f'Error getting new data from relay {self}. Error: unknown')`
`File "/home/pi/TerrariumPI/hardware/relay/__init__.py", line 162, in __get_hardware_value`
`File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/retry/api.py", line 33, in __retry_internal`
`File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/retry/api.py", line 74, in retry_decorator`
`return caller(func, *(extras + args), **kw)`
`File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/decorator.py", line 232, in fun`
`new_data = self.__get_hardware_value()`
`File "/home/pi/TerrariumPI/hardware/relay/__init__.py", line 240, in update`
`2021-12-04 13:33:15,948 - ERROR   - hardware.relay        - Error getting new data from relay PCA9685-dimmer dimmer named 'Test' at address '0x40'. Error: unknown`
`2021-12-04 13:33:15,945 - ERROR   - hardware.relay        - Error getting new data from relay PCA9685-dimmer dimmer named 'Test' at address '0x40'. Error: '<' not supported between instances of 'NoneType' and 'int'`
`2021-12-04 13:33:15,441 - WARNING - hardware.relay        - Error getting new data from relay PCA9685-dimmer dimmer named 'Test' at address '0x40'. Error: unknown, retrying in 0.5 seconds...`

Those are the settings I'm trying to use. (unsure of the IC2 address but that's what I found)

[image]

theyosh, 2021-12-05:
Hi, it turns out the code was wrong, and you need to change the address.

The current address you have is the I2C address. That is now optional. As  I understand correctly, you have 8 connections, so you need to specify a relay number also. Not sure if they start with 0 as first or 1 as first.

So, the correct address will be `[Relay number],[I2C address]` where the I2C address is now optional. When omitted, I use 0.x40 as i2c address as that seems to be the default.

So, do a `git pull` and restart. Try to update the dimmer address and lets hope all code is correct now.

Jexkin, 2021-12-05:
Hi, in theory this should be correct, right?

[image]

The PCA9685 has 16 LED drivers/connections, labeled 0-15. I assume with `[Relay number]`, that is what you meant, correct?

This is the output I get when I click save:

`2021-12-05 14:34:13,145 - WARNING - hardware.relay        - Could not load hardware for relay PCA9685-dimmer dimmer named 'Test' at address '0,0x40': invalid literal for int() with base 10: '0x40', retrying in 0.5 seconds...`

I also tried `0,x40` and `0.x40` but both didn't work either.
Just gonna post the logs for both of those as well:
`2021-12-05 14:36:01,555 - WARNING - hardware.relay        - Could not load hardware for relay PCA9685-dimmer dimmer named 'Test' at address '0,x40': invalid literal for int() with base 16: '0xx40', retrying in 0.5 seconds...`

`2021-12-05 14:36:16,398 - WARNING - hardware.relay        - Could not load hardware for relay PCA9685-dimmer dimmer named 'Test' at address '0.x40': invalid literal for int() with base 10: '0.x40', retrying in 0.5 seconds...`

theyosh, 2021-12-05:
Crap, small type in the code. Do another `git pull` and I think it should work

Jexkin, 2021-12-05:
I can save the PCA9685 now, thanks!

I still get a weird bug though. Whenever I set the % to any % it says `Changed relay Test to 100%`, no matter if I set it to 10% or 90% or whatever. The LED changes to whatever % I set it, so I guess it's just a UI bug.

So far I hooked up the IRF520 module to the PCA9586 module and the brightness difference is barely visible compared to just the IRF520/IRF3708 module.
 
I also just tried to drive the LED strip with just the PCA9685 alone but that didn't work at all (connecting the LED strip to PWM and GND of the PCA module). Already tried to change the Frequency within 100Hz to 10000Hz but nothing worked.
If I connect the LED strip to the V+ and GND of the PCA module it just goes to a certain brightness, which doesn't change if I change the % value.

theyosh, 2021-12-05:
Hmm, I made a small update which I hope will fix the GUI so that it shows the right value.

But then, this is what it is...  I am sorry if this is not solving your dimming wish. And I hope it did not costs to much. As I do not have every hardware device at home, and I am not able to test all kind of hardware and dimming setups.

But I want to thank you for at least testing this dimmer, so I know that it works with TP4. And maybe you can use it somewhere else...

Jexkin, 2021-12-05:
Thank you for your efforts, I really appreciate it!

The small update somewhat works. The small notification box now outputs the correct % that I set it to but the bar still jumps from the set % to 100% after some time.

As you have [this](https://custom-build-robots.com/electronic/raspberry-pi-led-dimmer-pca9685-servo-controller/8840?lang=en#prettyPhoto) guide linked on the PCA9685 page, I'll just buy one (or five) of those L298N modules and connect it to the PCA9685. Maybe that will work somehow. If not then I'll just use the IRF520 module and be happy with the dimming that I can get! And thanks again for developing TP :)

theyosh, 2021-12-06:
Hmm, strange, I expected to be fixed with the code on line https://github.com/theyosh/TerrariumPI/blob/4.x.y.z/hardware/relay/pca9685-dimmer_relay.py#L38

Are you a bit handy with code? Then it would help me if you can try to see what value you will get from `self._device['device'].get_pwm(self._device['switch'])` so that we can calculate the dimmer value back to a value between 0 - 100.

Or, if you can and willing to give me SSH access to your PI though port forwarding, then I can debug it myself. If so, send login details to: github.com@theyosh.nl

Jexkin, 2021-12-06:
Hi, I managed to log some values

0 % - 0
10% - 409
20% - 819
30% - 1228
40% - 1638
50% - 2047
60% - 2457
70% - 2866
80% - 3276
90% - 3685
100% - 4095

Every time TP updated the bar jumped back to 100%, the little notification box did say the correct value every time though.

Hope this helps.

theyosh, 2021-12-06:
Hmm, that was a bit predictable seeing the list now. So I updated the code once more, and I hope it is working now correctly. Do a `git pull` and restart. Lets hope this will fix it

Jexkin, 2021-12-06:
The updated code definitely did something, thanks!
It doesn't jump to 100% on every update anymore but the green notification box says the value that was previously set.
E.g. I set it from 100% to 10% and it'd say 100% instead of 10%. I then set the value from 10% to 20% and the box said 10% (pic).

[image]

theyosh, 2021-12-07:
Hmm, found that small bug also when putting to 0% and 100%. Do a `git pull` and restart. That should be fixed.

About the notification message, it is a bit strange. Does the other dimmer have the same issue? And are you using a RaspberryPI 4or 3?

Jexkin, 2021-12-07:
Before I did a `git pull` I tested if I have the same issue regarding the notification with the other relays and yes.
But the `git pull` seemed to have fixed both of the issues!
No more errors when I try to set it to 0% or 100% and the notification message now says the correct % value.

Oh, and currently I'm using a Pi3 for all my testing but once everything is done I'll move/install TP on my Pi4, I hope everything will work as usual then.

theyosh, 2021-12-07:
Hmm, I am afraid that my code did not fix the notification. What I have experienced with a RPi3, with more database actions now, it sometimes is a bit slow with updating the relays. So the toggle is done, relays is changed state, but the update to the database is a bit slow. And therefore, it readouts the old value and give that back to the notification.

On a RPi4 I have not seen this anymore. As the RPi4 has a faster CPU and therefore there is no delay with the database. I will see somewhere next year how to properly fix this.

theyosh, 2021-12-07:
But nice that the 0% and 100% are correct now. Thanks for testing and debugging.

Jexkin, 2021-12-07:
So far the notifications didn't give me any wrong readouts. In the first few days of testing I didn't notice any wrong readouts either, it began after testing the PCA9685 module. For me at least it seemed fixed but maybe some other people still have that issue.

As for now, thank you for all the fixes! I'll try to order and test the L298N modules until the end of next week. My plan so far is to just hook up the L298N module to the PCA9685 dimmer module and test everything. So yeah, I'll keep you updated on that one! Thanks!

Jexkin, 2021-12-10:
Alright, I just tested the L298N modules but I can't control them with the PCA9586 in any way. The LEDs are just permanently on. It was worth a try.

So I guess it's not really possible to get a wider brightness range with different hardware. 10% will be 10% regardless if I use the IRF520 module or a combination of the PCA9685 and IRF520 module. Not sure if there's anything else I can try. So yeah, feel free to close this issue unless you can think of something else I can try :)

If you decide to close the issue then I'll say thank you for your efforts!

theyosh, 2021-12-10:
Hmm, that is too bad.. You have tried all my possible dimmers. And I do not have any other solutions right now.

The only thing I can say, if you find some hardware that you can control with some code on your PI, please open a new issue with that info. Then we can see if we can incorporate that into TP4.

Thanks for testing this out and also finding some bugs :) To bad we could not make it work...

closed 2021-12-10 by theyosh