theyosh/TerrariumPI #629: [Question] Timer functionality

fujexo, 2021-11-24:
Good evening!

Maybe I didn't find it in the documentation, maybe I'm just not understanding the area settings.
I'm trying to setup a timer which works the following way:
* Turn on for 15 minutes
* Turn off for 45 minutes

This should happen the whole day and night, but only during the day would be fine too. Would be great if someone can help me to set this up.

I would like to add this to the documentation too (if it isn't there already and I missed it), maybe we could add a "often used configurations" page or something like that?

fujexo, 2021-11-25:
Another thing I want to do: Every evening, after the lights have turned of, let the rain pump relay run for 30 seconds or so. Is that also possible?

theyosh, 2021-12-01:
Hi,

the timer documentation is not yet finished :( https://theyosh.github.io/TerrariumPI/setup/#areas I need to describe the forms, but that is a lot of work. But if you want to contribute that is possible: https://github.com/theyosh/TerrariumPI/tree/4.x.y.z/docs

Also I have already a dummy page at: https://theyosh.github.io/TerrariumPI/scenarios/ where I guess this kind of setups should be described.

[image]
I think the above example should be what you asked. 

Set mode to timer. Then you enter the operating window (Begin and End time). Within this period, you have 15 minutes the power of, and off for 45 minutes. And then on again etc until the End time is passed. By entering the values 0 at Power on time and Settle time.

If you want to change this to only during the day, you can select the light status option to be on. And at the light source option, you can either select weather (based) or on the main lights of your enclosure.

Then about the sprayer after the lights have turned off. That could a bit tricky to make. What is the setup of your main lights area. Do you use a dimmer? Because that could make it a bit harder to do. Because when the lights go off, they are set to off, while the dimmer is still running. So if you have a dimmer setup of 30 min to go off, you have to take that in account with the spray timer....

fujexo, 2021-12-02:
Hi Yosh!

Thanks, the timer works now, just as you described :tada: 
[image]

For my lighting setup, I just use two relays (no dimmer) with a little bit of delay, only 10 minutes or so:

[image]
[image]

Now that I know how the timers are working, I've tried my luck. But it doesn't turn on.

[image]

PS: Sorry for the language of the Webinterface, forgot to switch it back to english..

theyosh, 2021-12-02:
You mean the last option. Rain after lights go out.... 

But, for your lights, I suggest to toggle the 'Hauptlichter' to on. This will make sure the lights are run as first area per run. I think it is logical to make the lights run first, as it is used for the 'Licht Status' check. So it could be that this will also fix the rain option when the lights go out. Not sure how this react at the moment

fujexo, 2021-12-04:
Hi Yosh, thanks for your feedback. I was testing the setup yesterday with the following settings applied:
[image]
[image]

If I got that right, if the lights go our after 4 PM, then the rain pump should power on for 30 seconds and then turn off until the next day starts. I don't know why the rainpump didn't turn on, had to toggle it manually.

Might it be that it interfers with my other area which has the same relay configured? I use it to keep the humidity at a decent level based on the sensor readings.
[image]

Is it possible to configure sensor and timer mode in the same area?

theyosh, 2021-12-05:
Hmm, using the same relay in two areas will not work. 

Also I see a disabled mode in the second screenshot. So that area is not operational when you select disabled. 

And it is possible to select the timer mode and use sensors as well. But that will not help your case. As that would trigger only the relays when the timer is true, AND a low alarm on the sensors. And at my terrarium, the humidity rises when the lights go out. So that will never hit the low alarm of my sensors.

But, there is maybe another way. What kind of hardware is your Regenpumpe?

fujexo, 2021-12-06:
I made sure to test it in timer mode. But I disabled it afterwards as I suspected it could interfere with the other Area as it has the same relay configured.

I'm using a GPIO relay to turn on the rain pump.

theyosh, 2021-12-22:
Sorry for the late reply. There is a trick here..

If you make 2 python scripts that both can toggle the same relay (GPIO pin) outside TP4, then you could add two `script` type relays based on the two python scripts. 
Because with the script, you can instruct TP4 not to readout the state, and 'trust' on the database state.

That means when 1 script relay will toggle the relay, the other script relay will not change. So then you can control the same hardware relay, with two 'virtual' relays in TP4.

But this is not tested, and I do not know what the side effects could be.. There is an example script at: https://github.com/theyosh/TerrariumPI/blob/main/contrib/external_switch.py

You will have to add some code to toggle the GPIO pin at line 37 and 44.

theyosh, 2022-01-06:
Is the above clear? And did you had time to test this?

fujexo, 2022-01-09:
Had no yet time to test this. The instructions are clear, once I got it sorted and found the time, I'll report back.

closed 2022-02-25 by theyosh

kaju666, 2025-03-16:
@theyosh 
Dont want to make a new thread, but i cant find out how to set replay to power on for 10 seconds two times per day.
I want to run mister on the morning and midnight and run it for 10 seconds.
Can you help please :)?

[image]

theyosh, 2025-03-16:
> [@theyosh](https://github.com/theyosh) Dont want to make a new thread, but i cant find out how to set relay to power on for 10 seconds two times per day. I want to run mister on the morning and midnight and run it for 10 seconds. Can you help please :)?
> 

Running in the morning and mid night is a bit hard, due to day change. So what you can do is using a begin time of 00:01 which is one minute over midnight. And an end time of 09:00 hours in the morning.

Next set the on duration on 1 (minute) and the off duration of 535 minutes. This will allow the relay to be on 00:01 and 08:57 for 1 minute.

So than set the Power on time to 10 seconds, and the settle time to 60 seconds.

This should trigger the relay for 10 seconds on 00:01 and around 08:57.

https://theyosh.github.io/TerrariumPI/scenarios/#timer-wizard

[image]

theyosh, 2025-03-16:
The on time and settle time should match or be greater than the on period. So 50 should work, but just to be sure, I added an extra 10 seconds. I am not sure how precise the timing is, so adding extra time is just for extra security that it will only run once.