theyosh/TerrariumPI #630: [Question] Disable relay based on second sensor

fujexo, 2021-11-24:
Good evening!

My current setup has a rain pump relay. I'm turning the pump on based on low humidity readings from a BME280 sensor.
I've got a 10 liters water tank which I'm monitoring using an HC-SR04 sensor. I've already setup a water tank area which displays the current water level.
My goal is to create some area or sensor which disables the rain pump relay when the tank is almost empty to prevent it from running dry.  I didn't find anything like that in the documentation.

[image]

Would be great if someone can help me to set this up.

I would like to add this to the documentation too (if it isn't there already and I missed it), maybe we could add a "often used configurations" page or something like that?

theyosh, 2021-12-01:
Hi,

at this time, this is not possible. If I understand correctly, you want the sprayer not toggle on when there is not enough water in the water tank. That sounds a bit like: #249 But in a different way. The heater should not go in, when there is not enough water.

So I am thinking that we should make areas depend on each other. That way, you can setup the watertank you have right now. With the distance sensor alarm settings, you can create 'alarm' states on the water tank when almost empty, or to full.

If we use that state with, in your case, the sprayer area (humidity) will not toggle on when the watertank is in an alarm state. That should prevent from running dry.

About the documentation.. I still need to make a lot, but that takes more time than I expected. And also, writing in not my native language is sometimes a bit hard. But, if you want, you can contribute to it: https://github.com/theyosh/TerrariumPI/tree/4.x.y.z/docs

fujexo, 2021-12-02:
Hi Yosh!

Thanks for the feedback!
No problem if it isn't implemented, that would just be a great feature. For now, I've worked around the water tank level with a script relay which turns a LED cube on which I built some years back. That should give enough indication to refill the tank :)

[image]
(I'll post some images to the right issue once I've rebuilt my case - not yet happy with it).

For the documentation, I'll try to to some updates if I get some spare time. I really like your project so that's a good time investment :)

theyosh, 2021-12-06:
So, I was just testing area dependencies and it looks like it works nicely. Now you can add the water tank area as a dependency for the humidity area. And then the water tank is in alarm mode, the sprayer will not operate. This should prevent running dry.

After a quick test, it looks OK!. So if you want, you could give this a try. It will not produce warning messages (yet). As I do not know yet how often this will happen...

Do a `git pull` and restart. You should see a new setting in the area forms.

fujexo, 2021-12-06:
This is huge :+1:
I've just updated my instance, works exactly as I've wanted. Doesn't turn on the rain pump relay when there is not enough water in the tank. After filling it up, it turns it on automatically.

For testing, I've just configured the sensor alarms to simulate the case. But that should reflect the real thing good enough :)

theyosh, 2021-12-10:
Closing this issue, as with the new [release 4.1.0](https://github.com/theyosh/TerrariumPI/releases/tag/4.1.0) this is now possible.

closed 2021-12-10 by theyosh