theyosh/TerrariumPI #631: [BUG]: LCD Display isn't working

kebs19, 2021-11-26:
Hello!
I'm trying to put the LCD Display to work, but I can't make it..
First some background:

The LCD is correctly connected since i2c can detect it at 0x23:

i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- 23 -- -- -- -- -- -- -- -- -- -- -- --
(...) 

Using a Rasp Zero W, Raspbian buster:

 cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"

When I add a LCD in the notifications setup, I used the address "23" and "0x23", but the lcd still shows a line full of squares. 

I tried to turn on some notifications (I don't know what to fill in those fields) and tried to turn on the 'show name' to get the display to at least show date/time.

Is there a step I'm missing?

theyosh, 2021-11-28:
Hi,

for TP4 the display is not working yet. I need to fix that. And the notification system is also not yet fully developed. I hope to fix that in the next month.

theyosh, 2021-12-22:
Hi, I think I fixed the code, so that it should work... Only the placeholders data is not complete (UI).

But, you should now be able to add a display. And when you have added one, you should at least some startup and shutdown messages on your display.

The code is currently in the **4.x.y.z branch**.

kebs19, 2021-12-23:
I tried a git pull and then reinstall  but now I'm getting fatal errors... when I try to debug the system I get the following:
pi@terrariumpi4:~ $ sudo service terrariumpi restart
Job for terrariumpi.service failed because a fatal signal was delivered to the control process.
See "systemctl status terrariumpi.service" and "journalctl -xe" for details.

pi@terrariumpi4:~/TerrariumPI $ source venv/bin/activate
(venv) pi@terrariumpi4:~/TerrariumPI $ python terrariumPI.py
Illegal instruction

The worst part is that the system isn't working anymore :(

theyosh, 2021-12-23:
> (venv) pi@terrariumpi4:~ /TerrariumPI $ sudo python terrariumPI.py

Do this without sudo. https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

If that does not help, please rerun the installer: `sudo ./install.sh`

But I hope when you run it without sudo it should work. If that still produces an ` ImportError: No module named gevent` then a rerun of the installer is needed.

theyosh, 2021-12-23:
Crap, maybe my fault. I have added new hardware so there are new modules needed. So a rerun of the installer is needed. That should fix it.

Maybe a rule of thumb to remember: When after a `git pull` there are errors, just rerun the installer `sudo ./install.sh` That is 99% the fix.

kebs19, 2021-12-24:
hey there :) 

Nope, I tried to re-run the installer (sudo ./install.sh) but the error continues. it says the following:

pi@terrariumpi4:~ $ cd /home/pi/TerrariumPI/
pi@terrariumpi4:~ /TerrariumPI $ sudo service terrariumpi stop
pi@terrariumpi4:~ /TerrariumPI $ source venv/bin/activate
(venv) pi@terrariumpi4:~/TerrariumPI $ python terrariumPI.py
Illegal instruction

I tried to:
sudo pip3 install gevent

but it says:
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: gevent in /usr/lib/python3/dist-packages (1.3.7)
Requirement already satisfied: greenlet>=0.4.14 in /usr/lib/python3/dist-packages (from gevent) (0.4.15)

(I don't know if it helps somehow)
Thanks for your support :)

theyosh, 2021-12-24:
Hmm this is strange... With TP3 this is indeed the correct way of fixing. But for TP4 it is a bit different.

The ` Illegal instruction` message looks strange.... not seen it before, or only when there where more problems with the PI.

So, the last thing you could, is to stop TP4. Then move the `venv` folder to something else like `venv.old` and then rerun the installer. The venv folder is inside TerrariumPI folder. By renaming it, you can always go back if the new install did not fix it.

All the python packages are installed in the venv folder. By removing it and then reinstalling will enforce to install all packages again from online. As the old packages are gone.

When the venv folder is not removed, the installer will only update packages that are newer. So that is not a full install then.

Now you have installed something system wide, which I would not expect to work.

Also, do a git pull again, as I have made again new updates for the notification part.

Summary:
1. Stop TP4: `sudo service terrariumpi stop`
2. Enter the folder: `cd /home/pi/TerrariumPI` 
3. Move venv folder: `mv venv venv.old`
4. Rerun installer: `sudo ./install.sh`

If this does not fix it.... then it could be wise to start preparing a new SD card and make backup of this one.... get some strange feeling your SD card is start to fail...

kebs19, 2021-12-24:
Hello,

Now I tried two things:
To reinstall TPI on my previous system, but with the same results, and
I tried to install TPI in a different Rasp 4, with a different microsd, following the instructions from the site, as it was a completely new setup.

It didn't work, and when I try to debug, here's what I got:

pi@terrariumpi:~ $ sudo service terrariumpi stop
pi@terrariumpi:~ $ cd /home/pi/TerrariumPI/
pi@terrariumpi:~ /TerrariumPI $ source venv/bin/activate
-bash: venv/bin/activate: No such file or directory
pi@terrariumpi:~ /TerrariumPI $ python terrariumPI.py
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 4, in <module>
    from gevent import monkey
ModuleNotFoundError: No module named 'gevent'

thanks for your kind support and merry christmas :D

theyosh, 2021-12-24:
OS Bullseye is not supported. You need to use the older Buster Lite image: https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy (Raspberry Pi OS Lite (Legacy))

theyosh, 2021-12-24:
Hmm, So you are now also testing on a different PI, so your animals stay happy? ;)

kebs19, 2021-12-24:
> 

actually i'm back to basics: using a mechanical timer and the automatic heater. since it's summer here in Brasil, the weather is quite warm so the systems are 'kinda' working offline. 
I'm making a Rasp OS Buster Lite image to try. I'll send you a message in a couple of minutes :)

kebs19, 2021-12-25:
Hello, what I found out so far:
The issue is with the Rasp zero W. I can swap a non-working card in a rasp zero w and put it into a rasp 4 and it works nicely. 
Even if I setup a card on a rasp4 (and it works there) when I swap it to the rasp zero w it stops working. 
The funny thing is that the card that was working on the rasp zero (and stopped working) starts to work nicely when I use it on the rasp 4 (preserving my switches, DB and history from my previous system).
It seems that the problem is related to using Rasp Zero W, which is awkward cause I was using it with TerrariumPI 4.0.0 with no issues

kebs19, 2021-12-26:
I give up, I tried everything I could possibly know, even trying to pull the 4.000 version from github, but it just doesn't work at all. Tried deleting venv folder, everything.
The only thinkg I didn't try was restore a working instalation, since I did the git pull before backing things up. 
The chip shortage is hitting hard on us and a Rasp 3 is priced at over USD120, which is completely outrageous for a 35 bucks SBC

theyosh, 2021-12-27:
Damm, that is strange... But there is still another solution.

You could use Docker to run TP4. https://theyosh.github.io/TerrariumPI/install/#docker

That could make it work again. But not sure if this will work on a Raspberry PI zero.

I have not tested my software with Zero W. I can try to see if I can get it to work. But this is a strange issue..... I wish I could help more.

kebs19, 2021-12-27:
Hey there :) 

I tried to install using the docker image, and it says:

pi@terrariumpi:~ $ sudo docker-compose up -d
Pulling terrariumpi (theyosh/terrariumpi:4.1.0)...
4.1.0: Pulling from theyosh/terrariumpi
2aa85085c988: Pull complete
9b32ab3e4a3c: Pull complete
c60dd621177a: Pull complete
0819ba4e41da: Pull complete
971662f95f9f: Pull complete
6d114548c736: Pull complete
01d4405280cb: Pull complete
214cf8e7f53a: Pull complete
8e80bdc19a74: Pull complete
8f28337eba2e: Pull complete
3cef3ce7f2b2: Pull complete
221e86449af1: Pull complete
3a613edb2f61: Pull complete
cc6d862cf093: Pull complete
57fa285e325b: Pull complete
Digest: sha256:a95ec7223d5a190a906e655ea1754c634132515385359ac11055c1129122b0d7
Status: Downloaded newer image for theyosh/terrariumpi:4.1.0
Creating pi_terrariumpi_1 ... done

But when I try to access the web interface the system doesn't load. How can I debug the system installed with docker? I'm not very good when it comes to programming languages...

theyosh, 2021-12-27:
Hmm, ok, now it is getting a bit complicated... Because I think I found the original problem with the Pi Zero. There are two libraries that do not work in the Pi Zero and causing the crashes.

So I would first try that, before we are going to debug Docker. As I do not know much about Docker, it will be a very long way, which I hope to skip with by using the original installation again.

So, if you have installed TP4 on your Pi Zero at the manual way, using the `sudo ./install.sh` command, then you can do the following to make it work. After installation, do the following steps:

1. Become root: `sudo -i`
2. Enter TP4 folder: `cd /home/pi/TerrariumPI`
3. Load Python3 evironment **as root**: `source venv/bin/activate`
4. Replace Python libraries: `pip install numpy==1.21.4 lxml==4.6.4`

After this, the software should be working again. Start with `sudo service terrariumpi start`

And a quick Docker tip:
run the following command where the docker-compose.yaml file is: `docker-compos logs` should give some output. If not, then it still could be that the docker image has the same Python issue as running normally. Then make sure you have run `docker-compose down` so it will not start after a reboot.

kebs19, 2021-12-27:
hello, I tried the steps you told me, and had the following results:

pi@terrariumpi:~ $ sudo -i
root@terrariumpi:~# cd /home/pi/TerrariumPI
root@terrariumpi:/home/pi/TerrariumPI# source venv/bin/activate
(venv) root@terrariumpi:/home/pi/TerrariumPI# pip install numpy==1.21.4 lxml==4.6.4
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting numpy==1.21.4
  Downloading https://www.piwheels.org/simple/numpy/numpy-1.21.4-cp37-cp37m-linux_armv6l.whl (12.3 MB)
     |████████████████████████████████| 12.3 MB 11 kB/s
Collecting lxml==4.6.4
  Downloading https://www.piwheels.org/simple/lxml/lxml-4.6.4-cp37-cp37m-linux_armv6l.whl (6.0 MB)
     |████████████████████████████████| 6.0 MB 17 kB/s
Installing collected packages: numpy, lxml
  Attempting uninstall: numpy
    Found existing installation: numpy 1.21.5
    Uninstalling numpy-1.21.5:
      Successfully uninstalled numpy-1.21.5
  Attempting uninstall: lxml
    Found existing installation: lxml 4.7.1
    Uninstalling lxml-4.7.1:
      Successfully uninstalled lxml-4.7.1
Successfully installed lxml-4.6.4 numpy-1.21.4

pi@terrariumpi:~ $ sudo service terrariumpi start
Job for terrariumpi.service failed because a fatal signal was delivered to the control process.

and when I try to debug the installation, I have the same error:

pi@terrariumpi:~ $ cd /home/pi/TerrariumPI/
pi@terrariumpi:~/TerrariumPI $ source venv/bin/activate
(venv) pi@terrariumpi:~/TerrariumPI $ python terrariumPI.py
Illegal instruction

kebs19, 2021-12-28:
I just reformated the sd card and started all over again, installed RPI and when I try to install the named versions of numpy and lxml I have the following result:

root@terrariumpi: /home/pi/TerrariumPI# source venv/bin/activate
(venv) root@terrariumpi: /home/pi/TerrariumPI# pip install numpy==1.21.4 lxml==4.6.4
Looking in indexes: https: //pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: numpy==1.21.4 in ./venv/lib/python3.7/site-packages (1.21.4)
Requirement already satisfied: lxml==4.6.4 in ./venv/lib/python3.7/site-packages (4.6.4)

But the system have the same error:

(venv) root@terrariumpi: /home/pi/TerrariumPI# python terrariumPI.py
Illegal instruction

theyosh, 2021-12-28:
I have updated the code so that the numpy and lxml package should be already the right version. So that should not be needed anymore.

This morning (in europe ;) ) I found that also `gevent` could cause issues. So try the following:

1. Become root `sudo -i`
2. Go to TP4 folder: `cd /home/pi/TerrariumPI`
3. Enable python environment: `source venv/bin/active`
4. Downgrade gevent: `pip install gevent==21.8.0`

The you could try to start it manually: `python terrariumPI.py` I hope this will make it work. 
And I am trying to make the installer fix this if it is installed on a Pi Zero. But that takes more then an hour to test :(

kebs19, 2021-12-28:
Hello there,

I think that gevent was the culprit... What I did to make TPI starts:

Install in a new formatted disk, when asked to reboot, chose to not reboot. Then I did the 'pip install gevent==21.8.0' as you said in the previous message. 
TPI starts and I tried to add a new LCD 16x2 display at 0x23 address, but when I reboot it hangs and gives the following messages when debugged:

(venv) pi@terrariumpi:~/TerrariumPI $ python terrariumPI.py
2021-12-28 11:52:52,703 - INFO    - terrariumEngine       - Starting TerrariumPI 4.1.0 ...
2021-12-28 11:52:53,111 - INFO    - terrariumEngine       - Loaded 28 settings in 0.39 seconds.
Traceback (most recent call last):
  File "terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 105, in __init__
    self.notification.load_services()
  File "/home/pi/TerrariumPI/terrariumNotification.py", line 259, in load_services
    self.services[service['id']] = terrariumNotificationService(service['id'], service['type'], service['name'], service['enabled'], setup)
  File "/home/pi/TerrariumPI/terrariumNotification.py", line 404, in __init__
    self.load_setup(setup)
  File "/home/pi/TerrariumPI/terrariumNotification.py", line 423, in load_setup
    self.setup['device'] = terrariumDisplay(None, setup_data['hardware'], setup_data['address'], None if not terrariumUtils.is_true(setup_data['show_title']) else f'TerrariumPI {self.setup["version"]}')
  File "/home/pi/TerrariumPI/hardware/display/__init__.py", line 106, in __init__
    self.clear()
  File "/home/pi/TerrariumPI/hardware/display/__init__.py", line 315, in clear
    self._write_line(self.title[:self.width].ljust(self.width),0,'black')
TypeError: _write_line() takes 3 positional arguments but 4 were given

And when i hit ctrl+c:

^CKeyboardInterrupt
2021-12-28T15:01:16Z
Exception ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/monkey.py", line 911, in _shutdown
    orig_shutdown()
  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown
    t.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/thread.py", line 121, in acquire
    acquired = BoundedSemaphore.acquire(self, blocking, timeout)
  File "src/gevent/_semaphore.py", line 180, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_semaphore.py", line 249, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_abstract_linkable.py", line 521, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait
  File "src/gevent/_abstract_linkable.py", line 487, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 490, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 442, in gevent._gevent_c_abstract_linkable.AbstractLinkable._AbstractLinkable__wait_to_be_notified
  File "src/gevent/_abstract_linkable.py", line 451, in gevent._gevent_c_abstract_linkable.AbstractLinkable._switch_to_hub
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
KeyboardInterrupt:

theyosh, 2021-12-28:
Do a `git pull` and restart. I think than we should be there....

kebs19, 2021-12-28:
working now :D :D :D :D :D 

[image]

theyosh, 2021-12-28:
Pffffff, that was a journey to find it. But, now we know, I have updated the installer to install the older python packages for Pi Zero. So next time, it should work without any manual changes.

So, the notification system is a bit different then on TP3. I hope you can work it out how to use it. Use the question mark icon to get help about using placeholders in messages.

theyosh, 2022-01-05:
So, after a week, it is still working? Can we close this issue?

kebs19, 2022-01-06:
Sure, it's working like a charm :D

closed 2022-01-06 by kebs19