theyosh/TerrariumPI #632: [BUG]: Webinterface after restart not available

DanielDTWO, 2021-12-01:
[image]
[image]
[image]
[image]
**More info**
Please look first at the [FAQ](https://github.com/theyosh/TerrariumPI/wiki/FAQ) or [Known Issues](https://github.com/theyosh/TerrariumPI/wiki/Known-issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: [4]
 - Python version: [3]
 - Raspberry PI: [4]

**Describe the bug**
After a restart, because the webinterface was weriting offline. It was not possible to enter the webinterface. It happends during setting a area.

**To Reproduce / Installation steps**
See above and the debuging pictures.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Please add screenshots of the [debug screens](https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi):

**Additional context**
Add any other context about the problem here.

theyosh, 2021-12-01:
Hmm, looks like a strange (invalid) setup. Could be a bug. Easiest now is to remove the files `data/terrariumpi.db*` and restart TP4. You should then start over with your setup. If the error happens again, could you then try to make a screenshot of the area you want to add? Maybe you are doing something which I did not think off.

PS, screenshots with a mobile phone.... that a first for me :P

DanielDTWO, 2021-12-01:
Same happend again!

It was necessary to remove the terrariumpi.db also!

I guess it is in combination with light, humidity and day/ night weather!

[Image.jpeg][Image.jpeg]

[Image.jpeg][Image.jpeg]

Mit freundlichen Grüßen / regards

Ing. Daniel Zötsch

D-TWO Stage- & Lightcreations OG
Hart 48
8160 Naas
Österreich

Mobile: +43 664 / 51 00 490<tel:+43%20(0)664%20/%2051%2000%20490>
Mail: ***@***.******@***.***>
Net: www.d-two.at<http://www.d-two.at/>

Planning - Operating - Rental - SFX - Production Service

*We use Vectorworks Spotlight - L8 - Depence2*

-- Von meinem IPhone gesendet --

This message and any attachments are solely for the use of the intended recipients. They may contain privileged and/or confidential information or other information protected from disclosure. If you are not an intended recipient, you are hereby notified that you received this email in error and that any review, dissemination, distribution or copying of this email and any attachment is strictly prohibited. If you have received this email in error, please contact the sender and delete the message and any attachment from your system.

D-TWO Stage & Light Creations OG

Hart 48, 8160 Naas / Austria

Firmenbuchgericht/ Court of registry: Handelsgericht Graz
Firmenbuchnummer/ Company registration: FN 492323 f
VAT-Nr.: ATU73367509

Thank You

theyosh, 2021-12-02:
Hmm, if you tried to attach images in the email, it did not work. I do not see screenshots of the area setups...

DanielDTWO, 2021-12-02:
[Image.jpeg][Image.jpeg][Image.jpeg][Image.jpeg]

Mit freundlichen Grüßen / regards

Ing. Daniel Zötsch

D-TWO Stage- & Lightcreations OG
Hart 48
8160 Naas
Österreich

Mobile: +43 664 / 51 00 490<tel:+43%20(0)664%20/%2051%2000%20490>
Mail: ***@***.******@***.***>
Net: www.d-two.at<http://www.d-two.at/>

Planning - Operating - Rental - SFX - Production Service

*We use Vectorworks Spotlight - L8 - Depence2*

-- Von meinem IPhone gesendet --

theyosh, 2021-12-02:
Please upload trough web. Email does not work. I do not see any screenshots

DanielDTWO, 2021-12-02:
[image]

DanielDTWO, 2021-12-02:
[image]
[image]
[image]

fujexo, 2021-12-02:
Hey, I've just encountered something similar after unplugging my webcam today afternoon. It somehow restarted itself and didn't come back up anymore. Plugging the webcam back in didn't work.

I was on git commit `e0bbb4726979975b45519ecc3b3d2beb6b3415d8`. After playing around with older commits a little bit, I've just pulled the newest commit, did a `sudo ./install.sh` and voilà, there we are online again...

Wasn't able to reproduce it unfortunately. Just for completeness, here are the logs of my instance:

```
2021-12-02 16:47:24,364 - INFO    - terrariumEngine       - Starting TerrariumPI 4.0.0 ...
2021-12-02 16:47:24,431 - INFO    - terrariumEngine       - Loaded 28 settings in 0.07 seconds.
2021-12-02 16:47:24,438 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Bowil,CH&appid=xxx
2021-12-02 16:47:24,870 - WARNING - retry.api             - 'description', retrying in 0.5 seconds...
2021-12-02 16:47:25,743 - WARNING - retry.api             - 'description', retrying in 0.5 seconds...
2021-12-02 16:47:26,646 - ERROR   - terrariumEngine       - Loading weather exception: 'description'
2021-12-02 16:47:27,075 - INFO    - terrariumEngine       - Loading existing sensors from database.
2021-12-02 16:47:28,898 - INFO    - terrariumEngine       - Loaded sensor miflora temperature named 'MiFlora bluetooth sensor measuring temperature' at address 'c4:7c:8d:6d:d7:9a' with value 21.50C in 1.82 seconds.
2021-12-02 16:47:28,906 - INFO    - terrariumEngine       - Loaded sensor miflora light named 'MiFlora bluetooth sensor measuring light' at address 'c4:7c:8d:6d:d7:9a' with value 115.00lux in 0.01 seconds.
2021-12-02 16:47:28,910 - INFO    - terrariumEngine       - Loaded sensor miflora moisture named 'MiFlora bluetooth sensor measuring moisture' at address 'c4:7c:8d:6d:d7:9a' with value 51.00% in 0.00 seconds.
2021-12-02 16:47:28,914 - INFO    - terrariumEngine       - Loaded sensor miflora fertility named 'MiFlora bluetooth sensor measuring fertility' at address 'c4:7c:8d:6d:d7:9a' with value 2306.00µS/cm in 0.00 seconds.
2021-12-02 16:47:28,951 - INFO    - terrariumEngine       - Loaded sensor bme280 temperature named 'Temperatur unten' at address '76' with value 23.04C in 0.04 seconds.
2021-12-02 16:47:28,955 - INFO    - terrariumEngine       - Loaded sensor bme280 humidity named 'Luftfeuchtigkeit unten' at address '76' with value 67.64% in 0.00 seconds.
2021-12-02 16:47:28,971 - INFO    - terrariumEngine       - Loaded sensor hc-sr04 distance named 'Wassertank' at address '32,31' with value 10.38cm in 0.02 seconds.
2021-12-02 16:47:29,006 - INFO    - terrariumEngine       - Loaded sensor bme280 humidity named 'Luftfeuchtigkeit oben' at address '77' with value 41.46% in 0.03 seconds.
2021-12-02 16:47:29,008 - INFO    - terrariumEngine       - Loaded sensor bme280 temperature named 'Temperatur oben' at address '77' with value 30.62C in 0.00 seconds.
2021-12-02 16:47:29,024 - INFO    - terrariumEngine       - Scanning for new sensors ...
2021-12-02 16:47:35,129 - INFO    - terrariumEngine       - Loaded 9 sensors in 8.05 seconds.
2021-12-02 16:47:35,130 - INFO    - terrariumEngine       - Loading existing relays from database.
2021-12-02 16:47:35,432 - INFO    - terrariumEngine       - Loaded relay gpio-inverse relay named 'Licht links' at address '11' value 100.00 in 0.28 seconds.
2021-12-02 16:47:35,474 - INFO    - terrariumEngine       - Loaded relay gpio-inverse relay named 'Licht rechts' at address '12' value 100.00 in 0.04 seconds.
2021-12-02 16:47:35,481 - INFO    - terrariumEngine       - Loaded relay gpio-inverse relay named 'Regenpumpe' at address '13' value 0.00 in 0.01 seconds.
2021-12-02 16:47:35,488 - INFO    - terrariumEngine       - Loaded relay gpio-inverse relay named 'Ventilatoren' at address '15' value 0.00 in 0.01 seconds.
2021-12-02 16:47:35,491 - WARNING - hardware.relay        - Error changing relay Remote power switch (API) relay named 'Luftpumpe' at address 'http://aquarium-airpump.hcorp.c0d3.ch/api/relay/0?apikey=xxx' to state 0.0. Error: 'terrariumRelayRemote' object has no attribute '_set_hardware_value', retrying in 0.5 seconds...
2021-12-02 16:47:35,993 - WARNING - hardware.relay        - Error changing relay Remote power switch (API) relay named 'Luftpumpe' at address 'http://aquarium-airpump.hcorp.c0d3.ch/api/relay/0?apikey=xxx' to state 0.0. Error: 'terrariumRelayRemote' object has no attribute '_set_hardware_value', retrying in 0.5 seconds...
2021-12-02 16:47:36,494 - ERROR   - hardware.relay        - Error changing state for relay Remote power switch (API) relay named 'Luftpumpe' at address 'http://aquarium-airpump.hcorp.c0d3.ch/api/relay/0?apikey=xxx' to 0.0 :Error changing relay Remote power switch (API) relay named 'Luftpumpe' at address 'http://aquarium-airpump.hcorp.c0d3.ch/api/relay/0?apikey=8097F948D7D49668' to state 0.0. Error: 'terrariumRelayRemote' object has no attribute '_set_hardware_value'
2021-12-02 16:47:36,525 - INFO    - terrariumEngine       - Loaded relay remote relay named 'Luftpumpe' at address 'http://aquarium-airpump.hcorp.c0d3.ch/api/relay/0?apikey=8097F948D7D49668' value 0.00 in 1.04 seconds.
2021-12-02 16:47:37,015 - INFO    - terrariumEngine       - Loaded relay script relay named 'Blinkstick' at address '/usr/local/bin/blink.py' value 0.00 in 0.49 seconds.
2021-12-02 16:47:37,022 - INFO    - terrariumEngine       - Scanning for new relays ...
[... cut ...]
  File "/home/pi/TerrariumPI/terrariumArea.py", line 727, in load_setup
    super().load_setup(data)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 286, in load_setup
    self._time_table()
  File "/home/pi/TerrariumPI/terrariumArea.py", line 207, in _time_table
    sunrise = copy.copy(self.enclosure.weather.sunrise if 'weather' == self.mode else self.enclosure.weather.sunset - datetime.timedelta(hours=24))
AttributeError: 'NoneType' object has no attribute 'sunrise'
```

After that, it just kept hanging.

theyosh, 2021-12-03:
Hmm, I see the following:

```
2021-12-02 16:47:24,438 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Bowil,CH&appid=xxx
2021-12-02 16:47:24,870 - WARNING - retry.api             - 'description', retrying in 0.5 seconds...
2021-12-02 16:47:25,743 - WARNING - retry.api             - 'description', retrying in 0.5 seconds...
2021-12-02 16:47:26,646 - ERROR   - terrariumEngine       - Loading weather exception: 'description'
```
And I tested the weather url you are using, and that is missing some data. So I need to make an update for that. Not sure if that is the case, but it could be

DanielDTWO, 2021-12-03:
> [image] [image] [image] [image] **More info** Please look first at the [FAQ](https://github.com/theyosh/TerrariumPI/wiki/FAQ) or [Known [...]

Should the bug fix solve this problem?

theyosh, 2021-12-03:
@fujexo I guess some loading of the weather data failed for unknown reason. But I made a small update to tackle it. Do a `git pull` and restart.

@DanielDTWO The screenshots are not helping this way. Could you make the screenshots on a desktop/laptop? So that I can see more. Also, with this much area's it is hard to debug. So you should create a setup where the error happens, but not with all the area's added. Then it is almost impossible to debug.

But I get the feeling you did not configure relays on the main lights area part for the day. As that is needed. You main lights need to have daylight relays in use. Could that be the issue? Could you check that? Also, only 1 area can be main lights...

theyosh, 2021-12-03:
@DanielDTWO No, my remark about you have zero relays configured at the main lights area. You should have selected relays for the day section in the main lights area.

DanielDTWO, 2021-12-03:
@theyosh: during the first „crash“ relays were added, but in the last, no relays was added!

fujexo, 2021-12-04:
> @fujexo I guess some loading of the weather data failed for unknown reason. But I made a small update to tackle it. Do a `git pull` and restart.

Great, thanks! Updated today.

DanielDTWO, 2021-12-05:
Attached a new Debug:

(venv) pi@terrarium:~/TerrariumPI $ python terrariumPI.py
2021-12-05 19:02:18,862 - INFO    - terrariumEngine       - Starting TerrariumPI 4.0.0 ...
2021-12-05 19:02:18,929 - INFO    - terrariumEngine       - Loaded 28 settings in 0.06 seconds.
2021-12-05 19:02:18,936 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Gratkorn,AT&appid=077d558317e8ed89bfd399d16b985846
2021-12-05 19:02:20,240 - INFO    - weather.openweathermap_org_weather - Loaded new historical weather data (48 measurements) from 2021-12-03 02:00:00 till 2021-12-05 01:00:00 in 0.63 seconds.
2021-12-05 19:02:20,242 - INFO    - weather               - Loaded new weather data in 1.303 seconds.
2021-12-05 19:02:20,675 - INFO    - terrariumEngine       - Loading existing sensors from database.
2021-12-05 19:02:21,594 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-0315a1aa02ff' with value 30.44C in 0.91 seconds.
2021-12-05 19:02:22,460 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-0315a1acf8ff' with value 24.62C in 0.87 seconds.
2021-12-05 19:02:22,497 - INFO    - terrariumEngine       - Loaded sensor bme280 humidity named 'BME' at address '0x77' with value 28.67% in 0.04 seconds.
2021-12-05 19:02:22,500 - INFO    - terrariumEngine       - Loaded sensor bme280 pressure named 'BME' at address '0x77' with value 957.11hPa in 0.00 seconds.
2021-12-05 19:02:22,504 - INFO    - terrariumEngine       - Loaded sensor bme280 temperature named 'BME' at address '0x77' with value 25.70C in 0.00 seconds.
2021-12-05 19:02:22,575 - WARNING - terrariumEngine       - Measurement for sensor jsn-sr04t distance named 'Distanz' at address '29 ,31' of 997.22cm is outside valide range 0.00cm to 200.00cm during startup in 0.07 seconds. Will be updated in the next round.
2021-12-05 19:02:22,592 - INFO    - terrariumEngine       - Loaded sensor veml6075 uva named 'UVA' at address '0x10' with value 0.00µW/cm² in 0.01 seconds.
2021-12-05 19:02:22,595 - INFO    - terrariumEngine       - Loaded sensor veml6075 uvb named 'UVB' at address '0x10' with value 15.00µW/cm² in 0.00 seconds.
2021-12-05 19:02:22,598 - INFO    - terrariumEngine       - Scanning for new sensors ...
2021-12-05 19:02:30,506 - INFO    - terrariumEngine       - Loaded 8 sensors in 9.83 seconds.
2021-12-05 19:02:30,508 - INFO    - terrariumEngine       - Loading existing relays from database.
2021-12-05 19:02:30,764 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'HQI 1' at address '11' value 0.00 in 0.23 seconds.
2021-12-05 19:02:30,771 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'HQI 2' at address '13' value 0.00 in 0.01 seconds.
2021-12-05 19:02:30,778 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Heizung' at address '19' value 0.00 in 0.01 seconds.
2021-12-05 19:02:30,784 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Wasserfall' at address '16' value 0.00 in 0.01 seconds.
2021-12-05 19:02:30,790 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Pumpe' at address '18' value 0.00 in 0.01 seconds.
2021-12-05 19:02:30,797 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Ventilator' at address '21' value 0.00 in 0.01 seconds.
2021-12-05 19:02:30,805 - INFO    - terrariumEngine       - Loaded relay dc-dimmer dimmer named 'LED' at address '22' value 0.00 in 0.01 seconds.
2021-12-05 19:02:30,807 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2021-12-05 19:02:41,080 - INFO    - terrariumEngine       - Loaded 7 relays in 10.57 seconds.
2021-12-05 19:02:41,081 - INFO    - terrariumEngine       - Loading existing buttons from database.
2021-12-05 19:02:41,091 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.01 seconds.
2021-12-05 19:02:41,091 - INFO    - terrariumEngine       - Loading existing webcams from database.
2021-12-05 19:02:41,098 - INFO    - terrariumEngine       - Loaded 0 webcams in 0.01 seconds.
2021-12-05 19:02:41,099 - INFO    - terrariumEngine       - Loading existing enclosures from database.
Traceback (most recent call last):
  File "terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 152, in __init__
    self.__load_existing_enclosures()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 938, in __load_existing_enclosures
    new_enclosure.update()
  File "/home/pi/TerrariumPI/terrariumEnclosure.py", line 118, in update
    area_states[area_id] = self.areas[area_id].update(read_only)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 861, in update
    super().update(read_only)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 583, in update
    toggle_relay = self._is_timer_time(period)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 391, in _is_timer_time
    for time_schedule in self.setup[period]['timetable']:
KeyError: 'timetable'

theyosh, 2021-12-06:
I did a small code change. Do a `git pull` and restart. Lets hope it will continue.

DanielDTWO, 2021-12-06:
THX worked :)

theyosh, 2021-12-06:
Ok, then I would like to close this issue, as it is not clear if the original issue is still there. So can we close this issue?

closed 2021-12-06 by DanielDTWO