theyosh/TerrariumPI #633: New Sequent relay boards

morrell7, 2021-12-06:
Hi again Joshua,

I've recently seen my current Sequent Microsystems board start to error and my enclosure almost hit as low as 20 degrees Celsius!
A power cycle and cold start had solved it in the short term - obviously causes issues and delay in the software.

This is just a request to work in the new boards they have into v4.1 please.

Links to libraries:

https://github.com/SequentMicrosystems/4relind-rpi & https://github.com/SequentMicrosystems/8relind-rpi

Best regards & happy holidays!

Mark

theyosh, 2021-12-06:
Just added them :) Do a `git pull` and rerun the installer. The installer will install the needed libraries. So you need to run the installer after pulling the latest code.

morrell7, 2021-12-06:
Thanks for the fast response :)

theyosh, 2021-12-07:
Are you able to test this? And/or do you still have some old V3 or older boards? As I have changed the complete code for this kind of hardware. It would be nice if you could test it with at least 1 hardware device. I am pretty sure if 1 is working, they are all working.

theyosh, 2021-12-10:
Closing, this is tested and works.

closed 2021-12-10 by theyosh