theyosh/TerrariumPI #634: [Question]:

Speedster25, 2021-12-18:
Hello, first thanks for your efforts!

I installed your TerrariumPi as you described on a clean PI OS Lite. Everything went well till i tried to get on the interface. 
Here is nothing to reach.
I tried to start the service and got this error:

lines 4376-4437/4437 (END)
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit terrariumpi.service has entered the 'failed' state with result 'exit-code'.
Dec 18 14:38:12 raspberrypi systemd[1]: Failed to start TerrariumPI.
░░ Subject: A start job for unit terrariumpi.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit terrariumpi.service has finished with a failure.
░░
░░ The job identifier is 4081 and the job result is failed.
Dec 18 14:38:17 raspberrypi systemd[1]: terrariumpi.service: Scheduled restart job, restart counter is at 59.
░░ Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ Automatic restarting of the unit terrariumpi.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Dec 18 14:38:17 raspberrypi systemd[1]: Stopped TerrariumPI.
░░ Subject: A stop job for unit terrariumpi.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A stop job for unit terrariumpi.service has finished.
░░
░░ The job identifier is 4145 and the job result is done.
Dec 18 14:38:18 raspberrypi systemd[1]: Starting TerrariumPI...
░░ Subject: A start job for unit terrariumpi.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit terrariumpi.service has begun execution.
░░
░░ The job identifier is 4145.
Dec 18 14:38:18 raspberrypi python[718]: Traceback (most recent call last):
Dec 18 14:38:18 raspberrypi python[718]:   File "/home/pi/TerrariumPI/terrariumPI.py", line 4, in <module>
Dec 18 14:38:18 raspberrypi python[718]:     from gevent import monkey
Dec 18 14:38:18 raspberrypi python[718]: ModuleNotFoundError: No module named 'gevent'
Dec 18 14:38:18 raspberrypi systemd[1]: terrariumpi.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit terrariumpi.service has exited.
░░
░░ The process' exit code is 'exited' and its exit status is 1.
Dec 18 14:38:18 raspberrypi systemd[1]: terrariumpi.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit terrariumpi.service has entered the 'failed' state with result 'exit-code'.
Dec 18 14:38:18 raspberrypi systemd[1]: Failed to start TerrariumPI.
░░ Subject: A start job for unit terrariumpi.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit terrariumpi.service has finished with a failure.
░░
░░ The job identifier is 4145 and the job result is failed.
~
~

Do you know a solution?

Thanks a lot! 
Christian :)

theyosh, 2021-12-18:
Hmm, did you use the latest OS? That is bullseye. That is not supported yet. You need to use the older buster image. 

https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy

Speedster25, 2021-12-18:
Okay Thanks ill try :)

Speedster25, 2021-12-19:
Hello, to switch to the older OS solved the problem! 
Thanks a lot! You created an awesome product!!
Regards 
Christian

closed 2021-12-19 by Speedster25