theyosh/TerrariumPI #637: [FEATURE]: Show git revision in version string when admin

cnelsonsic, 2021-12-24:
**Is your feature request related to a problem? Please describe.**
When I'm deploying software it's nice to be able to see exactly what I've deployed, the overall version is good, but a git hash in addition would be better.

**Describe the solution you'd like**
I'd like to be able to show the git revision to admin users and maybe expose it via api.

**Additional context**
The ability to show admin-defined environment variables in the footer would also be very nice, maybe by a block of jinja an admin could type in a textbox.

I might end up implementing this one if it sounds okay to @theyosh and/or isn't already a thing.

theyosh, 2021-12-26:
Hi, I see what you mean. And I think it is handy to have. And for docker, it is not that hard to make. During the build we can figure out the last commit and use that hash.

And when you use git to clone this repo, then I can make some code that reads out the last git hash during startup.

But, people that just download the zip file of a release, those people will not have any git information. So then a git hash should already be added to the code, which is a bit hard, as it is only available after creating a release. So I think, for those, that information is not available..

theyosh, 2022-04-11:
This is now added in the new release. When you are logged in, you will see the commit hash in the source code at the header and in the footer of the GUI.

closed 2022-05-05 by theyosh