theyosh/TerrariumPI #639: Installation Issues on Raspberry Pi Zero and Debian Bullseye based Pi OS

mcb5637, 2022-01-04:
**More info**
I do know that the Pi Zero is not supported, thats why i do not label this as a bug report. If this is not relevant, please just close it.
All of this happened a few weeks ago, so something might have changed.
The installer script install.sh has multiple issues that prevent it from sucessfully installing TerrariumPI.
I got it to work properly after a lot off fiddling with the script and manual copying commands to console.
Everything I needed to do is down below, split into the categories of what likely caused it.

**Setup:**
 - TerrariumPI version: 4.0 (later updated to 4.1 with the same problems)
 - Python version: 3.9.2 (default for Bullseye Pi OS)
 - Raspberry PI: Zero

**Describe the bug**
the installation script finishes in a few minutes without an error message and starts the TerrariumPi service, which immediately fails.

**To Reproduce / Installation steps**
- Connect to the Pi Zero via ssh (and do first setup, if needed)
- install git
- clone this repo and checkout the 4.x.y.z branch
- execute sudo install.sh
- wait for it to finish

**Expected behavior**
Normal installation happens.

**Issues likely caused by Pi OS / Python version**
- packages `libopenexr23` and `libilmbase23` can not be installed by apt
   - this prevents the installer from installing any apt modules, as apt just exits with an error message
   - changing them to the 25 versions, as recommended by apt, works
- pip package `opencv-python-headless==4.5.4.60` does not build/download correctly, removing the version specifier works (still needs a lot of time to build it)
   - this hangs installation for hours, after the apt problems are fixed (before nothing python related seemed to work anyway)
- python gettext translate install got changed quite a bit (and i think are deprecated now). I had to remove the True from install to get it to work properly
   - https://github.com/theyosh/TerrariumPI/blob/d99938c4f4883d9cb2ba617d8f95928ef6ca49a0/terrariumEngine.py#L292
   - https://github.com/theyosh/TerrariumPI/blob/d99938c4f4883d9cb2ba617d8f95928ef6ca49a0/terrariumWebserver.py#L51

**Issues likely caused by Pi Zero**
- normal users cannot access GPIO pins, so TerrariumPI has to be started as root
   - edited `/etc/systemd/system/terrariumpi.service` to do this

**Issues i am unsure what caused them**
- pip package `RPi.GPIO` cannot be installed into the python venv, and fallback to the global one does not work. only way to fix this is to remove the venv and use the global python environent instead
   - manually installed pip requirements and edited `/etc/systemd/system/terrariumpi.service` to do this

theyosh, 2022-01-04:
Bullseye OS is not support. Please use buster (older) OS.

theyosh, 2022-01-04:
And use the 4.x.y.z branch for better Pi Zero support.

mcb5637, 2022-01-04:
Well, i did not see a requirement for buster anywhere (but maybe i just missed it).
And its running now on bullseye, so im probably not gonna change it. I just wanted you to know what i have done, so you can potentially fix it for others as well.

theyosh, 2022-01-05:
> pip package RPi.GPIO cannot be installed into the python venv, and fallback to the global one does not work. only way to fix this is to remove the venv and use the global python environment instead 

Does this work and help? https://askubuntu.com/a/1330210 Because this last issue of the list is a bit nasty. Now you have installed the packages system wide? That could cause issues with OS package versions...

I also found some fixes for the issues you have mentioned. The only problem now is that you cannot use the Raspberry PI camera, as that is totally different on Bullseye OS.

More updates are coming...

mcb5637, 2022-01-05:
I'll test that out when i have access to the Pi again (probably in a few weeks, was a present for someone else).
Yes, i have the packages installed systemwide. There is nothing else using the Pi, so i don't think i will have any issues. I'm also not using the camera.

theyosh, 2022-01-05:
Not in a hurry, and maybe not needed. I am setting up a Pi Zero with Bullseye. And I will see if at least the installer will run and if it will at least startup.

So, I would say, do not change your setup right now. As it is working, leave it :)

I will do some more testing, and I think I can make it all work.

theyosh, 2022-01-06:
Ok, I updated the installer and code to make it work in Pi Zero with Bullseye. Still the RPI cameras do not work, but the software will install all the correct modules and setup.

So in short:
- We are user root by default on a Pi Zero
- Apt packages are updated with correct version numbers on a Pi Zero
- PIP package opencv-python-headless is now using a lower version so we have a wheel package which does not need to compile during installation on a Pi Zero
- Python gettext calls are updated for all RPIs
- Python package RPi.GPIO is now working on Bullseye.

So all together, I think the next time you update your setup and run the `installer.sh` it should revert your changes to the new working setup. So the python env will be used again then. As I am also updating the systemd service scripts.

If you have time and willing to test this, that would be great. I did a full install on a Pi Zero with Bullseye, and it did fully install and run afterwards. I have not tested any sensor hardware.

theyosh, 2022-01-06:
And thanks for looking into this. Made my work a lot easier!

mcb5637, 2022-01-06:
I will test it out, when i have access to it again (should be a few weeks, as i said, it was a present for someone else).
I'll report back how it goes.

theyosh, 2022-02-13:
Hi, I close this issue now, as I have made a new release. You can reopen this issue anytime to need to.

mcb5637, 2022-03-13:
Finally got around updating, and it seems the installation is working as expected now.

theyosh, 2022-03-14:
Nice! Thanks for the feedback.