theyosh/TerrariumPI #640: Script Sensors

Zanzibar312, 2022-01-06:
First off, this software is amazing. However I know little of programming, and hit a wall with sensor scripts. 
I am trying to control a heat may with a thermistor using an arduino as a analog to digital converter. I can get my python script to output the temperature in degrees Celcius, but TerrariumPI keeps telling me:
 "Invalid reading from sensor Script sensor temperature named 'Heat Mat' at address '/TerrariumPI/test/script.py', retrying in 0.5 seconds..."

Python script: 
import serial
if __name__ == '__main__':
    ser = serial.Serial('/dev/ttyACM0', 9600, timeout=1)
    ser.reset_input_buffer()
while True:
        if ser.in_waiting > 0:
            rawserial = ser.readline()
            cookedserial = rawserial.decode('utf-8').strip('\r\n')
            temperature =cookedserial[0]+cookedserial[1]
            print(int(temperature))

Other info: currently running in docker V4.1, Python 3, and all of my other hardware is working.
When I run the script independently of terrariumpi, I get consistent temperature readings. 

Any help would be greatly appreciated.

theyosh, 2022-01-06:
Hmm, the 'issue' here is Docker :P

I think I missed this in the Docker setup. But, looking at your code, I think you can make it work by updating the docker-compose.yaml file.

Add a new volume like this:

```
- /opt/terrariumpi/scripts:/TerrariumPI/scripts
```

To the volumes list. And then you can place your script on your Pi on location: `/opt/terrariumpi/scripts` and then add it in TP4 with the path: `/TerrariumPI/scripts/[your_script].py`

Then I think it should work.

theyosh, 2022-01-06:
And I hope you can readout the serial in Docker. That is something I have not tested yet.

Zanzibar312, 2022-01-06:
I have been running the following to test the script in docker:
docker exec -u 0 -it pi_terrariumpi_1 bash
python test/script.py

This will out put consistent temperatures in terminal while testing. Also the arduino has the TX light blinking at the 1 sec interval for sending data. TerrariumPI seems to not recognize the data. My last ditch idea is that print(int()) is not giving usable data. Which may explain the "invalid reading". I will try to edit the yaml and add a new folder though.

theyosh, 2022-01-06:
In your code:

```python
import serial
if name == 'main':
  ser = serial.Serial('/dev/ttyACM0', 9600, timeout=1)
  ser.reset_input_buffer()
  while True:
    if ser.in_waiting > 0:
      rawserial = ser.readline()
      cookedserial = rawserial.decode('utf-8').strip('\r\n')
      temperature =cookedserial[0]+cookedserial[1]
      print(int(temperature))
```
I think this is it, as you did not use the code markup. So I can't see the indentation.

Why is there a while loop? And should that not be ended when there is a result? TP4 cannot handle continues running scripts.

I think you should break your while loop like this:

```python
import serial

ser = serial.Serial('/dev/ttyACM0', 9600, timeout=1)
ser.reset_input_buffer()
while True:
  if ser.in_waiting > 0:
    rawserial = ser.readline()
    cookedserial = rawserial.decode('utf-8').strip('\r\n')
    temperature =cookedserial[0]+cookedserial[1]
    print(float(temperature))
    break
```
And you do not need the `if name == 'main':` for this. And use `float` so you will get more precise values.

Zanzibar312, 2022-01-06:
I ended up editing the contrib/script_sensor.py to the following: 
`import serial

ser = serial.Serial('/dev/ttyACM0', 9600, timeout=1)
ser.reset_input_buffer()
rawserial = ser.readline()
cookedserial = rawserial.decode('utf-8').strip('\r\n')
temperature = float(cookedserial)
print(temperature)`

This seems to have fixed it. Although the data update rate is slow. Thank you for your pointers. Again, huge fan of the software and effort you have put into this. It has saved me some money, and has much more flexibility than anything I can find.

closed 2022-01-06 by Zanzibar312