theyosh/TerrariumPI #641: [BUG]: Can restart 3.10.0 without authentication?

Swiftnesses, 2022-01-09:
Hello,

I'm still running version 3, as my Pi Zero struggles to install version 4 (need to figure it out).

I've just realised that I can restart / shutdown TerrariumPI without authentication (setting is set to, 'changes'). I simply click cancel on the authenticate login box on the page and then click one of the buttons.

Is there a quick workaround I can implement to fix this? My instance is exposed to the internet for family / friends.

Best regards and Happy New Year.

theyosh, 2022-01-09:
Hmm, this sounds a bit strange. As far I can judge, this is not possible, so I guess that your browser has saved the login somehow.

And I am unable to test or make changes to TP3. As I am not running or developing it anymore for about a year.

For TP4, you should now use the 4.x.y.z branch, which has been tested on Pi Zero. Make sure you have the os Buster (Old legacy). So if you have an extra SD card, flash it with a new old os Buster, and install TP4 from the 4.x.y.z branch. It should work without any changes. And then configure TP4.

Later on, you could migrate the relay history if you want. That is the best I can advise now.

theyosh, 2022-01-27:
Did you find any solution for this? Or was there something else making it look like the authentication is not working? As I cannot change this anymore, I would like to close this issue.

closed 2022-02-07 by Swiftnesses