theyosh/TerrariumPI #642: Can I transfer TerrariumPi configurations?

cyber-shifter, 2022-01-09:
It's been a while since I checked it, but I'm finally upgrading my enclosures to TP4! I wanted to see if it was possible for me to build out my configuration on a test Pi, and then copy over the configuration to another instance of TP (running the same version). I feel like it should just be able to lift and drop in the other one, since the purpose of backing up the configuration is being able to have it to put on a new machine in case of critical failure. Please let me know what you think.

theyosh, 2022-01-11:
Hi,

it is possible to transfer the database files from on Pi to another, as long both are running the same TP4 version (cause of migrations updates in the database).

The only thing you have to fix after moving the database, is the admin password. As the password is encrypted based on the CPU serial, moving the database to a different PI, will cause all passwords unreadable. https://theyosh.github.io/TerrariumPI/features/#data-encryption

and https://theyosh.github.io/TerrariumPI/faq/reset-authentication/ (I need to update the path, as the database is now in the data folder)

cyber-shifter, 2022-01-19:
Good to know! I'm going to do the change-out a different way, but I still wanted to know this in order to restore the TP4 config from backup if something catastrophic happens. Thanks!

closed 2022-01-19 by cyber-shifter