theyosh/TerrariumPI #643: Dimmer Power

cyber-shifter, 2022-01-11:
So I got some IRF530 boards (just like the ones linked in the the source code, actually) and got everything hooked up proper. They "work", in the sense that they function as expected. However, it's the case of the full output of the dimmer only being half of the power output total. I remember this being a thing with TP3, but when I went into the code to look around for the value to change, it was all different. Is this something that can be adjusted with TP4? I'm dimming LED's to imitate sunlight. It's not critical that I dim them (they are currently switched), but it would be nice to have working. I'm happy to edit the code necessary to do this.

theyosh, 2022-01-11:
For dimmers, you have a small `calibration` option. Click on the edit form on the word `calibration` and you will get more option. There you can change the dimmer HZ. With that you can change it a bit.

For more information and some tests: https://github.com/theyosh/TerrariumPI/issues/628

cyber-shifter, 2022-01-11:
I tried that. I know historically that lowering the frequency can result in brighter LED's, but there wasn't much change between 240Hz and 10000 Hz. I know measuring the voltage this way probably isn't accurate, but I'm only getting 7.24v on the output from the dimmer pack, with a 12v input. It seems I am having the opposite problem as the person in #628. I am using a different dimmer though, I think. He started with IRF520, but they used an  IRF3708?

cyber-shifter, 2022-01-19:
@theyosh any ideas? I can't seem to tag jexkin (from #628)

theyosh, 2022-01-28:
Actually, I have no clue. As we tried everything in #628 to get the right calibration and dimming settings. Even it is the opposite way of what you have, I think I have no ideas. Sorry..

theyosh, 2022-01-28:
Maybe #645 could be an option, but that is still not sure

theyosh, 2022-02-16:
Hmm the old code is here: https://github.com/theyosh/TerrariumPI/blob/3.x.y.z/terrariumSwitch.py#L1167 Maybe that helps..?

I think I know what you mean. In the past I had two values to play with. The frequency in Hz, and a max dim value. The max dim value is gone, and I now use a different library for dimming. Based on your story, it looks like that is not always working well.

But, could it also be that you use a wrong GPIO pin? TP4 does both hardware and software PWM. The PI has 4 dedicated GPIO pins for hardware PWM. So maybe you should use one of those GPIO pins: https://pinout.xyz/pinout/pwm#

Else I have to make a some kind of legacy dimmer relay option... But that needs some thinking and testing.

Could you first see which GPIO pin you are using at the moment, and if that is not one of the 4 in the link above, try one of those 4. If that does not make a difference, please report back, and I will see if I can make a legacy dimmer option.

theyosh, 2022-02-25:
Hmm, I think I found what you mean with changing some extra values. In the old code, there was a MAX dim value which could be use the change the dimming range of the dimmer. That has been removed. But it looks like you need it for your dimmer.

That means I have to look into the dimming code, as this will affect all dimmers. Give me a week so see what I can do.

theyosh, 2022-06-17:
Do you have an update? Is you dimmer working correctly now you have the 'old' dimming option in place?

theyosh, 2022-06-26:
Closing due to no response

closed 2022-06-26 by theyosh

cyber-shifter, 2024-03-28:
My apologies, life got very chaotic after this, and I put the project down. I finally have a free Pi and am working on this again. I'm changing things about with the setup, so I won't reopen this issue. But I do have some coming. I'll make new issues for those. 

--cyber