theyosh/TerrariumPI #645: Controlling a Meanwell LED driver from TerrariumPi?

victorhooi, 2022-01-14:
I need to be able to control a Meanwell LED driver - [HLG-120H-54AB](https://www.meanwell-web.com/en-gb/ac-dc-single-output-led-driver-mix-mode-cv-cc-with-hlg--120h--54ab).

This uses a "Dimming with 1-10Vdc 10V PWM resistance".

I did see that TerrariumPi supports a [Universal AC MAINS Dimmer - MPDMv4.1](https://www.tindie.com/products/next_evo1/universal-ac-mains-dimmer-mpdmv41/), however, my understanding is that this is designed for connecting to AC mains (e.g. 240V), and then directly connecting to devices to drive them.

However, in this case, I simply want to be able to control the Meanwell driver already in place, using the control line.

What is the best way of getting this to work in TerrariumPi?

theyosh, 2022-01-27:
Hi, sorry for the late reply, but this is a bit hard... I have no experience with this kind of relays. But. after reading https://www.tme.com/Document/e15d3c7d1e6bea6e51f91444a47d1fdc/HLG-120H-SPEC.pdf I am thinking that it is possible to use a mosfet like dimmer and a 10Volt power source, to dim this device.

On page 4 you see some graphs 
[image]

And it looks like you can use the Dim + and - line as the output of your dimmer. And the dimmer needs a 10 volt power source. So that when you dim max, 0-1 Volt is getting through, and when no dim, 10 Volt is getting through.

And that voltage is used to dim the leds on the meanwell.

At least, this is what I make out of it. Testing is as always at own risk ;)

theyosh, 2022-02-13:
Damm, that would be very nice.

About the PWM, I am not sure, but I think you could try the `DC Dimmer` or the `IRF520 dimmer`. I think those two could be working. Also make sure you set the right Mhz in the calibration part.

Else. if you find some Python code that works with that transistor, then I can create a dimmer out of it for TP4.

theyosh, 2022-06-26:
Closing due to no response

closed 2022-06-26 by theyosh