theyosh/TerrariumPI #646: [FEATURE]: On/Off duration-periode in seconds not minutes

kahuwi14, 2022-01-15:
Hey @theyosh.

Happy new year and I hope, you're doing fine!

I'm finally on V4 now (exactly 4.1) and switched to the docker-version: Great work so far!!!

During the configuration, I found and changed some translation-issues but found something that could be improved imho:
In my setup, it would be more suitable, if the on/off duration would be in seconds -> finetune a sprayer works better, if you can change the duration in seconds not minutes.
The other way round: define longer periodes in seconds seems to be easier for me.

That wasn't the case in v3, was it?

Anyway, I'm open for a conversation about that. ;)

Greats

kahuwi14, 2022-01-15:
I think I was confused about the fields and didn't  unterstand it the right way.

I'm really not sure any more about the difference between On/Off-duration an power on time..
I think, we should find a better/more specific description for these fields. What do you think?

I'll then try to fix the translation again.

theyosh, 2022-01-15:
Translations could be off... I am depending on other people for that.

But does this help: https://theyosh.github.io/TerrariumPI/setup/#areas

kahuwi14, 2022-01-15:
`But does this help: https://theyosh.github.io/TerrariumPI/setup/#areas`
Kind of.... 
It was/is a bit confusing to me, but if there's nobody else with these "problems" don't mind. ;)

theyosh, 2022-01-31:
[timer.html.zip](https://github.com/theyosh/TerrariumPI/files/7974049/timer.html.zip)
If you download the zip file, unpack it, and run in the browser, does it make more clear? I hop this will make it more visual and clear what the values do

theyosh, 2022-01-31:
And an online version as well: https://theyosh.github.io/TerrariumPI/scenarios/

theyosh, 2022-02-13:
I am closing as I have tried to make it more clear and adding a interactive widget on the site to explain this. For now this is the max I can do.

closed 2022-02-13 by theyosh